aws/amazon-freertos #1408: Porting of LWIP for esp32

ashwini-kumar-87, 2019-10-15:
**Is your feature request related to a problem? Please describe.**  
We are trying to create an AP in AWS FreeRTOS for esp32 device. The current version of IDF which AWS FreeRTOS supports is 3.1.3 which doesn't have any porting documentation for LWIP. There is a miss-match too in the AWS FreeRtos software folder contents when we download the software from AWS Console (AWS IoT Core) compared to this github repo releases. For same release version the folders are missing if we download software from aws console.

**Describe the solution you would like.**  
As per the IDF documentation for simple wifi example, LWIP is required in order to create an AP through IDF way.
Currently AWS porting documentation has no proper guidelines to port LWIP. It would be a great help if we have a proper and defined way to port LWIP. IDF also have released a new version of IDF which should also be included in upcoming AWS FreeRTOS releases.  

**Additional context**  
This is issue is also related to another issue which is open - https://github.com/aws/amazon-freertos/issues/362

Thank you!

qiutongs, 2019-10-16:
Can you elaborate the mismatch you saw from AWS Console and Github release? Which folders you expected are missing from downloaded software from AWS Console?

mahavirj, 2019-11-14:
https://github.com/aws/amazon-freertos/pull/1519

abhidixi11, 2019-11-30:
Hello ashwini-kumar-87 ,
Since the PR seems to be merged,  I'm closing this issue.  If you have any further questions, please feel free to reopen, or create a new issue.
Thanks!

closed 2019-11-30 by abhidixi11

ashwini-kumar-87, 2020-01-02:
Thanks @mahavirj and @abhidixi11 to taking this. My apologies for not being able to keep track of this. I will try out this fix, thanks again.