aws/amazon-freertos #1414: MQTT demo does not work

Qspace, 2019-10-16:
**Describe the bug**
Network error occurs when Run MQTT demo with https://github.com/aws/amazon-freertos-ble-android-sdk on Android 7.0 

**System information**
- Which hardware board or part numbers? ESP32
- IDE used? VScode
- Operating System [Windows|Linux|MacOS]? Linux
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)? 201906.00_Major-966-g2dd5da127

- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server] BLE MQTT demo
[image]

- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 

**To reproduce**
Steps to reproduce the behavior:

1. ...
2. ...

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
Add any other context about the problem here.

Thank you!

qiutongs, 2019-10-16:
The log you pasted says "Failed to create BLE network connection after 0 milliseconds."
It seems IOT_BLE_MQTT_CREATE_CONNECTION_WAIT_MS was set to 0.
Did you modify the config by any chance? t might not be waiting enough before the pairing completes.

https://github.com/aws/amazon-freertos/blob/790ec2c5c471003ad8745c7c70e83d9ca1092d2f/libraries/c_sdk/standard/ble/src/services/mqtt_ble/iot_network_ble.c#L128

Qspace, 2019-10-17:
Because I use BT4.1 device so I must use just-work pairing method.
I changed IOT_BLE_NUMERIC_COMPARISON_TIMEOUT_SEC back to 30 but still got issue
[image]

`/**
 * @brief The flag to enable secure connection for the device.
 *
 * By default secure connection will be enable on all device.
 */
#ifndef IOT_BLE_ENABLE_SECURE_CONNECTION
    #define IOT_BLE_ENABLE_SECURE_CONNECTION    ( 0 ) //Q_mod 1 to 0
#endif

/* Config if set, requires encryption to access services and characteristics */

/**
 * @brief Configuration to force encryption to access all characteristics of services.
 */
#ifndef IOT_BLE_ENCRYPTION_REQUIRED
    #define IOT_BLE_ENCRYPTION_REQUIRED    ( 0 ) //Q_mod 1 to 0
#endif

/**
 * @brief Configuration to enable numeric comparison for authentication.
 *
 * If the configuration is set to 0 and #IOT_BLE_ENABLE_SECURE_CONNECTION is set to 1, then
 * device will use just works pairing.
 */
#ifndef IOT_BLE_ENABLE_NUMERIC_COMPARISON
    #define IOT_BLE_ENABLE_NUMERIC_COMPARISON    ( 0 ) //Q_mode 1 to 0
#endif

/**
 * @brief Timeout in seconds used for BLE numeric comparison.
 */
#ifndef IOT_BLE_NUMERIC_COMPARISON_TIMEOUT_SEC
    #define IOT_BLE_NUMERIC_COMPARISON_TIMEOUT_SEC    ( 30 ) //Q_mode 30 to 0
#endif`

ravibhagavandas, 2019-10-17:
Hi,
Please note that Amazon FreeRTOS supports **BLE versions 4.2 and higher**. Any prior BLE versions have not been tested for compatibility and is not recommended.

To enable just works pairing, you want to still have BLE connection as encrypted but with IO capabilities as None. So you need to modify the following configurations:

```c
#define IOT_BLE_ENABLE_NUMERIC_COMPARISON        ( 0 )
#define IOT_BLE_ENABLE_SECURE_CONNECTION         ( 0 )
#define IOT_BLE_INPUT_OUTPUT                     ( eBTIONone )
```

ravibhagavandas, 2019-11-01:
@Qspace 
Did the above answer helped in setting up the demo ? Do you have any other issues?

abhidixi11, 2019-12-04:
Hello Qspace.
I'm closing this issue, if you have any further questions please feel free to re-open. 
Thanks.

closed 2019-12-04 by abhidixi11