aws/amazon-freertos #1429: OTA data partition gets into invalid state causing further updates fail

bgintz-rsc, 2019-10-17:
### Platform:  ESP32
### AFR Branch/Commit: master/dfb8d0e
#### Issue:  A sequence of OTA attempts will set the boot flags such that subsequent OTA's will fail.

### Procedure to reproduce

1. Load code normally into the factory partition - device boots into factory image. Boot flags reported as:  "W (18421) esp_ota_ops: otadata partition is invalid, factory/ota_0 is boot partition"

2. Attempt to OTA an image with the same revision as the one running (into 0x1a0000)
   - Device reboots with the following boot flags (before version check) : 
      - flags/seq:0x1/0x1, pflags/seq:0xffffffff/0x0 
    - Version check fails:  
       - "Failing job. We rebooted and the version is still the same".  
       - Job status reports: "rejected: 0x23000000"
       - Boot flags set to:  aflags/seq:0x3/0x1, pflags/seq:0xffffffff/0x0
   - Device reboots
       - Bootloader reports "ota data partition invalid, falling back to factory"
       - boot flags still set to: aflags/seq:0x3/0x1, pflags/seq:0xffffffff/0x0
       - (note: active image is set to invalid)
3. Attempt another OTA this time with an image that has a higher revision (so it should succeed)
      - normal bootloader messages:
         - "boot: Loaded app from partition at offset 0x1a0000"
         - "boot: Set fw to pending validation"
      - boot flags: aflags/seq:0x3/0x1, pflags/seq:0x1/0x1
      - Update fails: 
         - "Job in self test but platform state is not!"  (prvOTAUpdateTask log message)
         - Msg: {"status":"FAILED","statusDetails":{"reason":"rejected: 0x26000000"}}

### Partition table:
I am using a partition table with a factory, test and two ota partitions as shown below.  

nvs,      data, nvs,     0x10000,  0x6000
otadata,  data, ota,     0x16000,  0x2000
phy_init, data, phy,     0x18000,  0x1000
factory,  0,    factory, 0x20000,   1500K
ota_0,    0,    ota_0,   ,          1500K
ota_1,    0,    ota_1,   ,          1500K
test,     0,    test,    ,           200K
calib,       data, nvs,	 ,         0x3000   
data,  data, nvs,     ,        0x10000

qiutongs, 2019-10-17:
We are actively working on this problem.

pvyawaha, 2019-10-18:
Hello,

Thank you for sharing the details about the issue you are facing. The official port supports only two application partition ota_0 and ota_1 without the factory image partition. The OTA pal is also developed and tested for two application partitions one of which is active and other one is passive. The default partition table is 

nvs,      data, nvs,     0x10000,  0x6000
otadata,  data, ota,     0x16000,  0x2000
phy_init, data, phy,     0x18000,  0x1000
ota_0,    0,    ota_0,   0x20000,  1500K
ota_1,    0,    ota_1,   ,         1500K
storage,  data, nvs,     ,         0x10000

The issue seems to be due to addition of a third partition for factory image and from the part of logs that you shared it seems the boot flags are not correctly set for active partition (should be 0x01 and not 0x03) during the process which results in job getting rejected. But if you do not need the factory image partition the fastest solution is using two application partitions ota_0 and ota_1.

If you have requirement to support three application partitions on the device including factory image we can look into how we can provide solution for it.

bgintz-rsc, 2019-10-18:
Thanks you for your response.  Yes - we had determined that was the cause and agree with your comment on the issue being the boot flags in an incorrect state as reported in this issue.  We were however unaware that the ESP32 factory image partition is not supported by AFR.  We believe this is a valuable feature to permit deployed devices in the field to be recovered from what may essentially be a "bricked" state due to some unforeseen OTA failure.

ahmedwahdan, 2019-10-18:
Hello @bgintz-rsc 
Could you please mention the debugger are you using for ESP32? How do you cope with the 2 break points limitation?
Thanks and sorry for the non-related question.

bgintz-rsc, 2019-10-19:
This is my environment

- Eclipse IDE
- openocd-esp32
- ESP-PROG (JTAG)

pvyawaha, 2019-11-04:
Hello,

The fix for using factory partition along with ota_0 and ota_1 partition is provided in the branch -

https://github.com/aws/amazon-freertos/commits/fix/espressif-factory-partition

The fix is located in the file vendors/espressif/boards/esp32/ports/ota/aws_esp_ota_ops.c  

The partition table is updated in this branch to use factory partition and the size of partitions are adjusted for testing on smaller flash esp32 devices.

nvs,      data, nvs,     0x10000,  0x6000
otadata,  data, ota,     0x16000,  0x2000
phy_init, data, phy,     0x18000,  0x1000
factory,    0,    factory,   0x20000,  1200K
ota_0,    0,    ota_0,   ,         1200K
ota_1,    0,    ota_1,   ,         1200K
storage,  data, nvs,     ,         0x10000

Please let us know if this works for you and if you have more questions.

abhidixi11, 2019-12-04:
Hello  bgintz-rsc ,
I'm closing this issue.  Please feel free to re-open if you have any further questions. 
Thanks!

closed 2019-12-04 by abhidixi11