aws/amazon-freertos #1436: crash when using IotNetworkInterface->setReceiveCallback with lwip porting

eddid, 2019-10-19:
hi there,

I meet crash when using `IotNetworkInterface->setReceiveCallback`, you could recurring the issue with following code piece

```
#include "platform/iot_network.h"
#include "platform/iot_network_afr.h"
void Echo_ReceiveCallback(void *pNetworkConnection, void *pReceiveContext)
{
    char buffer[32];
    const IotNetworkInterface_t *pNetworkInterface = (const IotNetworkInterface_t *)pReceiveContext;
    size_t recvLength;

    recvLength = pNetworkInterface->receive(pNetworkConnection, (uint8_t *)buffer, sizeof(buffer));
    vLoggingPrintf("%s %d %.*s\n", __FUNCTION__, recvLength, recvLength, buffer);
}

void handle_echo_task(void *pvParameters)
{
    const IotNetworkInterface_t *pNetworkInterface = IOT_NETWORK_INTERFACE_AFR;
    IotNetworkServerInfo_t *pServerInfo = (IotNetworkServerInfo_t *)pvParameters;
    void *pNetworkConnection = NULL;
    size_t msgLength, sendLength;
    IotNetworkError_t networkStatus = IOT_NETWORK_SUCCESS;

    IotSdk_Init();

    networkStatus = pNetworkInterface->create(pServerInfo, NULL, &pNetworkConnection);
    if (networkStatus != IOT_NETWORK_SUCCESS)
    {
        vLoggingPrintf("%s:%d networkStatus=%d\n", __FUNCTION__, __LINE__, networkStatus);
        goto exit;
    }

    networkStatus = pNetworkInterface->setReceiveCallback(pNetworkConnection, Echo_ReceiveCallback, (void *)pNetworkInterface);
    if (networkStatus != IOT_NETWORK_SUCCESS)
    {
        vLoggingPrintf("%s:%d networkStatus=%d\n", __FUNCTION__, __LINE__, networkStatus);
        goto exit;
    }

    vTaskDelay(1);

    msgLength = strlen("IOT_NETWORK_INTERFACE_AFR");
    sendLength = pNetworkInterface->send(pNetworkConnection, "IOT_NETWORK_INTERFACE_AFR", msgLength);
    if (sendLength != msgLength)
    {
        vLoggingPrintf("%s:%d %d != %d\n", __FUNCTION__, __LINE__, sendLength, msgLength);
        goto exit;
    }

exit:
    if (NULL != pNetworkConnection)
    {
        networkStatus = pNetworkInterface->close(pNetworkConnection);
        vLoggingPrintf("%s:%d networkStatus=%d\n", __FUNCTION__, __LINE__, networkStatus);

        networkStatus = pNetworkInterface->destroy(pNetworkConnection);
        vLoggingPrintf("%s:%d networkStatus=%d\n", __FUNCTION__, __LINE__, networkStatus);
    }

    IotSdk_Cleanup();
    vTaskDelete(NULL);
}
```

step
1. add above code to project and register command like following
```
    if (0 == strcmp(argv[1], "echo"))
    {
        static IotNetworkServerInfo_t xServerInfo;

        if (argc < 4)
        {
            bk_printf("Usage: %s echo url port\n", argv[0]);
            return;
        }
        xServerInfo.pHostName = argv[2];
        xServerInfo.port = atoi(argv[3]);
        /* Create the task to run unit tests. */
        xTaskCreate( handle_echo_task,
                     "echo_task",
                     mainTEST_RUNNER_TASK_STACK_SIZE,
                     &xServerInfo,
                     tskIDLE_PRIORITY + 1,
                     NULL );
        return;
    }
```
2. compile project and burn firmware
3. run an echo server
4. start up board and connect with WIFI
5. connect to echo server with above command like "xxx echo echo_server echo_port"

pvyawaha, 2019-10-24:
Hello, 

Thank you for reporting this , we are actively working on this and will get back soon.

ravibhagavandas, 2019-11-01:
@eddid 
The issue seems related to the known issue we have in lwip port that, receiving, sending and closing sockets in different threads is not safe if `LWIP_NETCONN_FULLDUPLEX ` is not defined. For more details you can check #461.
Are you using the lwip port from Amazon FreeRTOS under [https://github.com/aws/amazon-freertos/tree/master/libraries/3rdparty/lwip](libraries/3rdparty/lwip) ?
If so, please watch here for more updates, as we are fixing some regressions in full duplex mode in our port for lwip.

eddid, 2019-11-04:
> @eddid
> The issue seems related to the known issue we have in lwip port that, receiving, sending and closing sockets in different threads is not safe if `LWIP_NETCONN_FULLDUPLEX ` is not defined. For [...]

hi @ravibhagavandas ,
I have test with LWIP_NETCONN_FULLDUPLEX enabled, and the issue still exists.

regards,
eddid

ravibhagavandas, 2019-11-05:
@eddid  Can you point to the LWIP version or commit ID you are using so that I can quickly reproduce the issue locally ?
Also there was a recent fix made for `LWIP_NETCONN_FULLDUPLEX ` in PR #1496. Did you include those changes as well in testing ?

eddid, 2019-11-06:
@ravibhagavandas
I port baseline 201906.00_Major to our EVB, so the version of LWIP should be STABLE-2.0.3, you could refer to the code at:
https://github.com/bekencorp/bk7231u_freertos_aws

the issue still exists when test with latest code of amazon-freertos, and I could workaround this issue with two methods:
a) modify lwipopts.h like mediatek
https://github.com/aws/amazon-freertos/blob/master/vendors/mediatek/boards/mt7697hx-dev-kit/aws_tests/config_files/lwipopts.h
b) predefine IOT_NETWORK_RECEIVE_TASK_PRIORITY to lowest priority in config
```
#define IOT_NETWORK_RECEIVE_TASK_PRIORITY      ( tskIDLE_PRIORITY )
```

the different of lwipopts.h between mediatek and us is
```
#define MEM_USE_EXTERNAL_MGMT          (1)
```
we using system heap to malloc/free memory for lwip, other than lwip private heap,
so it might trigger task switch when malloc memory in lwip 
```
void *pvPortMalloc( size_t xWantedSize )
{
	...
	vTaskSuspendAll();
	...
	( void ) xTaskResumeAll();
	...
}
```

so, above two workaround just avoid some task switch when "NetRecv" running, that will avoid "NetRecv" task in pending state when another task close the socket.
a better solution should be close the socket safely, 
and the test solution should be enhance the robustness of lwip :-)

ravibhagavandas, 2019-11-06:
Hi @eddid ,

I am **not able to reproduce** the crash using above code with amazon freertos latest commit id `c066b4998e263ff9e06487bfd21ebb50ee8450ec` and [lwip version 2.1.2 ](https://github.com/aws/amazon-freertos/tree/master/libraries/3rdparty/lwip) (part of third party folder in amazon freertos),  on Mediatek board. 

Here are the logs:
```
loader init
fota partition base: 0x266000, length: 0x18a000, offset from end: 0x200
magic ver: -1
fota: TMP is empty, skip upgrade
jump to (0x10079000)
0 0 [wrap] Starting up
1 34 [wrap] Write certificate...
2 718 [wrap] WiFi module initialized. Connecting to AP...
3 718 [wpa_supplicant] ========= Supplicant Ready =======4 6865 [wrap] WiFi Connected to AP. Creating tasks which use network...
5 6867 [echo_task] [INFO ][INIT][6867] SDK successfully initialized.
6 6921 [echo_task] handle_echo_task:37 Network creation succeeded
7 6922 [echo_task] handle_echo_task:60 networkStatus=0
8 7223 [echo_task] handle_echo_task:63 networkStatus=0
9 7223 [echo_task] [INFO ][INIT][7223] SDK cleanup done.
```

However with older version, `201906.00_Major` which uses LWIP `STABLE-2.0.3` version, I am able to reproduce the crash using the same code on Mediatek board. Logs:

```
 545 [wrap] WiFi module initialized. Connecting to AP...
2 546 [wpa_supplicant] ========= Supplicant Ready =======3 6149 [wrap] WiFi Connected to AP. Creating tasks which use network...
4 6150 [echo_task] [INFO ][INIT][6150] SDK successfully initialized.
5 6199 [echo_task] handle_echo_task:38 Network creation succeeded
6 6200 [echo_task] handle_echo_task:61 networkStatus=0

In Bus Fault Handler
SCB->CFSR = 0x00008200
Bus fault: A precise data access error has occurred @x43ffffff
SCB->BFAR = 0x43ffffff
r0  = 0x2001754c
r1  = 0x44000000
r2  = 0x43fffffd
r3  = 0x00000000
```

Also from the issue #461 , its mentioned that the bug exists in `lwip 2.0.3 version` 

> We have confirmed the behavior that @caleb-garrett reported here. With LwIP 2.0.3 this will be a problem because this version of LwIP does not support closing a socket from a thread different from the one you are reading the same socket with.

So, could you bump up the LWIP version to `2.1.2` and retest with LWIP_NETCONN_FULLDUPLEX mode enabled ? Additionally to get all the latest bug fixes, you can pull the LWIP code from our third party folder [here](https://github.com/aws/amazon-freertos/tree/master/libraries/3rdparty/lwip)

eddid, 2019-11-07:
hi @ravibhagavandas ,
test with lasted code with `LWIP_NETCONN_FULLDUPLEX` enable, the bug could not be reproduced.
thank you for your help.

closed 2019-11-07 by eddid

ravibhagavandas, 2019-11-07:
Thanks for confirming. I have closed the related PR which has the workaround, which is no longer needed.