aws/amazon-freertos #1458: OTA blocks retransmission

ahmedwahdan, 2019-10-25:
**Describe the bug**
The OTA agent has continuous receiving of duplicate OTA blocks 

**System information**
- Board : ESP32
- Operating System : Windows
- 08.2019
- Project [Greengrass+OTA]

**Screenshots or console output**
[image]

Thank you!

pvyawaha, 2019-10-25:
Hello,

Thank you for reporting this issue and sharing logs.

Please set the request timer to a higher value in the OTA agent config using the config parameter  otaconfigFILE_REQUEST_WAIT_MS . From the logs you shared it seems the request are being sent faster than the data blocks being received. You can also control the number of blocks requested by lowering the OTA config otaconfigMAX_NUM_BLOCKS_REQUEST . Please let me know how this works for you and if you have further questions.

pvyawaha, 2019-11-04:
Hello, 

Please reopen this issue if you have more questions related to this or post a new issue for questions on other topics. Closing this issue now.

closed 2019-11-04 by pvyawaha