aws/amazon-freertos #1463: Support T-HEAD processor

cui632, 2019-10-28:
Hi , 
I am desired to port T-HEAD processor in FreeRTOS. the porting project is done. I wonder to know how to request to merge it into FreeRTOS. 

Vincent

RichardBarry, 2019-10-28:
Grateful if you can provide a link to the T-HEAD processor - thanks.

cui632, 2019-10-28:
https://www.t-head.cn/product/s802?spm=a2ouz.12987056.0.0.429248abc8ds69&lang=en

Richard, please have a look. 
We still have RISC-V core which is scheduled to support FreeRTOS in future 

发自我的iPhone

cui632, 2019-10-29:
Hi Rechard

Have you check the processor link I paste?

Vincent 

发自我的iPhone

RichardBarry, 2019-10-29:
It is not clear what the core is from the link.  Could you please contact me using the 'business contact' link on https://www.freertos.org/contact so we can discuss in more detail - thanks.

lundinc2, 2019-11-06:
Hello, I am closing this issue as the conversation has moved to email. Please reopen if you would like to discuss this further via this forum.

closed 2019-11-06 by lundinc2