aws/amazon-freertos #1468: HTTPS (TLS) fails if MQTT connection exists

bgintz-rsc, 2019-10-29:
***Overview***
We are  using the AFR HTTPS client to GET a file.  The code is a modified version of the synchronous HTTPS get demo code (modified for our host and path).
- If the HTTP client (GET)  is run before connecting to the AWS broker (but after connecting to WiFI it succeeds.  
- If the GETis executed after connecting to the broker it fails with a TLS handshake error.  

***Additional Troubleshooting Steps***
- Disconnecting the MQTT session will permit the HTTPS connect to succeed
- Placing the HTTPS connect and GET in its own task still fails if there is an MQTT session  

***Architectural Questions***
Since they are separate connections (sockets) one would not expect any interaction between the two connections.  
- Is it required to synchronize use of the WiFi network (via a mutex for example) when two tasks are using different sockets? 
- Is there anything else that we need to be concerned with to permit the network to maintain an MQTT broker connection while making intermittent HTTPS GETs? 

**System information**
- ESP32
- Eclipse
- OSX
- AFR:  master dfb8d0e2f (Oct 15, 2019 9:43 pm)

**Note**
Previously reported via email.  Reposted here for visibility

Thank you!!  

Best
Bobby

pvyawaha, 2019-10-29:
Hello,

Thank you for sharing details about the issue you are facing.

The HTTP client (GET) should work after connecting to broker with the sequence of operation that you mentioned. The TLS handshake error that you are getting seems to be due to memory allocation failures during that process as enough memory is not available.

 We tried this using two tasks , one is used for connecting to broker and after that another is doing the HTTP requests for downloading file. However the default configuration parameter `configENABLED_NETWORKS `  in file `aws_iot_network_config.h` was updated to disable the BLE network so that we have enough heap available. 

Please let us know if  this works for you and if you also have requirement for using BLE in same project along with MQTT & HTTP.

bgintz-rsc, 2019-10-29:
Thank you for the quick response. Our app does not have BLE enabled at the time we are trying this and I checked the heap.  Will test again and verify that this is not a memory allocation issue.

ravibhagavandas, 2019-11-05:
@bgintz-rsc 
Were you able to verify if this was a memory allocation issue ? Could you post the logs from your test ?

bgintz-rsc, 2019-11-05:
I have not been able to confirm that now but looking at heap size and stack allocations.  I have confirmed that we free up the BLE allocation when we boot with Wifi.

bgintz-rsc, 2019-11-05:
Hello https://github.com/ravibhagavandas

After doing some analysis I have found that I can run both connections if there is enough heap available. However to accomplish this I had to disable almost everything in my system to and then did some further profiling of heap allocation.

In my test I snapshot heap before and after calling IotMqtt_Connect().

I record 125K free heap before the call and 54K after the call.  So the open MQTT connection is consuming 72Kbytes of RAM.  Is this expected behavior?   Is there any way to reduce this?

Thank you

lundinc2, 2019-11-06:
Hello @bgintz-rsc ,

I have profiled based on dfb8d0e using cmake and the default MQTT demo we provide in AFR.   

I called xPortGetFreeHeapSize() before and after calling IotMqtt_Connect(), the amount of heap consumed by this API call was 59.4K. 

> Is this expected behavior?
This is about how much heap usage you can expect.

> Is there any way to reduce this?
To reduce the heap impact of MQTT it is necessary to tweak the lower level network libraries MQTT depends on, for example TCP and TLS libraries. 

Does this satisfy your question?

abhidixi11, 2019-12-04:
Hello bgintz-rsc ,

I'm closing this issue, assuming you don't have any further question.  If you do have more questions, please reopen the issue.
Thanks.

jnorgan, 2020-01-02:
@lundinc2 What modifications to the network libraries would you recommend? I'm working on a device that needs BLE, an active connection to MQTT, AND the ability to create a separate HTTPS connection for file retrieval. I am running out of heap as seems to be expected for this use case.

lundinc2, 2020-01-03:
@jnorgan,

What device are you using? Also what network stacks are you using to run MQTT, is it FreeRTOS+TCP and mbedTLS?

jnorgan, 2020-01-03:
@lundinc2 
> What device are you using? Also what network stacks are you using to run MQTT, is it FreeRTOS+TCP and mbedTLS?

I'm using an ESP32. Yep, using FreeRTOS+TCP and mbedTLS.

ravibhagavandas, 2020-01-03:
@jnorgan 
In your use case, does BLE needs to be kept running when creating MQTT and HTTPS connections. If not, you can disable BLE stack and re-enable it later when needed. PR #1663 adds this functionality along with a sample usage to disable unused networks and re-enable it later. Disabling BLE stack can reclaim close to 20K of memory for ESP32.

lundinc2, 2020-01-06:
@jnorgan,

I will split this post into two for readability.

Regarding FreeRTOS+TCP, here is a guide for what you can modify in order to reduce the memory footprint, it will take some tweaking but you should be able to adjust the default configurations to fit your constraints. I want to emphasize that you will need to test these values yourself, I cannot recommend an optimal solution as I am not aware of what kind of traffic your device will see, and there is a risk that you will start to see failures when tuning the TCP resources.

An example: 
https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_TCP/TCP_IP_Configuration_Examples.html

Configuration outline:
https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_TCP/TCP_IP_Configuration.html

lundinc2, 2020-01-06:
@jnorgan,

Regarding mbedTLS on the ESP32 platform.

As a precursor to my example, please see this thread with an explanation from Dan #1250. This is still a work in progress but it should guide you to a smaller memory footprint.

I have adjusted some of the settings for our hosted ESP32 project. See this example on my fork:
https://github.com/lundinc2/amazon-freertos/commit/e65185e673b2d6ee5a58467cf8f52679f66919aa

Note that I have disabled all mbedtls key exchange algorithms except ECDHE_ECDSA. It is necessary to use the different root CA (ATS3) as it is the EC based root CA used by IoT Core. 
You can read more about this here: https://docs.aws.amazon.com/iot/latest/developerguide/server-authentication.html. 

Also it is important that you use the ATS endpoint, this will not work for IoT Core endpoints that lack the ATS postfix. (See above link). 

Lastly I recommended that you use ECC based keys. Here is an example of how you could generate a key and certificate to test connecting with IoT Core. https://aws.amazon.com/blogs/iot/elliptic-curve-cryptography-and-forward-secrecy-support-in-aws-iot-3/ 

You can further optimize mbedTLS, by observing the maximum TLS packet sizes coming in and out, by enabling MBEDTLS_DEBUG. Then adjust the inbound and outbound mbedTLS buffer sizes. 

Note: All of these mbedtls configuration changes can be done with the idf menuconfig command.

By following these steps, this was the evolution of my TLS Handshake:
Baseline: RSA Cert + Private key, with default handshake algorithms
54116 Bytes
 
EC only handshake  + RSA Certs (No buffer size changes):
43200 Bytes
 
EC handshake + Buffer size changes + RSA Certs:
34116 Bytes
 
EC handshake + Buffer size changes + EC Certs:
28836 Bytes

I hope this helps, and please reply here if this is helpful, or if you should need any further help.

lundinc2, 2020-01-07:
I will note that various network conditions should be checked. This was test on an strong WiFi signal, with a simple MQTT connection to IoT Core. Using a different endpoint or using OTA will need further testing, to ensure buffer size changes do not break them.

dcgaws, 2020-04-13:
Hello @bgintz-rsc please reopen if you have additional questions or data to add on this topic. Thank you!

closed 2020-04-13 by dcgaws