aws/amazon-freertos #1489: With external Wi-Fi module FreeRTOS Qualification is possible?

prvthkr, 2019-10-31:
Hi,

We have ported our MCU to [amazon-freertos](https://github.com/aws/amazon-freertos) and able to connect with the AWS cloud using external Wi-Fi module [ESP8266](https://en.wikipedia.org/wiki/ESP8266).

Now, we want to qualify the [FreeRTOS Qualification](https://docs.aws.amazon.com/freertos/latest/qualificationguide/afr-qualification.html) but our MCU does not have either Wi-Fi or ethernet, will this MCU is eligible for the FreeRTOS Qualification?

Is it compulsory to have either Wi-Fi or Ethernet for the FreeRTOS Qualification? 
Or, 
With an external Wi-Fi module, it will qualify?

Could anyone, please help me with the above question?
Sorry for my bad English.

dnoal, 2019-10-31:
To your question:  Is your configuration eligible for FreeRTOS Qualification?  YES!  We have existing qualifications that use Wifi modules similar to your case and offload various functions to the Wifi module including TCP/IP and TLS.  An example is the STM32L4 Discovery Kit that uses an Inventek Wi-Fi module.  Please go ahead and follow the DQP submittal process (https://aws.amazon.com/partners/dqp/)  and let us know how we can help with your efforts.  See this document which specifically shows how you can identify that features like TCP and TLS are offloaded:  https://docs.aws.amazon.com/freertos/latest/qualificationguide/freertos-qualification.html.

prvthkr, 2019-11-04:
Hi @dnoal, Thank you very much for your help.
I'll procedure further if any issues will come, will definitely create the new issue for the same. 

Thanks

**I'm closing this thread (issue).!**

closed 2019-11-04 by prvthkr