aws/amazon-freertos #1507: TimeOut_t doesn't work with vTaskSetTimeOutState

D252874, 2019-11-07:
**Describe the bug**
A clear and concise description of what the bug is.

**System information**
- somlabs VisionCB-RT-STD
- MCUXpresso
- Windows
- V10.0.1
- Custom Application
- 
```cpp
void modenTerminalTask(void* param){
	TickType_t xTicksToWait = 5000;
	TimeOut_t xTimeOut;
	char text[2] = {0};
	char resived[2] = {0};
	while(1)
	{
		LPUART_ReadBlocking(LPUART1, (uint8_t*)text, 1);
		LPUART_WriteBlocking(LPUART2, (uint8_t*)text, 1);
		vTaskSetTimeOutState(xTimeOut);

		while(strlen(resived) >= 1)
		{
			LPUART_ReadBlocking(LPUART2, (uint8_t*)resived, 1);
			if (xTaskCheckForTimeOut( &xTimeOut, &xTicksToWait ) != pdFALSE)
			{
				printf("break");
				break;
			}

		}
		LPUART_WriteBlocking(LPUART1, (uint8_t*)resived, 1);
		resived[0] = 0;
	}
}
```

**Expected behavior**
that my ide doesn't give an error on this line
		vTaskSetTimeOutState(xTimeOut);

**To reproduce**
Steps to reproduce the behavior:
NVT

**Code to reproduce the bug**
look above

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

thanks in advance
if it is not an bug what should i use instead then

lundinc2, 2019-11-07:
Hello @D252874,

I see on the line
```c
 vTaskSetTimeOutState(xTimeOut);
```
A TimeOut_t struct is passed in, but vTaskSetTimeOutState takes a pointer to a TimeOut_t struct.

So if you change said line to
```c
 vTaskSetTimeOutState(&xTimeOut);
```
That should fix the error you are seeing on that line.

D252874, 2019-11-11:
oke this fixed it

closed 2019-11-11 by D252874