aws/amazon-freertos #1514: BLE advertisement does not work since minInterval and maxInterval are zero

cy-vaira, 2019-11-08:
**Describe the bug**
BLE advertisement does not work with the BLE GATT Server demo. 

**System information**
- Which hardware board or part numbers?
Cypress CY8CPROTO-062-4343W (partner qualified). Should be able to reproduce the issue on any device. 
- IDE used
ModusToolbox 2.0 IDE
- Operating System [Windows|Linux|MacOS]
Windows
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
201908.00
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
BLE GATT Server demo

**Expected behavior**
Device should appear on the list when scanned for BLE devices. 

**To reproduce**
Steps to reproduce the behavior:
Please replace /cypress/boards/CY8CPROTO_062_4343W with any other board that supports BLE. 

1. Add #define IOT_BLE_ADD_CUSTOM_SERVICES ( 1 ) to vendors/cypress/boards/CY8CPROTO_062_4343W/aws_demos/config_files/iot_ble_config.h 
2. Open vendors/cypress/boards/CY8CPROTO_062_4343W/aws_demos/config_files/aws_demo_config.h, comment out #define CONFIG_MQTT_DEMO_ENABLED, and add #define CONFIG_BLE_GATT_SERVER_DEMO_ENABLED.
3. Open vendors/cypress/boards/CY8CPROTO_062_4343W/aws_demos/config_files/aws_demo_config.h, update  #define democonfigNETWORK_TYPES     ( AWSIOT_NETWORK_TYPE_BLE)
4. Open vendors/cypress/boards/CY8CPROTO_062_4343W/aws_demos/config_files/aws_iot_network_config.h and update #define  configENABLED_NETWORKS  ( AWSIOT_NETWORK_TYPE_BLE | AWSIOT_NETWORK_TYPE_WIFI )
5. Build and Program. 
6. Once the device has started up, scan for the device using any BLE app. 

Thank you!

cy-vaira, 2019-11-08:
The issue is resolved when the .minInterval & .maxInterval of _advParams are set to non-zero value. This structure is in libraries/c_sdk/standard/ble/src/iot_ble_gap.c file. 

`
static IotBleAdvertisementParams_t _advParams =
{
    .includeTxPower    = true,
    .name              = { BTGattAdvNameShort,          IOT_BLE_DEVICE_SHORT_LOCAL_NAME_SIZE},
    .setScanRsp        = false,
    .appearance        = IOT_BLE_ADVERTISING_APPEARANCE,
    .minInterval       = IOT_BLE_ADVERTISING_CONN_INTERVAL_MIN,
    .maxInterval       = IOT_BLE_ADVERTISING_CONN_INTERVAL_MAX,
    .serviceDataLen    = 0,
    .pServiceData      = NULL,
    .manufacturerLen   = 0,
    .pManufacturerData = NULL,
    .pUUID1            = ( BTUuid_t * ) &_advUUID,
    .pUUID2            = NULL
};
`

lundinc2, 2019-11-08:
Hello @cy-vaira,

I am investigating this issue.

ravibhagavandas, 2019-11-09:
Hi @cy-vaira 

The connection intervals (min and max) is set to 0 to use the stack specific default values. Our BLE HAL layer API specification has this mentioned [here](https://github.com/aws/amazon-freertos/blob/master/libraries/abstractions/ble_hal/include/bt_hal_manager_adapter_ble.h#L100), though I agree documentation is unclear. 

I have added a PR to fix the documentation #1515 

So the vendor implementation needs to use stack specific default values if the min and max connection intervals are set to 0.

Does that answer your question ?

cy-vaira, 2019-11-12:
Hi @ravibhagavandas, 

Thanks for the response. I believe end-users will be often configuring the connection interval parameters. What is the way for an end-user to do that?

ravibhagavandas, 2019-11-18:
@cy-vaira  

Thanks for your response.

After checking through the code, I think  it should also pass in`IOT_BLE_ADVERTISING_CONN_INTERVAL_MIN` an `IOT_BLE_ADVERTISING_CONN_INTERVAL_MAX`  as you described above (so that end user can configure the connection intervals). We are working on a fix for this.

closed 2019-11-19 by ravibhagavandas

ravibhagavandas, 2019-11-23:
Hi @cy-vaira 

I wanted to let you know that we have to revert back the above change for connection interval parameter in this PR #1545.

Reason being, the allowable size of the advertisement data is 31 bytes. With the existing parameters, advertisement data adds up to 30 bytes as follows
```Flags - 3 bytes(length - 1 byte, type -1 byte, value - 1 byte)
Service UUID 128 bit - 18 bytes (length - 1 byte, type -1 byte, value - 16 bytes)
Short local name - 6 bytes (length - 1 byte, type -1 byte, value - 4 bytes)
TX power - 3 bytes (length - 1 byte, type -1 byte, value - 1 byte)
```
So we have removed slave connection interval params from advertisement data (by setting the values to 0 in advertisement params) and enabling them in scan response data.
The parameter is optional as per bluetooth specification and can be included in either advertisement or scan response data.

 PR #1545 also updates the HAL API documentation mentioning the same.

cy-vaira, 2019-12-05:
Hi @ravibhagavandas, 

We see an issue with your solution. It's not a good idea to define the advertising interval via a macro (IOT_BLE_ADVERTISING_INTERVAL). Advertising interval is something user may want to change during the run time. For example, advertise rapidly for some time and switch to a lower advertising rate if no connection has been made. This is to save power. 

Either there should be a GAP API to specify the advertising interval or the IotBleAdvertisementParams_t structure should have a member for specifying it.

ravibhagavandas, 2019-12-05:
@cy-vaira 

With the current HAL API, we don't expose this parameter to the user, since advertisement interval is a low level controller parameter and its often better for vendors to configure this with sane values based on power consumption.
However the use case of setting advertisement intervals at run-time makes sense. We will look into how to best support this in our HAL layer.

PR #1534 (2019-11-19): Pass BLE connection interval params from user
Pass BLE connection interval params from user

Description
-----------
Add ability to pass connection interval parameters from user in the advertisement message.
This PR also fixes #1514

Checklist:
----------

- [x] I have tested my changes. No regression in existing tests.
- [ ] My code is Linted.

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
PR #1658 (2019-12-31): Add advertisement interval support in BLE HAL
 Add advertisement interval support in BLE HAL

Description
-----------
Add new params `usMaxAdvInterval` and `usMinAdvInterval` in `BTGattAdvertismentParams_t` structure to configure advertisement intervals.

This also fixes #1514 

Checklist:
----------

- [ ] I have tested my changes. No regression in existing tests.
- [ ] My code is Linted.

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.