aws/amazon-freertos #1521: AWS FreeRTOS ESP32 Smartconfig Support

ncohen17, 2019-11-12:
Hello, 

I would like to know if the AWS FreeRTOS on the ESP32 supports the SmartConfig WiFi setup feature ? 

Thank you 
Netanel

lundinc2, 2019-11-12:
Hello @ncohen17,

We currently do not support that feature in our code base.

ncohen17, 2019-11-12:
Is it still possible to connect to the AWS IoT Core using the ESP-IDF and be able to use the Smartconfig feature ?

dnoal, 2019-11-12:
Absolutely yes.  The ESP IDF is integrated with the AWS IoT embedded C SDK (https://github.com/espressif/esp-aws-iot).  You can review this thread on our AWS Forum (https://forums.aws.amazon.com/thread.jspa?threadID=310846&tstart=0) for a review of the differences between Amazon FreeRTOS and the ESP-IDF.

ncohen17, 2019-11-12:
I checked the link you sent me I don't see the differences between using Amazon FreeRTOS and the ESP-IDF.
What are the advantages of using the  Amazon FreeRTOS or the ESP-IDF ?

dnoal, 2019-11-12:
These two distributions have a lot in common.  Both are based up FreeRTOS.  Both facilitate secure authenticated communication with cloud services like AWS IoT.  Both include an OTA update solution.  As you have seen, there are some features supported in the ESP-IDF that are not supported in Amazon FreeRTOS.  Amazon FreeRTOS runs on microcontrollers from many silicon partners while the Espressif IDF is intended for use only with Espressif silicon.  We at AWS continue to work closely with our Espressif partners to close the feature gap so that Espressif customers can have similar experiences on either SW platform.

yuhui-zheng, 2019-11-26:
Since this thread has been in active for last two weeks, I'm closing it for now. 
Tagged with "Feature Request" for future reference.

closed 2019-11-26 by yuhui-zheng

chegewara, 2019-12-26:
> Hello @ncohen17,
> 
> We currently do not support that feature in our code base.

I dont agree with that answer. Smartconfig is supported and its working with AFR for long time, even before release-1.5.
Here is code that is implementing smartconfig:
https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c#L199

I tested it and i am using it.

lundinc2, 2019-12-27:
@chegewara,

You are right, that was a bad answer. I apologize as that was poorly worded. 

A better answer is that we do not host any examples explicitly showcase the Smartconfig WiFi feature, but since this is an open source project, it is always possible to port code to it.

Thank you for pointing this out.