aws/amazon-freertos #1535: How to build a Custom Application!

monkeytronics, 2019-11-19:
The AWS Demos are great in that they demonstrate what can be done. However, in the iot_demo_runner.h it says:

`* Only one demo can be configured at a time`

which I feel is a major limitation. 

**Problem 1:**
We're all trying to create an custom application that includes a selection of the building blocks showcased in the demos. So it would be great if a guide existed which helped us integrate them into our own custom applications. If it is the case that the above stated limitation (only one at a time) would require some careful planning on the part of the designer, guidance on that would be of immense benefit too. 

**Problem 2:** 
In our applications, we don't want to use the code all buried in the demo runner files. It would make more sense to integrate the calls in custom code, working outside of the amazon freertos library. So when we pull the library for updates, we don't obliterate all our code! So basically, how do we add the components one by one to an existing project.

**A Specific Example:** 
1. I have an application that will send data to the cloud over MQTT. I've gotten the MQTT demo to run. How would I replicate this functionality without using the demo code, which is very deeply hard baked into the amazon freertos. I.e. how do we call it from our code. 
2. Next I want to add OTA to the application. Again, we would really benefit from a worked example of how to **add** this functionality to an existing project, and what gotchas to look out for, especially in the context of the other functions I'm using (MQTT & BLE)
3. BLE. I'm pretty sure virtually everyone is going to need to set up their devices with their end users, and BLE is the way we do it with this branch of freertos. Once again, guidance of how to add this to an existing application would be so incredibly useful. 

Thanks all!

lundinc2, 2019-11-19:
Hello @monkeytronics,

We will consider adding this feature. In the meantime you can pull code sections from the various demos or refer to the library APIs and flows found here: https://docs.aws.amazon.com/freertos/latest/userguide/dev-guide-freertos-libraries.html.

Could you expand on the blockers you are experiencing in your example?

monkeytronics, 2019-11-20:
@lundinc2 Hey. In a nutshell, the blocker is that I'm not as good as you at this! Honestly, I'd be happy to integrate the MQTT / OTA / other libraries without having to know too much about what's going on under the hood. A lot of it goes above my head! I imagine that a power user, with a deep familiarity with the AFR project could create a basic template in a day or 2. And subsequently this would save many customers weeks or even months of time getting up to speed with the code. And it would also serve as an incredibly powerful learning tool. 

In terms of specific blockers, it's taken me 2 weeks to get JITP to work, even though I only had to write a few lines of code and run a few scripts. I spent most of that time confused, dredging forums and it's only the kindness of a more experienced user that I got it running! 

But great idea. Let's take a concrete example : OTA. I'm starting from a main.c outside of the AFR library: see [threaad 832]( https://github.com/aws/amazon-freertos/issues/832) and refering to the example in [integrate-ota-agent.html](https://docs.aws.amazon.com/freertos/latest/userguide/integrate-ota-agent.html). 

I've created a new folder as per [threaad 832]( https://github.com/aws/amazon-freertos/issues/832) and created the files 
components ->
   ota ->
      src ->
         ota.c
      include ->

And copied the ota demo code into ota.c. I added  to the top CMake.list : 

```
get_filename_component( EXTRA_COMPONENT_DIRS "components/ota" ABSOLUTE )
list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})
```
Then i'm basically starting a task in the main.c:

```
/* OTA Custom Lib */
#include <ota.h>
void runOta(void *pvParameter);
...

 xTaskCreate(&runOta, "ota", 8 * 2048, NULL, tskIDLE_PRIORITY + 6, NULL );
...

void runOta(void *pvParameter)
{
    vStartOTAUpdateDemoTask();
}
```
(N.B I removed the parameters from vStartOTAUpdateDemoTask( ) since they aren't used)
I used this approach to put in my custom application. that works fine. But what I've done is clearly not right. All the includes in the ota demo code won't work. Is there a better way to do this?

I get errors like: 

```
In file included from ../amazon-freertos-configs/iot_config.h:76:0,
                 from ../components/ota/src/ota.c:38:
../amazon-freertos-configs/iot_config_common.h:35:50: fatal error: platform/iot_platform_types_freertos.h: No such file or directory
 
../components/ota/src/ota.c:41:22: fatal error: iot_mqtt.h: No such file or directory

../src/main.c:232:5: error: implicit declaration of function 'vStartOTAUpdateDemoTask' 
```

The following line seems to work in the main.c but does not in the code copied from ota. Honestly, CMake is confusing me so much! 

`#include "iot_config.h"`

So I guess my basic first question is how do I configure and call OTA from within my main function. Note that I'm still running the MQTT demo magically in the background. Although it's not clear at all to me how to migrate this code up into my main either. 

I'm very, very confused by this.

yuhui-zheng, 2019-11-25:
Hi @monkeytronics,

It seem @tgsong has been responding in #832 thread in regards to CMake and OTA related. Commenting a little bit more on OTA/MQTT related. 

OTA service utilizes IoT Job service and publishes to Job MQTT topics. On the client side, OTA agent subscribes [these topics](https://github.com/aws/amazon-freertos/blob/master/libraries/freertos_plus/aws/ota/src/aws_iot_ota_agent.c#L230-L235) with specified MQTT handle. [OTA demo ](https://github.com/aws/amazon-freertos/blob/master/demos/ota/aws_iot_ota_update_demo.c#L239-L254)connects MQTT, does OTA routine, and disconnects MQTT. 

When Integrating OTA with other application using MQTT, user could make the design choice for having either one MQTT handle (cannot share topics across apps if to run together, MQTT layer is not aware the context), or having two separate handles (each app can manage its own connection, but needs to watch out for memory consumption). With TLS, each MQTT connection can be pretty big (up to ~70K), that some device may not fit more than one MQTT connection.

abhidixi11, 2019-12-02:
Hi @monkeytronics,

Does the above response answer your question?
Please let us know.
Thanks.

monkeytronics, 2019-12-02:
@abhidixi11 @yuhui-zheng Thanks for the responses. @tgsong did offer excellent advise. Unfortunately, in all honesty, even with that, I don't think it's really feasible for me to try and join all this together to make an application. The OTA documentation is out of date, and refers to code that is no longer used. The approach to get it working is to modify the demo application code, rather than instantiate the code cleanly outside the library. I'm sure expert users could get it working, but I just don't have the time to fiddle with something this complex in the hope that it will eventually work, and with the risk that it won't necessarily continue to work. 

The purpose of this post was really to ask - Is there a simple starting point available for developers who are not experts or contributors to the amazon-freertos project that will help them get going with the bare basics. This would look like a hello world including OTA, MQTT, something like the unified provisioning setup used by ESP-IDF, and with adequate documentation to use the BLE portion of it (since the STA approach isn't used). Unfortunately at present the answer is no. No starting point like that is provided. Which is too much of a technical and business hurdle for me to continue with this as a solution at present. 

But I appreciate your generous explanations and will keep my eye on afr to see when it does become more commercially ready to use.

abhidixi11, 2019-12-03:
Hello  monkeytronics ,
>>>The purpose of this post was really to ask - Is there a simple starting point available for developers who are not experts or contributors to the amazon-freertos project that will help them get going with the bare basics.

Thanks for getting back. Sorry to see that our current documentation and demo examples are not serving as a good starting point for you.  Our demos are designed to demonstrate use of a particular library.  Unfortunately at this time we don't have an example of a complete application. 

>>>The OTA documentation is out of date, and refers to code that is no longer used.

Can you please point us to OTA documentation that is out of date?  If you use  'release' branch the documentation should match the code. If you are using master, it may be different because we are updating the master branch constantly.
Thanks.

abhidixi11, 2019-12-04:
Hello  monkeytronics,
I'm closing this issue, please feel free to re-open if you have any  more questions.
Thanks!

closed 2019-12-04 by abhidixi11

msn444, 2020-01-04:
Couldn't agree more with monkeytronics' points. I've gone through exactly his frustration in my own attempts to pull together OTA, MQTT and BLE functionality into one app using the provided demos. It would be so much better if a single demo app existed outside of the AFR source tree that provided OTA, MQTT and BLE.

sameerdasarwad, 2020-01-06:
@msn444  can please put more info about your single demo app if possible

msn444, 2020-01-24:
@sameerdasarwad Generally it would be great if the demo provided basic functionality that is required in most applications. I'm envisioning a generic app that:

- Provides a generic interface for the developer to provide certificates and provisioning data from whatever production workflow is required. Currently the demos require the certs to be hard-coded in a header file buried in a few layers of abstraction. Ideally there'd be an interface at the application level to feed the certs.
- Similarly, provides a generic interface for providing WiFI credentials at the application level, along with the option to use the AFR BLE-based credentials scheme.
- Provides full WiFI/network connection/disconnection/reconnection handling.
- Provides an interface for publishing and subscribing to arbitrary MQTT topics, with topic callbacks demonstrated at the app level.
- Has OTA fully functional in a background thread.
- Has a functioning NimBLE stack, both for IP traffic and generic GATT connections. Provide a service and a couple of characteristics for demo purposes.

I can't think of many applications that wouldn't make use of the above feature set.

hassin23ayz, 2020-10-13:
@monkeytronics thanks for this, exactly my situation right now , were you able to solve ? found any base / template project ?