aws/amazon-freertos #1536: Don't manage to get the ESP32 projects compiled

htibosch, 2019-11-19:
**Describe the bug**
I don't manage to get the ESP32 projects compiled, even after a long effort.

**System information**
- Which hardware board or part numbers?
  Espressif Systems ESP-WROVER-KIT

- IDE used
  Eclipse for Espressif

- Operating System
  Windows 10

- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
  Today's version from GIT

- Project
  Both demo and testing projects for Espressif

**Expected behavior**
Install, compile, configure and run!

I am trying to get the ESP32 demo and test projects running.

My first question: do you recommend using Windows 10 or a Linux-based platform?

I am now working on W10. I set-up the tools by running `esp-idf-tools-setup-2.1.exe`. I am able to change the source files, compile them and flash the CPU. So far the software from Espressif.

Now the AWS/FreeRTOS part. I installed Eclipse and then the real trouble starts.

The aws_demos Makefile states:

~~~c
export AMAZON_FREERTOS_PATH := $(CURDIR)/../../../../..
ifndef IDF_PATH
export IDF_PATH := $(AMAZON_FREERTOS_PATH)/vendors/espressif/esp-idf
else
Don't mess with IDF_PATH
endif
~~~

My `CURDIR` points to "E:/Home/amazon-freertos/vendors/espressif/boards/esp32/aws_demos"

The "E:" gives a fatal error: 

~~~c
IDF_PATH cannot contain colons. If overriding IDF_PATH on Windows, use MSYS Unix-style /c/dir instead of C:/dir.  Stop.
~~~

The Make macro `$(CURDIR)` on Windows tends to start with a drive letter and a colon. I think I can not change that?

The Makefile does not allow a user-defined `IDF_PATH`:

~~~c
IDF_PATH is defined in your environement, it will not point to vendors/espressif/esp-idf.
To remove the variable run the command "unset IDF_PATH"
~~~

Any solution?

Third and last question:

At some other point, Make complained about ncurses not being installed. It also needed a native GCC compiler (which I had installed). I think it wants to build some kind of `KCONFIG`.

I just want to do some maintenance work on the FreeRTOS+TCP library. Isn't there a short-cut to get Eclipse compile and debug?

Thanks,
Hein Tibosch

htibosch, 2019-11-25:
Hi, could anyone experienced with the ESP32 please give a tiny hint?
Thanks, @htibosch

yuhui-zheng, 2019-11-25:
Hi @htibosch 

To your first question:

> My first question: do you recommend using Windows 10 or a Linux-based platform?

Both should work. 

And in your description as well as log, it seems you are using a version of ESP-IDF tool instead of using the esp-idf in aws/amazon-freertos. Just checking, were you following ESP getting started guide? 

Amazon FreeRTOS ESP getting started guide: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

esp-idf in aws/amazon-freertos: https://github.com/aws/amazon-freertos/tree/master/vendors/espressif/esp-idf

htibosch, 2019-11-26:
Thank you, yuhui-zheng!

I have a few questions left:

Which setup should I use, 1.1 or 2.0 ?

	esp-idf-tools-setup-1.1.exe
	esp-idf-tools-setup-2.0.exe

Note that they are very different. I used the newest one.

Should I use mingw32 or mingw64, which is compatible with Python 2.7.14, which is 64-bits?

What do I need MINGW for?

> Version 3.1.5 of the ESP-IDF (the version used by Amazon FreeRTOS) does not
> support the latest version of the ESP32 compiler. You must use the compiler
> that is compatible with version 3.1.5 of the ESP-IDF (see the links above).
> To check the version of your compiler, run "xtensa-esp32-elf-gcc --version". 

The above text does not say which compiler version can be used? I now have 5.2.0.

I followed Sarena's advice (https://github.com/aws/amazon-freertos/issues/903#issuecomment-508846496)

	cd <root of amazon-freertos repo>
	cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -GNinja -S . -B build
	cd build
	ninja
	cd ..
	// I can even flash the part
	python vendors/espressif/esp-idf/tools/idf.py  erase_flash -B build -p COM7

Where can I choose between aws_demos and aws_tests ?
It tells to run "aws configure", but where is the command "aws"?

yuhui-zheng, 2019-11-26:
I believe ```esp-idf-tools-setup-1.1.exe``` && ```mingw32``` is the right combination for us. (From that Getting Started Guide I attached. For everything we put up there, it has been tested.) Though, I have a teammate telling me ``esp-idf-tools-setup-2.1.exe``` also worked for him. (https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html) 

I'm attaching build log from a colleague, in which you'll find compiler version and etc. 
[esp_windows.txt](https://github.com/aws/amazon-freertos/files/3893767/esp_windows.txt)

As for switching between test and demo, could try adding this arg in CMake command line:
```-DAFR_ENABLE_TESTS=1```
May need to delete the build directory between different targets though. 

```aws configure``` is the command to configure AWS CLI. (Configure AWS CLI with usernames, credentials and regions etc.) With AWS CLI, user could programmatically access AWS services. Need to install AWS CLI though. 

AWS Command Line Interface: https://docs.aws.amazon.com/cli/latest/userguide/cli-chap-welcome.html

htibosch, 2019-11-27:
At @yuhui-zheng : thanks a lot for your patience and hints, I've got it all running now.

I have used "esp-idf-tools-setup-2.1.exe", and I did not install MSYS or MINGW ( I'm allergic ).

To other Windows users : make sure PATH contains the locations of:

~~~
    git.exe
    xtensa-esp32-elf-cc.exe
    ninja.exe
    mconf-idf.exe
~~~

And make sure that the source dir is correct in `tools\aws_config_quick_start\configure.json`, e.g. :

~~~
	"afr_source_dir":"E:/home/amazon-freertos"
~~~

Finally the command to flash and run that worked for me:
~~~
python vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p COM7 -B build
~~~

Thanks,
Hein

htibosch, 2019-11-27:
Hurray, I managed to changes the source code and see that it works!

One (probably) last question:

I added a temporary source file to the project:

~~~
libraries/freertos_plus/standard/freertos_plus_tcp/source/iperf3.c
~~~

and then I recreate the "build" directory, and run Nija. But why doesn't it get picked-up?

Now I make changes to the component:

~~~
vendors/espressif/boards/esp32/components/freertos_tcpip/freertos_plus_tcp/component.mk
~~~

to add the new source file **explicitly**, but the file seems to be ignored.

~~~
COMPONENT_OBJS += $(AMAZON_FREERTOS_AFR_PLUS_DIR)/standard/freertos_plus_tcp/source/iperf3.o
~~~

Is there another action necessary after adding sources or after changing "component.mk" ?
Thanks, again

yuhui-zheng, 2019-11-27:
Nice! 

Could take a look at ```CMakeLists.txt``` under ```libraries/freertos_plus/standard/freertos_plus_tcp/```. -- https://github.com/aws/amazon-freertos/blob/master/libraries/freertos_plus/standard/freertos_plus_tcp/CMakeLists.txt.

This file defines what sources && headers are included in FreeRTOS+TCP component.

htibosch, 2019-11-29:
Thanks Yuhui-Zheng, that works!
I know enough by now to be able to do my job: checking NetworkInterface.c and +TCP.
One last remark: the presence of both CMakeLists and *.mk/Makefile was confusing for a first-time user like me.
Is GNU make also being used to build ESP32?
Hein

closed 2019-11-29 by htibosch