aws/amazon-freertos #1543: Mediatek MT7697HD (MT41) - Failed to provision device certificate 49

symdeb, 2019-11-22:
1. FreeRTOS and credentials from AWS console download
2. Copied aws_client_credentials.h and  aws_client_credential_keys.h to demos/include
3. Update SSID and password in the file
4. Compile with Keil MDK 5 in Windows 10
5. Connect MT41 board and  Flash 
6. Reset board, serial output is:

fota partition base: 0x266000, length: 0x18a000, offset from end: 0x200
magic ver: -1
fota: TMP is empty, skip upgrade
jump to (0x10079000)
0 0 [wrap] Starting up
1 11 [wrap] Write certificate...
2 11 [wrap] ERROR: Failed to provision device certificate. 49
3 559 [wrap] WiFi module initialized. Connecting to AP...
4 560 [wpa_supplicant] ========= Supplicant Ready =======5 5553 [wrap] WiFi Connected to AP. Creating tasks which use network...
6 5554 [iot_thread] [INFO ][DEMO][5554] ---------STARTING DEMO---------

7 5555 [iot_thread] [INFO ][INIT][5555] SDK successfully initialized.
8 5565 [iot_thread] radio already on/off, it is not necessary do radio on/off again.9 5598 [inband] DISCONNECTED MAC - b0:6e:bf:9a:3a:6c10 8545 [iot_thread] [INFO ][DEMO][8545] Successfully initialized the demo. Network type for the demo: 1
11 8546 [iot_thread] [INFO ][MQTT][8545] MQTT library successfully initialized.
12 8546 [iot_thread] [INFO ][DEMO][8546] MQTT demo client identifier is MT41-T2 (length 7).
13 8780 [iot_thread] **ERROR: Private key not found. 14 8780 [iot_thread] TLS_Connect fail (0x7d4, a2n55qlxple7ub-ats.iot.us-east-1.amazonaws.com)**
15 8781 [iot_thread] [ERROR][NET][8781] Failed to establish new connection.
16 9081 [iot_thread] [ERROR][MQTT][9081] Failed to establish new MQTT connection, error NETWORK ERROR.
17 9082 [iot_thread] [ERROR][DEMO][9082] MQTT CONNECT returned error NETWORK ERROR.
18 9082 [iot_thread] [INFO ][MQTT][9082] MQTT library cleanup done.
19 9082 [iot_thread] [ERROR][DEMO][9082] Error running demo.
20 9083 [iot_thread] [INFO ][INIT][9083] SDK cleanup done.
21 9083 [iot_thread] [INFO ][DEMO][9083] -------DEMO FINISHED-------

yuhui-zheng, 2019-11-23:
Hi symdeb,

In your step 2,
> Copied aws_client_credentials.h and aws_client_credential_keys.h to demos/include

Could you double check the file names under ```/demos/include```, in the version of the code you are using? 

In our master branch, we currently have ```aws_clientcredential.h``` and ```aws_clientcredential_keys.h```. https://github.com/aws/amazon-freertos/tree/master/demos/include 

(If this indeed fix your issue, please kindly let us know which tutorial you've followed. We'd like to fix the inconsistency in docs.)

symdeb, 2019-11-23:
Correct,  those are the filenames. credentials.h with WIFI SSID and PSW and key had a private key and cert. Code and credentials from  
https://console.aws.amazon.com/iot/home?p=ifr&cp=bn&ad=c&region=us-east-1#/software/freertos Connect to AWS IoT - MT7697Hx-Dev-Kit 
Error seems to happen in the last step of xProvisionCertificate (aws_dev_mode_key_provisoning.c

yuhui-zheng, 2019-11-25:
> Correct, those are the filenames. credentials.h updated with WIFI SSID and PSW
The key.c contained the private key and cert. as from the AWS console

I meant the file names you were using do not seem to be correct. 
- expected file with name string```aws_clientcredential.h``` instead of ```aws_client_credentials.h```
- expected  file with name string ```aws_clientcredential_keys.h``` instead of ```aws_client_credential_keys.h``` 
I verified "Connect to AWS IoT - MT7697Hx-Dev-Kit" from AWS IoT console. And under ```./demos/include``` we have ```aws_clientcredential.h``` and ```aws_clientcredential_keys.h```. Could you update your file name, and make a clean build? MQTT demo should work out of box.

symdeb, 2019-11-26:
The filenames in the first post appear to be a typo. They are as what is expected; `aws_clientcredential.h and aws_clientcredential_keys.The failed provisioning problem remains.
The Keil temporary license for 7 days will expires in 2 days.  After that it seems I will need to use deeper pockets of  US $1500 to $3000 for a license to continue with this.

yuhui-zheng, 2019-11-26:
From the steps you've mentioned above, other than file names nothing jumped out to me as red flag. The symptom from the log, I'd double check ```keyCLIENT_CERTIFICATE_PEM``` and ```keyCLIENT_PRIVATE_KEY_PEM```. And a clean build make help.

I've verified in our test infrastructure, that MQTT demo at the HEAD of master branch of this repo worked on MediaTek a couple of hours ago. Would cloning this repo (instead of downloading from IoT console) an option for you? (I'm attaching build/runtime log, in case you need to check a couple of details.)

[MediaTek_MQTT_demo_ 01ff99b579adb040e0d0b03a4428567fd91de2c4.txt](https://github.com/aws/amazon-freertos/files/3893390/MediaTek_MQTT_demo_.01ff99b579adb040e0d0b03a4428567fd91de2c4.txt)
)

symdeb, 2019-11-27:
Will check but these files were taken without modification from AWS site and do contain the certificate and private key. Posted a related issue with this board [here](https://github.com/USIWP1Module/USI-MT7697Hx-Development-Kit/issues/1)
To port the code to GCC (Linux) should be possible as the 7697 supports GCC., this may remove the problem to be forced to use Keil.

symdeb, 2019-11-27:
Your flash log appears very different than what is shown in my uVision. it shows references  STM32L etc..Do you use the MDK and board's USB interface to flash?
Secondly, the instructions
[here ](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_mediatek.html) omit to mention to replace the .h files with the keys and certificates and Wifi settings.
Attached my observations following the steps cloning the github repo and using master branch. (folder (**AmazonFreertos**). Loaded into Keil, and cleaned.
Checked that the debug and flash setting are exactly as shown on that page.

A1. Build output
A2. Flash fail error 
[logs.zip](https://github.com/aws/amazon-freertos/files/3897129/logs.zip)

A3. 2nd attempt Flash & Verify OK (see log) 
A4. Boot fails, (data dump)
A5. Reconnect USB, Boot, stuck after this point

After this went 
[here](https://console.aws.amazon.com/iot/home?region=us-east-1#/software/freertos)
created a new thing and downloaded the file (folder **AmazonFreeRTOS**) then did the Keil thing again. (after a clean).   Same results as A.

Right now LED D2 is flickering and COM port does not show up. Seems this board can now go where where is is supposed to be: the trash can.

symdeb, 2019-11-28:
Could AWS consider to support the standard Linkit 7697 board. MT-41 concerns are

Killer issues:
-  No COM port after testing, unrecoverable.
- Can only use the (expensive) tool chain form Keil. No GCC support.
- Amazon FreeRTOS SDK code not working (as above)

Others:
- No hardware diagram available.
- Does not work with Link-it  MT7697 SDK sample codes
- Close to zero support from USI and Mediatek. This board is based on ACSIP outsourced by Mediatek to USI. Both for which there are no discussion forums or they are  down. 
- Build under Linux not possible (Keil MDK is Windows only)
- JTAG/SWD connector must soldered (not soldered)
- Expensive board compared to Link-it 7697
- Apperrantly few other developer use it.

Made an attempt to add Mediatek as an alternative to other chipsets but it did not work out.
Removed this board as a  candidate to build a solution.

abhidixi11, 2019-11-28:
Hello Symdeb,
Thanks for reporting the problems you are facing with Mediatek board. Sorry to see that you had limited time and could not get the board working.

I would like to answer some of your earlier question.
>>>First, your flash log appears very different than what is shown in my uVision.

The log attached was from our automation workflow, therefore it  looked different, although we are using the board as described in getting started guide, the output is similar to what you had posted. I tested the board with latest code and the snippet of the output is attached below.
[Mediatek_demo.txt](https://github.com/aws/amazon-freertos/files/3900431/Mediatek_demo.txt)

I'm assuming you were putting the board in PROGRAM mode before attempting to flash the board.

>>>Secondly, the instructions here omit to mention to replace the .h files with the keys and >>>certificates and Wifi settings.

The instructions for updating the keys, certificate, and Wi-Fi settings can be found on the "First Steps" page of our website. These are common instructions for all the supported boards.
https://docs.aws.amazon.com/freertos/latest/userguide/freertos-configure.html

>>>Could AWS consider to support the standard Linkit 7697 board.

Thanks for the request and detailed reasoning for it, I will create an internal issue to look into you request.
Please let us know if you have any further questions.

symdeb, 2019-11-28:
I flash the board with the Keil MDK. equested an RMA repair or replacement of the board. Even with a replacement I would still be out of luck to proceed with this due to the Keil license. Would still suggest on the MT7697 page to mention that to be able to connect to AWS to replace the key/certificate and WIFI credential files and only then build the project. Thanks for the ticket. Cheers.

abhidixi11, 2019-11-28:
Hi symdeb,
Unfortunately you got the bad board, _PROGRAM=> RESET => RELEASE RESET => and hen Flash from the menu_ is the correct sequence. 
Thanks for your input about getting started guide. The guide for MT7697 has link to 'first steps' page (second paragraph), which has all the information needed to setup keys. We will try to see if we can improve the wording to make it more visible, so that those instructions are not missed.
I'm closing this issue, please feel free to re-open if you have any further question.
Thanks!

closed 2019-11-28 by abhidixi11