aws/amazon-freertos #1555: Can not link the Xilinx demo project

htibosch, 2019-11-26:
**Describe the bug**
When compiling the Xilinx aws_demo project, linking fails, because of the command length.

**System information**
- Which hardware board or part numbers?
  Xilinx Zynq

- IDE used
  Xilinx Software Development Kit, release 2019.1

- Operating System
  Windows 10

- Version of Amazon FreeRTOS
  Todays version

- Project
  aws_demo

**To reproduce**
Steps to reproduce the behavior:

1.Import the aws_demo project into the work space
2. Press on Build All

**Additional context**
The problem is caused by the fact that cmd.exe can only accept a parameter length of 8192 bytes.
There seems to be [a solution](https://mcuoneclipse.com/2015/03/29/solving-the-8192-character-command-line-limit-on-windows), but I didn't get that working with Xilinx SDK. Mentioned Make.exe is too old.

The link command is 9891 bytes long :
~~~
    arm-none-eabi-gcc -mcpu=cortex-a9...
~~~

Thanks, Hein

abhidixi11, 2019-11-30:
Hello Hein,
Thanks for reaching out, apologies for delay in responding. 
I'm trying to understand if there is any difference between your setup and how build on our internal nightly test server works. I don't have Xilinx setup with me, so I'm attaching a log file for the recent nightly build, I'll try to get more detail logs on Monday if you are still stuck by then.
I'm assuming you are following getting started guide:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_xilinx.html
Here is build log:
[xlinxLogs.txt](https://github.com/aws/amazon-freertos/files/3906779/xlinxLogs.txt)

htibosch, 2019-11-30:
Hello, and thanks for the links.
Now I followed the "getting started" guidelines, but this build also ends in a command that is 11,708 bytes long, too long for Windows' cmd.exe.
In a meanwhile I'm installing XSDK 2019.1 on Ubuntu, so I can continue my work.
I have a pending PR for the NetworkInterface for Xilinx, and want to test it in the demo and test applications.
But still I'd like to get it built on Windows as well.

abhidixi11, 2019-12-02:
Thanks for update, I'll check  tomorrow if anyone here has seen character limit issue with XSDK. Meanwhile, I'm not sure if you have seen this thread from Xilinx forum, which talks about the same issue, may be you can create static libraries of .o files as mentioned there. 
https://forums.xilinx.com/t5/Processor-System-Design/SDK-Is-there-a-fix-for-linker-command-line-over-8192-characters/td-p/716361
Thanks,
Abhijit

htibosch, 2019-12-02:
Thanks Abhijit. I installed and configured the Xilinx project on Ubuntu, and indeed the problem does not occur. The length of the link command doesn't matter.

I had seen the solutions given on the Xilinx forum.
In my own projects, I have also encountered the parameter length limit of 8 KB. In a Makefile project, it is easy to solve:

- Create libraries and link with those libraries
- Pass the parameters to GCC in a file: `gcc @parameters.txt`

Building the Xilinx project is guided by Eclipse. We could add a second target that only creates a library.
I don't know how the second method can be implemented in Eclipse.

For now I will continue on Ubuntu.

abhidixi11, 2019-12-03:
Hey Hein,

Looks like we did face the same problem with latest the SDK on windows. We reverted to using earlier SDK in our build system. So for Windows, you can either use older SDK or do one of the things mentioned above to fix it.
I'm assuming currently you are unblocked, if so can we close this issue? I'll investigate to see if our documentation should be updated to mention this problem.
Thanks,
Abhijit

htibosch, 2019-12-04:
Hello Abhijit, indeed I'm not blocked by this issue.
What do you mean with an "earlier SDK", earlier than Xilinx SDK 2019.1? Could that make a difference?
For the longer term, I would recommend creating one or more libraries. That solution should work well in Eclipse. I don't mind making that change.

closed 2019-12-04 by htibosch

abhidixi11, 2019-12-04:
Hey Hein,
>>>"earlier SDK", earlier than Xilinx SDK 2019.1? 
I confirmed that  we are using the same version as above (SDK 2019.1). We are creating a new setup here to check if we see the same problem as what you are seeing. I'll post update here and check if we are missing any steps in guide. 
Thanks,
Abhijit

abhidixi11, 2019-12-05:
Hello Hein, 
FYI- Looks like we can build with SDK 2019.1 .  We tried with two different setups and didn't notice the problem that you are seeing. 
Thanks.

htibosch, 2019-12-06:
Thanks for reporting this.
Maybe there is a difference in tools that we use?
This is what I have:

cmd.exe --version:
Microsoft Windows [Version 10.0.18362.476]

make.exe --version:
GNU Make 4.1

rm.exe --version:
rm (GNU coreutils) 5.3.0

sh.exe --version:
GNU bash, version 4.4.23(1)-release (x86_64-pc-msys)

It seems to be that Windows has a parameter buffer of 32 KB, but cmd.exe seems to be limited to 8 KB. When I link the project, the parameters are cut off after 8192 bytes.

Thanks, Hein

htibosch, 2019-12-07:
All right, I went a long way but **I solved it**!

I downloaded the source code of Make 4.2.1 and ran the program in debug mode. I saw that sh.exe was called as:

	sh.exe -c "arm-none-eabi-gcc -Wall -O0 -g3 -IE:/Ho..."

The 11 KB parameters for linking got truncated by sh.exe to 8 KB. That caused my problem.

Now when I tell Make to use cmd.exe, there is no problem: Make will call the compiler directly, without cmd.exe.

Solution: tell Ecplise which make.exe to use and indicate your preferred `SHELL`:

- Right-click on the project
- Click on "C/C++ Build"
- Clear the option "Use default build command"
- At "Build command", fill in e.g. "`C:\PROGRA~2\GNU\bin\make.exe SHELL=C:\Windows\System32\cmd.exe`"

In fact it tells Make **not** use cmd.exe, which solves the parameter problem.

Hein