aws/amazon-freertos #1563: HTTP server (web server) support

rajneeshsinha, 2019-11-27:
Hello AWS team

I am working on ESP32 board with amazon freeRTOS. I have an requirement to use web server (HTTP server) to sending message on local web server and vice versa. I know Amazon freeRTOS still doesn't added support for HTTP server.

My questions are following:

1. Can I port or integrate the ESP-WHO or ESP-IDF HTTP_server code (available in components folder) into amazon freeRTOS platform?
2. If yes then can you please provide some guidance how to do that and what other dependencies comes out?
3. Amazon freeRTOS planning to add HTTP server support in near future?

Any quick help will be appreciated.

Thank you.

abhidixi11, 2019-11-29:
Hello rajneeshsinha,
I think you asked same question on the AWS forum (https://forums.aws.amazon.com/thread.jspa?threadID=313477).  I'm going to close this issue. If you have any further questions, feel free to reopen.

closed 2019-11-29 by abhidixi11

mahavirj, 2019-11-29:
@rajneeshsinha Please take a look at https://github.com/aws/amazon-freertos/pull/1571