aws/amazon-freertos #1566: The ESP32 port of PKCS#11 should use a configurable NVS partition

matteoscordino, 2019-11-27:
**Is your feature request related to a problem? Please describe.**  
A common manufacturing pattern has the per-device credentials flashed as a separate read-only partition. This pattern allows the per-device credentials to be decoupled from user data, and simplifies the implementation of "factory reset" functionality (the user partition can just be reinitialised, leaving credentials intact)

Currently this requires patching AFR manually on ESP32, because the ESP32 PKCS#11 PAL has the NVS partition and namespace hardcoded.

**Describe the solution you would like.**  
The NVS partition and namespace used by the PAL should be a configuration define.

**Additional context**  
I have a pull request available that fixes this, it's #1554 

Thank you!

abhidixi11, 2019-11-30:
Thanks for reaching out.  We will review your PR internally and let you know. Meanwhile, if you have any questions about this issue, feel free to reach out. 
Thanks.

dan4thewin, 2019-12-06:
I approved the PR.  It needs one more approver.  I expect we'll have this merged early next week.

closed 2019-12-06 by dan4thewin