aws/amazon-freertos #1569: ESP32 Nimble: prvBTSetAdvData assumes exactly 1 UUID to be advertised

matteoscordino, 2019-11-28:
**Describe the bug**
In `iot_ble_hal_gap.c`, function `prvBTSetAdvData()` never uses the `xNbServices` parameter and assumes there is exactly 1 UUID to be advertised if `pxServiceUuid` is not NULL.
At the same time, `_setAdvData` always gives a value to `pxServiceUuid` and sets `xNbServices` as 0 if there are no UUIDs to advertise.
This means that trying to advertise 0 UUIDs (e.g. to save advertisement space) does not work.

**System information**
- Which hardware board or part numbers? ESP32
- IDE used. Does not matter.
- Operating System. Does not matter.
- Version of Amazon FreeRTOS. master
- Project: Custom Application

**Expected behavior**
The xNBServices should be used to advertise the requested number of UUIDs.

**To reproduce**
Steps to reproduce the behavior:

Just use a custom advertising callback like the following:
```void IotBle_SetCustomAdvCb(IotBleAdvertisementParams_t *pAdvParams,
                           IotBleAdvertisementParams_t *pScanParams)
{
  pAdvParams->setScanRsp          = false;
  pAdvParams->includeTxPower      = false;
  pAdvParams->name.xType          = BTGattAdvNameShort;
  pAdvParams->name.ucShortNameLen = IOT_BLE_DEVICE_SHORT_LOCAL_NAME_SIZE;
  pAdvParams->appearance          = IOT_BLE_ADVERTISING_APPEARANCE;
  pAdvParams->minInterval         = IOT_BLE_ADVERTISING_CONN_INTERVAL_MIN;
  pAdvParams->maxInterval         = IOT_BLE_ADVERTISING_CONN_INTERVAL_MAX;
  pAdvParams->serviceDataLen      = 0;
  pAdvParams->pServiceData        = NULL;
  pAdvParams->manufacturerLen     = 0;
  pAdvParams->pManufacturerData   = NULL;
  pAdvParams->pUUID1 = (BTUuid_t *)&_advUUID;
  pAdvParams->pUUID2 = NULL;

  /// use the scan response to fit the full name of the device
  pScanParams->setScanRsp          = true;
  pScanParams->includeTxPower      = false;
  pScanParams->name.xType          = BTGattAdvNameComplete;
  pScanParams->name.ucShortNameLen = 0; // does not matter
  pScanParams->appearance          = IOT_BLE_ADVERTISING_APPEARANCE;
  pScanParams->minInterval         = IOT_BLE_ADVERTISING_CONN_INTERVAL_MIN;
  pScanParams->maxInterval         = IOT_BLE_ADVERTISING_CONN_INTERVAL_MAX;
  pScanParams->serviceDataLen      = 0;
  pScanParams->pServiceData        = NULL;
  pScanParams->manufacturerLen     = 0;
  pScanParams->pManufacturerData   = NULL;
  pScanParams->pUUID1 = NULL;
  pScanParams->pUUID2 = NULL;
}```

**Code to reproduce the bug**
```void IotBle_SetCustomAdvCb(IotBleAdvertisementParams_t *pAdvParams,
                           IotBleAdvertisementParams_t *pScanParams)
{
  pAdvParams->setScanRsp          = false;
  pAdvParams->includeTxPower      = false;
  pAdvParams->name.xType          = BTGattAdvNameShort;
  pAdvParams->name.ucShortNameLen = IOT_BLE_DEVICE_SHORT_LOCAL_NAME_SIZE;
  pAdvParams->appearance          = IOT_BLE_ADVERTISING_APPEARANCE;
  pAdvParams->minInterval         = IOT_BLE_ADVERTISING_CONN_INTERVAL_MIN;
  pAdvParams->maxInterval         = IOT_BLE_ADVERTISING_CONN_INTERVAL_MAX;
  pAdvParams->serviceDataLen      = 0;
  pAdvParams->pServiceData        = NULL;
  pAdvParams->manufacturerLen     = 0;
  pAdvParams->pManufacturerData   = NULL;
  pAdvParams->pUUID1 = (BTUuid_t *)&_advUUID;
  pAdvParams->pUUID2 = NULL;

  /// use the scan response to fit the full name of the device
  pScanParams->setScanRsp          = true;
  pScanParams->includeTxPower      = false;
  pScanParams->name.xType          = BTGattAdvNameComplete;
  pScanParams->name.ucShortNameLen = 0; // does not matter
  pScanParams->appearance          = IOT_BLE_ADVERTISING_APPEARANCE;
  pScanParams->minInterval         = IOT_BLE_ADVERTISING_CONN_INTERVAL_MIN;
  pScanParams->maxInterval         = IOT_BLE_ADVERTISING_CONN_INTERVAL_MAX;
  pScanParams->serviceDataLen      = 0;
  pScanParams->pServiceData        = NULL;
  pScanParams->manufacturerLen     = 0;
  pScanParams->pManufacturerData   = NULL;
  pScanParams->pUUID1 = NULL;
  pScanParams->pUUID2 = NULL;
}```

**Additional context**
I have a patch that I will post soon as a PR

abhidixi11, 2019-11-30:
Hello  matteoscordino,

Thanks for contacting us and providing a fix for potential issue.   We will internally review it and get back to you.  
Thanks!

ravibhagavandas, 2019-12-03:
Hi @matteoscordino 

Thanks for bringing this up.

I agree there is a deviation here in the port implementation, but the bluetooth core specification ([core supplement v8](https://www.bluetooth.com/specifications/bluetooth-core-specification/))  page 10 says:

_"packet or data block shall not contain more than one instance for each Service UUID data size. If a device has no Service UUIDs of a certain size, 16-, 32-, or 128-bit, the corresponding field in the extended inquiry response or advertising data packet shall be marked as complete with no Service UUIDs. An omitted Service UUID data type shall be interpreted as an empty incomplete-list."_ 

So I think, the packet can have one service UUID each of different types (16, 32 or 128 ) but if there are more than one UUID of a particular type then it should be marked as incomplete type and only the first UUID should be included.

Could you change the pull request accordingly ?

dan4thewin, 2019-12-08:
Hi, @matteoscordino .  On the related PR, https://github.com/aws/amazon-freertos/pull/1570 , it looks like one more change was suggested.  Would you have a look?  Thanks.

PR #1570 (2019-12-14): fix for ESP32 Nimble: prvBTSetAdvData assumes exactly 1 UUID to be ad…
fix for ESP32 Nimble: prvBTSetAdvData assumes exactly 1 UUID to be advertised

Description
-----------
fixes #1569 :
In iot_ble_hal_gap.c, function prvBTSetAdvData() never uses the xNbServices parameter and assumes there is exactly 1 UUID to be advertised if pxServiceUuid is not NULL.
At the same time, _setAdvData always gives a value to pxServiceUuid and sets xNbServices as 0 if there are no UUIDs to advertise.
This means that trying to advertise 0 UUIDs (e.g. to save advertisement space) does not work. Trying to advertise [... cut]
closed 2019-12-14 by gkwicker