aws/amazon-freertos #1573: FreeRTOS_sendto

matteoscordino, 2019-12-02:
**Describe the bug**
Not sure this is a bug, a limitation I'm unaware of or just me being clumsy, but it seems if I create a UDP socket and send data with the standard FreeRTOS socket library (`FreeRTOS_sendto` etc) from the "demo" thread (i.e. the one created by DEMO_Runner), the function returns successfully, but the data is never sent.
I have tried sending to a host on the same local network (1 hop), no DNS resolution and sniffing the network: the data just didn't leave the interface.

**System information**
- Which hardware board or part numbers? ESP32
- IDE used. CLI
- Operating System: Linux 
- Version of Amazon FreeRTOS: master
- Project: Custom 

**Expected behavior**
I would expect the sendto call to either fail, or transmit data.

**To reproduce**
Steps to reproduce the behavior:

1. create a socket with FreeRTOS_socket
2. Listen on a known host:port (e.g. with `sudo NC -ul 123`)
3. FreeRTOS_sendto to the host:port above
4. FreeRTOS_sendto returns successfully, but no data is sent

Thank you!

htibosch, 2019-12-03:
Hello Matteo,

> Not sure this is a bug,

No it isn't

> a limitation I'm unaware of

Yes it is

> or just me being clumsy

I wouldn't think so

This behaviour is known: it has to do with ARP resolution. Already since the oldest version of the library, a UDP packet can be turned into an ARP packet, in case the MAC-address is unknown.

There is a solution: call `FreeRTOS_OutputARPRequest()` with the IP-address of the peer ( network endian ) **before sending** the first UDP message.

Here is an example, just tested on a board:

~~~c
uint32_t xIPAddress = FreeRTOS_inet_addr_quick( 192, 168, 2, 5 );
MACAddress_t xMACAddress;
BaseType_t xRetry;

for( xRetry = 0; xRetry < 3; xRetry++ )
{
	uint32_t xUseAddress = xIPAddress;
	eARPLookupResult_t xResult = eARPGetCacheEntry( &( xUseAddress ), &( xMACAddress ) );
	if( xResult == eARPCacheHit )
	{
		/* The MAC-address is known. */
		break;
	}
	if( xResult == eCantSendPacket )
	{
		/* Something wrong with the IP-address. */
		break;
	}
	FreeRTOS_OutputARPRequest( xIPAddress );
	/* Wait 100 ms for a response. */
	vTaskDelay( pdMS_TO_TICKS( 100u ) );
}
~~~

This is a blocking version that takes between 0 and 300 ms. You could turn it into an asynchronous (non-blocking) version.

In my next post I will give more background information.

htibosch, 2019-12-03:
So why doesn't `FreeRTOS_sendto()` return an error code if the MAC-address is unknown?

Because `FreeRTOS_sendto()` is just an API, it will pass the network buffer to the IP-task, which will wake-up and work on it. We did not want the API to block and wait for a response from the IP-task. That would last an unknown time.
In fact, `FreeRTOS_sendto()` will return the length of the data if the network buffer could be created and if it was passed to the IP-task. It does not wait for the IP-task to work on it.

Why not hide it all, and do the ARP look-up automatically?

Because the IP-task doesn't want to block on any procedure at all. It could be implemented in an asynchronous way, but that would be quite complex and far from **deterministic**.

PS. A DNS look-up can be done in a non-blocking way, see `ipconfigDNS_USE_CALLBACKS`.

Note that **UDP-broadcasting** does not need address resolution: the hardware address `ff-ff-ff-ff-ff-ff` will be used.

One thing about `eARPGetCacheEntry()`:

~~~c
	uint32_t xUseAddress = xIPAddress;
	eARPLookupResult_t xResult = eARPGetCacheEntry( &( xUseAddress ), &( xMACAddress ) );
~~~

The function `eARPGetCacheEntry( &ip_address, &mac_address )` may also change the contents of the first parameter, the IP-address.
This happens when the IP-address is outside the current net-mask. In that case, the IP-address of the gateway shall be used, and the parameter will be overwritten. That is why I called it `xUseAddress`.
Hein

matteoscordino, 2019-12-03:
Thank you, @htibosch , that's exactly what was happening (and what I didn't know)
Great explanation, by the way!

closed 2019-12-03 by matteoscordino