aws/amazon-freertos #1576: xBLEStackTeardown in main.c

chegewara, 2019-12-03:
Can we fix `xBLEStackTeardown` when NimBLE is used, because in NimBLE we dont have `esp_bt_controller_mem_release`:
https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/application_code/main.c#L195

with something like that:
```
        esp_err_t xBLEStackTeardown( void )
        {
            BTStatus_t xRet = IotBle_Off(); // 
            const BTInterface_t* xBTinter = BTGetBluetoothInterface(); 
            xBTinter->pxDisable();
            xBTinter->pxBtManagerCleanup();
            return ESP_OK;
        }
```
esp32 master commit

abhidixi11, 2019-12-04:
Hello chegewara,
Thanks for the question. 

>>>because in NimBLE we dont have esp_bt_controller_mem_release:

Can you please explain on  what do mean by 'we don't have esp_bt_controller_mem_release' ?
Thanks.

chegewara, 2019-12-04:
`esp_bt_controller_mem_release` its not suppose to be used with NimBLE, its function from bluedroid stack.
NimBLE has its own function to deinit and it is already implemented in master branch:
https://github.com/aws/amazon-freertos/blob/390f9752d85495505e8ba0ad7b34e78cb4cd45ba/vendors/espressif/boards/esp32/ports/ble/nimble/iot_ble_hal_common_gap.c#L575

abhidixi11, 2019-12-04:
OK thanks for explanation chegewara.  We will look into it.

ravibhagavandas, 2019-12-04:
@chegewara 

As per ESP, `esp_bt_controller_mem_release`  function is used to release the bss data and other sections of the controller back to the heap.  This is applicable to NIMBLE port as well. A power cycle is required to reclaim this memory.

chegewara, 2019-12-05:
I will check it, thanks.
Here is issue i have opened earlier on espressif github
https://github.com/espressif/esp-idf/issues/4284

ravibhagavandas, 2019-12-05:
After checking  the below functions:
```
pxDisable();
pxBtManagerCleanup();
```
The functions needs to be part of `IotBle_Off()` but was disabled due to the bug with Bluedroid before. Now, since there is a NIMBLE implementation of `pxDisable() and pxBtManagerCleanup()` we will look into calling them from `IotBle_Off()` instead of `xBLEStackTeardown`

dhrishi, 2019-12-06:
@chegewara @ravibhagavandas 

`esp_bt_controller_mem_release` is neither specific to Bluedroid nor NimBLE which means both the host stacks can use it as it is a controller API.

It releases the controller memory for a particular mode (which you don't want to use) to heap. This API should be called before `esp_bt_controller_init()` or after `esp_bt_controller_deinit()`. The memory is reclaimed in the same boot cycle.
The only thing is the process cannot be reversed in the current boot cycle which means that if you want to use the mode whose memory you have already released then it requires a reboot.

You can see the entire documentation here: https://github.com/espressif/esp-idf/blob/release/v3.3/components/bt/include/esp_bt.h#L351

ravibhagavandas, 2019-12-10:
@dhrishi 
Thanks for summing up regarding `esp_bt_controller_mem_release`.
@chegewara 
We are working on changes to `IotBle_Off()` to include `pxDisable();` and `pxBtManagerCleanup(); `.

PR #1663 (2020-01-06): Fix BLE enable disable sequence
Fix enable/disable sequence in BLE to support disabling and re-enabling stack at runtime.

Description
-----------
* Refactor IotBle_On() and IotBle_Off() functions to be symmetrical - enable/teardown GATT services and BLE stack.
* Example usage in demo runner to disable non-used networks for demo . (Disable BLE stack to save power and reclaim RAM after WiFi provisioning when WiFi is connected).
* Tested on ESP32 and Nordic.

PR also fixes #1576

Checklist:
----------

- [x] I have tested my changes. No regression in existing tests.
- [ ] My code is Linted.

By  [... cut]
closed 2020-01-06 by ravibhagavandas