aws/amazon-freertos #1588: MQTT Demo code not working properly.

AagamMehta, 2019-12-06:
**Describe the bug**
I am trying to run demo code of MQTT but it stops infinitely after printing 'WIFI: SYSTEM_EVENT_STA_START'. I am attaching a log file for the same.

**System information**
- ESP32
- Linux
- 201910.00
- MQTT|

```cpp
 ~/amazon-freertos (master) $ ./vendors/espressif/esp-idf/tools/idf.py erase_flash flash monitor -p /dev/ttyUSB0 -B build
Setting IDF_PATH environment variable: /home/aagam/amazon-freertos/vendors/espressif/esp-idf
Checking Python dependencies...
Python requirements from /home/aagam/amazon-freertos/vendors/espressif/esp-idf/requirements.txt are satisfied.
Running esptool.py in directory /home/aagam/amazon-freertos/build
Executing "/usr/bin/python /home/aagam/amazon-freertos/vendors/espressif/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 460800 --after hard_reset erase_flash"...
esptool.py v2.8-dev
Serial port /dev/ttyUSB0
Connecting........_____....._____.
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 24:0a:c4:b0:0f:78
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 3.2s
Hard resetting via RTS pin...
Running make in directory /home/aagam/amazon-freertos/build
Executing "make -j 6 all"...
[  2%] Built target afr_3rdparty_tinycbor
[  2%] Built target idf_component_ulp
[  2%] Built target idf_component_log
[  3%] Built target partition_table
[  4%] Built target afr_kernel
[  6%] Built target idf_component_soc
[  7%] Built target idf_component_heap
[  7%] Built target idf_component_vfs
[  8%] Built target idf_component_newlib
[  8%] Built target idf_component_esp_ringbuf
[ 10%] Built target idf_component_freertos
[ 10%] Built target idf_component_tcpip_adapter
[ 11%] Built target idf_component_app_update
[ 11%] Built target idf_component_esp_event
[ 13%] Built target idf_component_driver
[ 14%] Built target idf_component_micro-ecc
[ 15%] Built target idf_component_spi_flash
[ 17%] Built target idf_component_app_trace
[ 17%] Built target idf_component_efuse
[ 17%] Built target idf_component_ethernet
[ 17%] Built target idf_component_xtensa-debug-module
[ 18%] Built target idf_component_bootloader_support
[ 19%] Built target idf_component_nvs_flash
[ 20%] Built target idf_component_pthread
[ 20%] Built target idf_component_smartconfig_ack
[ 20%] Built target esp32_linker_script
[ 21%] Built target idf_component_cxx
[ 21%] Built target idf_component_espcoredump
[ 22%] Built target idf_component_esp_adc_cal
[ 22%] Built target idf_component_expat
[ 23%] Built target idf_component_console
[ 23%] Built target idf_component_bt
[ 32%] Built target idf_component_wpa_supplicant
[ 34%] Built target idf_component_wear_levelling
[ 35%] Built target idf_component_sdmmc
[ 37%] Built target idf_component_freemodbus
[... 212 lines cut ...]
W (669) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (839) phy: phy_version: 4102, 2fa7a43, Jul 15 2019, 13:06:06, 0, 2
GAP procedure initiated: stop advertising.
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=480 adv_itvl_max=960
I (1149) wifi: wifi driver task: 3ffd10a0, prio:23, stack:3584, core=0
I (1149) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1149) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1179) wifi: wifi firmware version: c94f3e6
I (1179) wifi: config NVS flash: enabled
I (1179) wifi: config nano formating: disabled
I (1179) wifi: Init dynamic tx buffer num: 32
I (1179) wifi: Init data frame dynamic rx buffer num: 32
I (1189) wifi: Init management frame dynamic rx buffer num: 32
I (1189) wifi: Init management short buffer num: 32
I (1199) wifi: Init static rx buffer size: 1600
I (1199) wifi: Init static rx buffer num: 10
I (1209) wifi: Init dynamic rx buffer num: 32
I (1219) wifi: mode : sta (24:0a:c4:b0:0f:78)
I (1219) WIFI: SYSTEM_EVENT_STA_START

Done

```

Thank you!

elieDaan, 2019-12-06:
Yesterday I had this issue with using [amazon-freertos-examples](https://github.com/tgsong/amazon-freertos-examples). I had to erase_flash, don't use the amazon-config repo and then into the folder aws_quick_configuration delete your creds and redo the setup.

dan4thewin, 2019-12-08:
Hi, @AagamMehta .  Can you verify that your wifi SSID and password are correct in amazon-freertos/demos/include/aws_clientcredential.h ?

yuhui-zheng, 2019-12-13:
@AagamMehta Since this thread has been inactive for almost a week, I'm going to assume you were able to make some progress. I'm closing this thread for now, with additional message left below for reference. 

From the log you've posted, that run had issue setting MAC address. 
https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/esp-idf/components/esp32/system_api.c#L66

Static variable ```base_mac_addr``` is not set prior to calling ```esp_base_mac_addr_get()```, which will likely further cause IP stack failure.  ```base_mac_addr``` is set in ```esp_base_mac_addr_set()`` in the same file. I'm not sure how that run ended up there, but next time running into something similar could dump ```base_mac_addr``` to confirm.

closed 2019-12-13 by yuhui-zheng