aws/amazon-freertos #1589: Ninja Menuconfig is not functionning

elieDaan, 2019-12-06:
**Describe the bug**
After updated amazon-freertos repo and used it such as a subdirectory , related to this [
Document how to create a new cmake project outside the freertos source tree
](https://github.com/aws/amazon-freertos/issues/832), I canno't use ninja menuconfig, with or without IDF_SDK_CONFIG personnalized. 

But it's working with make menuconfig. 

**** Command used : 
=> cmake -S . -B build -DCMAKE_TOOLCHAIN_FILE=amazon-freertos/tools/cmake/toolchains/xtensa-esp32.cmake -GNinja

=> ./amazon-freertos/vendors/espressif/esp-idf/tools/idf.py menuconfig 

**System information**
- Which hardware board or part numbers? ESP32
- IDE used VS Studio
- Operating System MacOS
- Version of Amazon FreeRTOS : 201910.00-378-gd26ecc1dd
- Project [Custom Application] 
- If your project is a Custom Application, please add the relevant code snippet in the section 

**Expected behavior**
Run ninja menuconfig

***Terminal error***
Loading defaults file /Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/sdkconfig.defaults...
No such file or directory
FAILED: amazon-freertos/CMakeFiles/menuconfig 

**Code to reproduce the bug**

```cmake
# -------------------------------------------------------------------------------------------------
# Cmake to build project BOB and amazon freertos
# -------------------------------------------------------------------------------------------------
cmake_minimum_required(VERSION 3.13)

set(CMAKE_C_COMPILER /Users/XX/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc)
set(CMAKE_CXX_COMPILER /Users/XX/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++)

set(EXTRA_CFLAGS --save-temps -lm -l)
project(bob_v6)

file(GLOB SOURCES "/Users/XXDocuments/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/main/*.c" ABSOLUTE 
   
    "amazon-freertos/libraries/freertos_plus/standard/utils/src/iot_system_init.c"
    )
file(GLOB HEADERS "/Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/main/*.h" ABSOLUTE
)
add_executable(bob_v6 ${SOURCES} ${HEADERS})

# Tell IDF build to link against this target
set(IDF_PROJECT_EXECUTABLE bob_v6)

#Define the components
get_filename_component(ESPRESSIF_CODE_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/amazon-freertos/vendors/espressif/boards/esp32/aws_demos/application_code/espressif_code" ABSOLUTE)
get_filename_component(ESP_SPIFFS_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/spiffs" ABSOLUTE)
get_filename_component(ESP_SPIFFS_IMG_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/spiffs_image" ABSOLUTE)
get_filename_component(ESP_MKSPIFFS_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/mkspiffs" ABSOLUTE)
get_filename_component(ESP_SPIDRIVER_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/spidriver" ABSOLUTE)
get_filename_component(ESP_TFT_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/tft" ABSOLUTE)
get_filename_component(MQTT_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/mqtt_cartridge" ABSOLUTE)
get_filename_component(FOO_COMPONENT "${CMAKE_CURRENT_LIST_DIR}/components/foo" ABSOLUTE )

set(EXTRA_COMPONENT_DIRS    
                            #${EXTRA_COMPONENT_DIRS} 
                            ${FOO_COMPONENT}
                            ${ESPRESSIF_CODE_COMPONENT}
                            ${ESP_TFT_COMPONENT}
                            ${ESP_SPIFFS_COMPONENT}
                            ${ESP_SPIFFS_IMG_COMPONENT}
                            ${ESP_MKSPIFFS_COMPONENT}
                            ${ESP_SPIDRIVER_COMPONENT}
                            ${MQTT_COMPONENT}
)

list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})

# As of now there's no offical way to redefine config files outside of Amazon FreeRTOS source tree.
# This is a temporary approach to inject an include path so that this takes precedence over the
# config file directory inside Amazon FreeRTOS.

#include_directories(BEFORE amazon-freertos-configs)

# Bring in the include directories 
set(INCLUDE_DIRS ${INCLUDE_DIRS} 
    "amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/include"
    "amazon-freertos/demos/include/"
    "amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/include/"
    "${esp_idf_dir}/components/bt/bluedroid/api/include/api/" 
    "${AFR_ROOT_DIR}/vendors/espressif/esp-idf/components/bt/include/"
    )
include_directories(${INCLUDE_DIRS})

# include_directories(bob_v6 PRIVATE "amazon-freertos/demos/include")
# include_directories(bob_v6 PRIVATE "amazon-freertos/libraries/abstractions/platform/freertos/include")
# include_directories(bob_v6 PRIVATE "amazon-freertos/libraries/freertos_plus/standard/utils/include")
# include_directories(bob_v6 PRIVATE "amazon-freertos/libraries/freertos_plus/standard/utils/src")
# include_directories(bob_v6 PRIVATE "amazon-freertos/demos/dev_mode_key_provisioning/include/")
# include_directories(bob_v6 PRIVATE "amazon-freertos/libraries/abstractions/pkcs11/include/")
# include_directories(bob_v6 PRIVATE "amazon-freertos/libraries/3rdparty/pkcs11/")
# include_directories(bob_v6 PRIVATE "amazon-freertos/demos/network_manager/")
# include_directories(bob_v6 PRIVATE "amazon-freertos/libraries/abstractions/wifi/include/")

#Set SDK Config Defaults
set(IDF_SDKCONFIG_DEFAULTS "/Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/sdkconfig.defaults")

# Bring in the FreeRTOS code
# AFR_BOARD tells which board we need to target.
set(AFR_BOARD espressif.esp32_devkitc CACHE INTERNAL "")
set(EXTRA_CFLAGS --save-temps -lm)
add_subdirectory(amazon-freertos)

message("++++++++++++++++++++++++++++++++++")
message("++++          MY APP          ++++")
message("++++++++++++++++++++++++++++++++++")

message("IDF_EXTRA_COMPONENT_DIRS: ${IDF_EXTRA_COMPONENT_DIRS}")

message("INCLUDE_DIRS: ${INCLUDE_DIRS}")

message("Glob Source : ${SOURCES}")

message("Glob headers : ${HEADERS}")

message("IDF_CONFG: ${IDF_SDKCONFIG_DEFAULTS}")

target_link_libraries(bob_v6 PRIVATE AFR::demo_mqtt)
target_link_libraries(bob_v6 PRIVATE AFR::demo_ble)
target_link_libraries(bob_v6 PRIVATE AFR::demo_ble_numeric_comparison)

target_link_libraries(bob_v6 PRIVATE AFR::wifi)
target_link_libraries(bob_v6 PRIVATE AFR::utils)
target_link_libraries(bob_v6 PRIVATE AFR::ble)
target_link_libraries(bob_v6 PRIVATE AFR::ble_hal)
target_link_libraries(bob_v6 PRIVATE AFR::platform)
target_link_libraries(bob_v6 PRIVATE AFR::dev_mode_key_provisioning)
target_link_libraries(bob_v6 PRIVATE AFR::mqtt)
target_link_libraries(bob_v6 PRIVATE AFR::pkcs11)
target_link_libraries(bob_v6 PRIVATE 3rdparty::pkcs11)
target_link_libraries(bob_v6 PRIVATE AFR::serializer)
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
Add any other context about the problem here.

Thank you!

dan4thewin, 2019-12-08:
When you run cmake, it should create a ninja.build file, I think.  Would you share the contents of that file, please?

elieDaan, 2019-12-09:
There is the build.ninja attached after running ninja menuconfig [build.ninja](https://gist.github.com/elieDaan/8f1f9df6f6e102685616bfbc7c8cb7da)

tgsong, 2019-12-12:
@elieDaan I just figured where the issue is, inside this file `amazon-freertos/vendors/espressif/esp-idf/tools/cmake/kconfig.cmake` where the `menuconfig` target is defined, the dependency `mconf-idf` is referencing a wrong location.

On line 51, change `set(MCONF ${CMAKE_BINARY_DIR}/kconfig_bin/mconf-idf)` to `set(MCONF ${CMAKE_CURRENT_BINARY_DIR}/kconfig_bin/mconf-idf)`. Basically, the `mconf-idf` should point to `build/amazon-freertos/kconfig_bin/mconf-idf`, note there's a `amazon-freertos` subdirectory. I'll raise a PR to fix this problem and let Espressif know.

@mahavirj, Looks like the issue is also in the ESP-IDF upstream repo, can you verify this? https://github.com/espressif/esp-idf/blob/93a8603c545fb8e54741d6685146e2f3b874378d/tools/cmake/kconfig.cmake#L36

elieDaan, 2019-12-12:
OMG It's working perfectly ! Very good job @tgsong, thanks a lot because I spend a lot of time on this ;)

closed 2019-12-12 by elieDaan

mahavirj, 2019-12-13:
> @mahavirj, Looks like the issue is also in the ESP-IDF upstream repo, can you verify this? https://github.com/espressif/esp-idf/blob/93a8603c545fb8e54741d6685146e2f3b874378d/tools/cmake/kconfig.cmake#L36

@tgsong Correct fix should be: https://github.com/espressif/esp-idf/commit/ded0ac352d90cda800bd54ec5c71635ffe38229a
Seems like we did not backport this one to our `release/v3.3` branch, I will ensure that its taken care of and when we migrate ESP-IDF to next release in AFR, change will reflect here as well. (CC @shubhamkulkarni97)

tgsong, 2019-12-13:
@mahavirj, thanks, I raised a PR #1614 to fix for this version first.