aws/amazon-freertos #1606: esp32 ledc set freq error

chegewara, 2019-12-12:
ESP32 ledc driver error:
https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/esp-idf/components/driver/ledc.c#L443

https://github.com/espressif/esp-idf/issues/4476

yuhui-zheng, 2019-12-12:
Hi @chegewara,

Thank you for linking the issue from espressif/esp-idf. Could you elaborate what is the goal for raising issue in this repo? 

If there's an update from Espressif in esp-idf, I guess we could take the change from Espressif as we would normally do. (PR is required here.)

chegewara, 2019-12-12:
As long as there is no PR from esp-idf the issue still exists in this repository, at least for me.
If this is not proper way to post issues here feel free to close it and i wont open new issue.

yuhui-zheng, 2019-12-13:
It's totally fine to open issue here, just trying to see how we can help. 

When there is an official esp-idf release from Espressif, we'll bump esp-idf version once we've done our integration testing. We don't generally patch vendor code directly in this repo, since the next sync from upstream may override patches. 

And with a quick glance, it seems ```ledc_set_freq()``` in our repo looks the same with esp-idf v4.0 release (https://github.com/espressif/esp-idf/blob/release/v4.0/components/driver/ledc.c#L487). Would it be possible to submit PR to Espressif repo directly? (I'm speculating from your email @esp32.eu.org.)

chegewara, 2019-12-14:
I am not espressif employee, even if i own esp32.eu.org domain.

closed 2019-12-16 by chegewara