aws/amazon-freertos #1641: DeviceTester FreeRTOS_Integrity test fail due to non-determinism in test execution order.

shuopeng-deng, 2019-12-17:
The files {tests/include/aws_clientcredential_keys.h, tests/include/aws_clientcredential.h, tests/common/aws_test_runner.c} get modified from multiple tests.

If any tests that modifes the above file runs before FreeRTOS_Integrity tests then FreeRTOS_Integrity  test will fail to validate the hash of those files.

It seem that most of the time device tester executes FreeRTOS_Integrity before other tests. But that does not seem to be guaranteed. In the attached log, FreeRTOS_Integrity is executed after FullBLE tests and FullPKCS11_ECC tests.

As a result when FreeRTOS_Integrity runs, the above files has been modified by the BLE and PSCS11 test and FreeRTOS_Integrity fails.

Device tester version: 1.5.1

[test_manager.log](https://github.com/aws/amazon-freertos/files/3975178/test_manager.log)

dcgaws, 2019-12-20:
Hello, to confirm, are you able to successfully execute other test groups that specifically connect to the cloud? Those would include the full MQTT and Shadow integration tests, for example.

shuopeng-deng, 2019-12-20:
yes

kozelc, 2020-01-02:
Thank you Shuopeng for reporting this issue. I am able to confirm this is a problem with IoT Device Tester specific to the FullBLE test group, where the original source code files are not being restored after the test executes. I will work on fixing this issue.

To work around this issue, you can configure IDT to run in parallel testing mode (even if you are only using one device). To use parallel testing mode you must use the `{{testData.sourcePath}}` placeholder in the buildTool.command, flashTool.command, and vendorPath fields of the userdata.json. 
See the "Configure Build, Flash, and Test Settings" section of [Device Tester Docs](https://docs.aws.amazon.com/freertos/latest/userguide/qual-steps.html) for more details about using this placeholder.

Parallel testing mode behaves differently in how it modifies the source code for each test, which will circumvent the issue you are seeing where the BLE test modifies the source code.

xiyunl, 2020-01-16:
Hi @shuopeng-deng,

Were you able to fix the issue with parallel testing mode? Please feel free to let us know if you need any help.

qiutongs, 2020-02-11:
If there is still a problem, feel free to reopen this ticket or create a new one.

closed 2020-02-11 by qiutongs