aws/amazon-freertos #1646: SOCKETS_SO_WAKEUP_CALLBACK memory leak in secure_sockets

dougbenedict, 2019-12-18:
**Describe the bug**
The SOCKETS_SO_WAKEUP_CALLBACK processing for lwip creates a background thread which performs a select() on the socket and invokes the callback routine when there is activity on the socket.

Setting the SOCKETS_SO_WAKEUP_CALLBACK with a NULL callback parameter calls the routine  prvRxSelectClear() which is empty except for a /* TODO */ comment. This orphans the select thread rather than deleting it and cleaning up. Open and close enough sockets using the SOCKETS_SO_WAKEUP_CALLBACK option and the kernel heap will run out of memory.

In the freertos_plus_tcp iot_secure_sockets.c implementation, the SOCKETS_SO_WAKEUP_CALLBACK option is not handled at all.

**System information**
- Operating System Windows
- Discovered in 201910.00 but I see the code is the same in the current master
- Project Custom Application

**Expected behavior**
The spawned select task should be deleted and the resources reclaimed when the SOCKETS_SO_WAKEUP_CALLBACK option is called with a NULL callback. And the lwip and freertos_plus_tcp implementations should be consistent.

**To reproduce**
Steps to reproduce the behavior:

Using the lwip network stack.

1. Create a TCP socket using SOCKETS_socket():
socket = SOCKETS_Socket(SOCKETS_AF_INET, SOCKETS_SOCK_STREAM, SOCKETS_IPPROTO_TCP);
2. Install an activity callback using the SOCKETS_SO_WAKEUP_CALLBACK option:
 SOCKETS_SetSockOpt(socket, 0, SOCKETS_SO_WAKEUP_CALLBACK, &tcp_socket_callback, sizeof(&cy_tcp_socket_callback));
3. Use the TCP socket normally.
4. When done with the TCP socket, remove the callback:
  SOCKETS_SetSockOpt(socket, 0, SOCKETS_SO_WAKEUP_CALLBACK, NULL, 0);
5. Close the socket:
  SOCKETS_Close(socket);

The select task is now orphaned and the resources are not recovered.

htibosch, 2019-12-20:
I felt like responding to this post, until I saw that it is using lwIP.

However, the macro `SOCKETS_SO_WAKEUP_CALLBACK` seems to borrowed from FreeRTOS+TCP, in which it is defined with the same value:

~~~c
    #define FREERTOS_SO_WAKEUP_CALLBACK	( 17 )
~~~

This socket option binds a function to a socket:

~~~c
	case FREERTOS_SO_WAKEUP_CALLBACK:
	{
		/* Each socket can have a callback function that is executed
		when there is an event the socket's owner might want to
		process. */
		pxSocket->pxUserWakeCallback = ( SocketWakeupCallback_t ) pvOptionValue;
		xReturn = 0;
	}
	break;
~~~

The function `pxUserWakeCallback` will be called by the IP-task after every important event:

- An outgoing `connect()` succeeded
- A new connection is waiting for `accept()`
- Data has been sent
- Data has been received
- An error ( usually malloc ) has occurred
- The connection got broken

This is one of the methods that can be used to handle multiple sockets within a single task.
Usually `vTaskNotify()` will be called from the callback, so the task can work on it.

The lwIP variant does more or less the same: it will poll the socket by calling `lwip_select()` and call an application hook `rx_callback()` when there is data available.

I gladly leave this issue to maintainers of IoT secure sockets.

htibosch, 2019-12-21:
> In the freertos_plus_tcp iot_secure_sockets.c implementation, the
> SOCKETS_SO_WAKEUP_CALLBACK option is not handled at all.

In `freertos_plus_tcp/iot_secure_sockets.c`, all unknown options are handled by falling back to:

~~~c
    default:
        lStatus = FreeRTOS_setsockopt( pxContext->xSocket,
            lLevel, lOptionName, pvOptionValue, xOptionLength );
        break;
~~~

which will handle `FREERTOS_SO_WAKEUP_CALLBACK`. When called with a NULL pointer, the cal-back mechanism will be disabled.

However, the `iot_secure_sockets.c` module for Infineon, Nuvoton, TI, NXP, and ST do not seem to recognise the `SO_WAKEUP_CALLBACK` socket option, and return an error.

abhidixi11, 2019-12-23:
Thanks for reporting the problem @dougbenedict and thanks for looking into it @htibosch. We are already have an open internal ticket to investigate LWIP SOCKETS_SO_WAKEUP_CALLBACK behavior.  I'll leave the issue open till that ticket is resolved. 
Thanks.

AniruddhaKanhere, 2021-10-15:
@dougbenedict, apologies for the late reply.
We revisited this issue and I have created a PR https://github.com/aws/amazon-freertos/pull/3406 to try and fix this issue. Let me know if this works for you.

Thanks

closed 2021-10-18 by AniruddhaKanhere