aws/amazon-freertos #1649: Improve error message for Device Tester

shuopeng-deng, 2019-12-19:
When device tester fails, it is difficult to find the cause of the failure. Sometimes, no error message is generated at all. See attached logs.

[Full_TLS.log](https://github.com/aws/amazon-freertos/files/3985330/Full_TLS.log)
[metrics.log](https://github.com/aws/amazon-freertos/files/3985331/metrics.log)
[test_manager.log](https://github.com/aws/amazon-freertos/files/3985332/test_manager.log)

dcgaws, 2019-12-23:
Hello, do you observe any serial debug output from the device? And/or, if you attach a debugger, is the app starting?

I ask because the additional debugging effort this time will give us a better idea of how to capture the issue in the device tester output. Thank you!

kozelc, 2019-12-31:
Thank you for the feedback, one of our goals for 2020 is to improve the user experience of IoT Device Tester which includes error messaging.

For the specific issue you are facing based on your logs, "zero test case was executed", this would happen when Device Tester is able to read "---------STARTING TESTS---------" and "-------ALL TESTS FINISHED-------" from the device serial output without seeing any information about the test results in between.

Can you verify no changes have been made to the test code by running `run-suite --group-id FreeRTOSIntegrity` and verifying that it is passing.

If you were able to resolve the issue, please let us know how as that will help us know how we can detect issues and recommend solutions in our error messages.

sarenameas, 2020-01-14:
Hello @shuopeng-deng,

Were you able to follow the advice that @kozelc made? Are you still having issues?

qiutongs, 2020-02-12:
Thank you reporting the problem. If there is still an issue, feel free to re-open this or create a new one.

closed 2020-02-12 by qiutongs