aws/amazon-freertos #1651: Impossible to use touch pad interrupt

elieDaan, 2019-12-20:
**Describe the bug**
I need to uses the touch pad in my AFR system with interrupt mode. I used the [touch pad interrupt example](https://github.com/espressif/esp-idf/tree/release/v3.3/examples/peripherals/touch_pad_interrupt). 
So by polling (filter mode), the esp is able to read the touch capacitance value but with interrupt mode nothing happened. 

I also tried with a customized code but when I touch the sensor, it crash and reboot.

**System information**
- Which hardware board or part numbers? ESP32
- IDE used VS Code
- Operating System MacOS
- Version of Amazon FreeRTOS : 201910.00-258-g8b4ebca33
- Project [Custom Application]
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
Interrupt trigger and print (T activated !)

**Screenshots or console output**
*With example code : *
FILTER MODE : 
I (69705) Touch pad: T7 activated!
I (69705) Touch pad: value: 708; init val: 789
I (74905) Touch pad: Waiting for any pad being touched...
I (79905) Touch pad: Waiting for any pad being touched...
I (84905) Touch pad: Waiting for any pad being touched...
I (89905) Touch pad: Waiting for any pad being touched...
W (89905) Touch pad: Change mode...interrupt mode
I (94905) Touch pad: Waiting for any pad being touched...
I (99905) Touch pad: Waiting for any pad being touched...
I (104905) Touch pad: Waiting for any pad being touched...
I (109905) Touch pad: Waiting for any pad being touched...

*With customized code :* 
core=0 port_interruptNesting=1

../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:315 (vPortAssertIfInISR)- assert failed!
abort() was called at PC 0x400880ef on core 0
0x400880ef: vPortAssertIfInISR at /Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:258

ELF file SHA256: 1b2df2076b1659d447af8cab75873fb81b68832090181ae4db72de6983128556

Backtrace: 0x40087bf0:0x3ffbebf0 0x40087e1d:0x3ffbec10 0x400880ef:0x3ffbec30 0x400899fe:0x3ffbec50 0x40081d5b:0x3ffbec70 0x400d2b0d:0x3ffbec90 0x400da0c0:0x3ffbecb0 0x40081816:0x3ffbecd0 0x400d2be5:0x00000000
0x40087bf0: invoke_abort at /Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x40087e1d: abort at /Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/build/../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c:715

0x400880ef: vPortAssertIfInISR at /Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:258

0x400899fe: vTaskEnterCritical at /Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/build/../amazon-freertos/freertos_kernel/tasks.c:5068

0x40081d5b: touch_pad_clear_status at /Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/build/../amazon-freertos/vendors/espressif/esp-idf/components/driver/rtc_module.c:441

0x400d2b0d: handler_intr_rtc at /Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/build/../main/capacitive_sensor.c:66

0x400da0c0: rtc_isr at /Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/build/../amazon-freertos/vendors/espressif/esp-idf/components/driver/rtc_module.c:441

0x40081816: _xt_lowint1 at /Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/build/../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1105

0x400d2be5: task_check_door at /Users/XX/Documents/XX/Phase_4_AWS/bob_v7_gui/Developement_GUI_ESP32/build/../main/capacitive_sensor.c:156

**To reproduce**
Steps to reproduce the behavior:

1. Try with touch pad interrupt example provided by espressif
2. Try with my customized code

**Code to reproduce the bug**

[My customized code ](https://gist.github.com/elieDaan/ea301cb7a93689217ee5884b314e69d1
)

Thank you!

mahavirj, 2019-12-20:
@elieDaan  Can you please try with following fix and let me know if it helps to fix your issue:

```
diff --git vendors/espressif/esp-idf/components/driver/rtc_module.c vendors/espressif/esp-idf/components/driver/rtc_module.c
index d36acc6db..c07eb2ecb 100644
--- vendors/espressif/esp-idf/components/driver/rtc_module.c
+++ vendors/espressif/esp-idf/components/driver/rtc_module.c
@@ -785,9 +785,9 @@ uint32_t IRAM_ATTR touch_pad_get_status()
 
 esp_err_t IRAM_ATTR touch_pad_clear_status()
 {
-    portENTER_CRITICAL(&rtc_spinlock);
+    portENTER_CRITICAL_SAFE(&rtc_spinlock);
     SENS.sar_touch_ctrl2.touch_meas_en_clr = 1;
-    portEXIT_CRITICAL(&rtc_spinlock);
+    portEXIT_CRITICAL_SAFE(&rtc_spinlock);
     return ESP_OK;
 }
```

elieDaan, 2019-12-20:
You are so amazing guys ! Thanks you very much it's working.  Do you know what is the git command to update with your code ?

dcgaws, 2019-12-20:
Hello, I'm glad to hear your issue is resolved. Regarding git usage, I'll respectfully refer you to external documentation for that. Perhaps https://stackoverflow.com/questions/2249852/how-to-apply-a-patch-generated-with-git-format-patch/52418860.

closed 2019-12-20 by dcgaws

elieDaan, 2020-04-08:
Hey @mahavirj , I've updated aws with esp-idf as a submodule. Please add this fix to [esp-afr-sdk](https://github.com/espressif/esp-afr-sdk )

shubhamkulkarni97, 2020-04-09:
> Hey @mahavirj , I've updated aws with esp-idf as a submodule. Please add this fix to [esp-afr-sdk](https://github.com/espressif/esp-afr-sdk)

@elieDaan 

I have added this fix to esp-afr-sdk, which can be found here: https://github.com/espressif/esp-afr-sdk/commit/d10ef9c6db4160c64aa95f503b3f4f050ffd5445

Thanks