aws/amazon-freertos #1652: AFR as component esp32

chegewara, 2019-12-20:
With easy to setup project thanks to this docs https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project i am trying to start with hello-world example from esp-idf. Project is simple so preparing CMakeLists.txt should be easy:
```
# The following lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.13)

# include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(hello-world)

add_executable(my_app main/main.c)

# Tell IDF build to link against this target.
set(IDF_PROJECT_EXECUTABLE my_app)

# Override the configurations for Amazon FreeRTOS.
include_directories(BEFORE amazon-freertos-configs)

# Add amazon freertos as a subdirectory. AFR_BOARD tells which board to target.
set(AFR_BOARD espressif.esp32_devkitc CACHE INTERNAL "")
add_subdirectory(amazon-freertos)

# Link against the mqtt library so that we can use it. Dependencies are transitively
# linked.
target_link_libraries(my_app PRIVATE AFR::mqtt)
```
but im getting error:
```
Loading defaults file /home/chegewara/demos/afr-submodule/hello_world/build/sdkconfig.defaults...
-- IDF_VER: v3.3-163-g601a03e
CMake Error: File /home/chegewara/demos/afr-submodule/hello_world/.git/packed-refs does not exist.
CMake Error at build/amazon-freertos/CMakeFiles/git-data/grabRef.cmake:27 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  amazon-freertos/vendors/espressif/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:80 (include)
  amazon-freertos/vendors/espressif/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:90 (get_git_head_revision)
  amazon-freertos/vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:253 (git_describe)
  amazon-freertos/vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:384 (app_get_revision)
  amazon-freertos/vendors/espressif/boards/esp32/CMakeLists.txt:433 (idf_import_components)
  amazon-freertos/CMakeLists.txt:49 (include)

CMake Error at build/amazon-freertos/CMakeFiles/git-data/grabRef.cmake:28 (file):
  file failed to open for reading (No such file or directory):

    /home/chegewara/demos/afr-submodule/hello_world/build/amazon-freertos/CMakeFiles/git-data/packed-refs
Call Stack (most recent call first):
  amazon-freertos/vendors/espressif/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:80 (include)
  amazon-freertos/vendors/espressif/esp-idf/tools/cmake/third_party/GetGitRevisionDescription.cmake:90 (get_git_head_revision)
  amazon-freertos/vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:253 (git_describe)
  amazon-freertos/vendors/espressif/esp-idf/tools/cmake/idf_functions.cmake:384 (app_get_revision)
  amazon-freertos/vendors/espressif/boards/esp32/CMakeLists.txt:433 (idf_import_components)
  amazon-freertos/CMakeLists.txt:49 (include)

-- Project is not inside a git repository,                         will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Adding linker script /home/chegewara/demos/afr-submodule/hello_world/build/amazon-freertos/esp-idf/esp32/esp32_out.ld
-- Adding linker script /home/chegewara/demos/afr-submodule/hello_world/amazon-freertos/vendors/espressif/esp-idf/components/esp32/ld/esp32.rom.ld
-- Adding linker script /home/chegewara/demos/afr-submodule/hello_world/amazon-freertos/vendors/espressif/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script /home/chegewara/demos/afr-submodule/hello_world/amazon-freertos/vendors/espressif/esp-idf/components/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/chegewara/demos/afr-submodule/hello_world/amazon-freertos/vendors/espressif/esp-idf/components/esp32/ld/esp32.rom.spiram_incompatible_fns.ld
-- Component libraries: 
=========================Resolving dependencies==========================
module disabled: posix
reason:          posix::mcu_port is not defined by vendor.
dependency path: posix->posix::mcu_port

====================Configuration for Amazon FreeRTOS====================
  Version:                 201912.00
  Git version:             201912.00

Target microcontroller:
  vendor:                  Espressif
  board:                   ESP32-DevKitC
  description:             Development board produced by Espressif that comes in two 
                           variants either with ESP-WROOM-32 or ESP32-WROVER module
  family:                  ESP32
  data ram size:           520KB
  program memory size:     4MB

Host platform:
  OS:                      Linux-4.15.0-72-generic
  Toolchain:               xtensa-esp32
  Toolchain path:          /home/chegewara/esp/xtensa-esp32-elf
  CMake generator:         Ninja

Amazon FreeRTOS modules:
  Modules to build:        ble, ble_hal, ble_wifi_provisioning, common, crypto, defender, 
                           dev_mode_key_provisioning, freertos_plus_tcp, greengrass, 
                           https, kernel, mqtt, ota, pkcs11, pkcs11_implementation, 
                           platform, secure_sockets, serializer, shadow, tls, wifi
  Enabled by user:         ble, ble_hal, ble_wifi_provisioning, defender, greengrass, 
                           https, mqtt, ota, pkcs11, pkcs11_implementation, platform, 
                           secure_sockets, shadow, wifi
  Enabled by dependency:   common, crypto, demo_base, dev_mode_key_provisioning, 
                           freertos, freertos_plus_tcp, kernel, ota_http, ota_mqtt, 
                           pkcs11_mbedtls, secure_sockets_freertos_plus_tcp, serializer, 
                           tls, utils
  3rdparty dependencies:   http_parser, jsmn, mbedtls, pkcs11, tinycbor
  Available demos:         demo_ble, demo_ble_numeric_comparison, demo_defender, 
                           demo_greengrass_connectivity, demo_https, demo_mqtt, demo_ota, 
                           demo_shadow, demo_tcp, demo_wifi_provisioning
  Available tests:         
=========================================================================

-- Configuring incomplete, errors occurred!
See also "/home/chegewara/demos/afr-submodule/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "/home/chegewara/demos/afr-submodule/hello_world/build/CMakeFiles/CMakeError.log".
```

chegewara, 2019-12-20:
Ok, i think i found problem. 
Looks like `git init` is not enough and i need to create repository on git and make at least 1 commit.

tgsong, 2019-12-20:
@chegewara Yes, `git init` is not enough, I've seen this before, you need at least an initial commit.

chegewara, 2019-12-20:
Well, almost have it build. I copy/paste main.c from afr aws_demo folder i am getting different header files missing, depend on what AFR modules i add as target_link_libraries:
```
FAILED: CMakeFiles/my_app.dir/main/main.c.obj 
../main/main.c:63:41: fatal error: iot_network_manager_private.h: No such file or directory
```
```
../amazon-freertos/libraries/freertos_plus/aws/ota/src/http/aws_iot_ota_http.c:43:40: fatal error: aws_iot_ota_agent_internal.h: No such file or directory
```
```
../amazon-freertos/libraries/freertos_plus/aws/ota/src/http/aws_iot_ota_http.c:42:31: fatal error: aws_iot_ota_agent.h: No such file or directory
compilation terminated.
[23/25] Building C object CMakeFiles/my_app.dir/amazon-freertos/demos/wifi_provisioning/aws_wifi_connect_task.c.obj
```
and probably more

chegewara, 2019-12-20:
I started with target_link_libraries from getting started, which is simple and adds only AFR::mqtt, then i added all libraries i will need eventually:
```
target_link_libraries(my_app PRIVATE AFR::demo_mqtt AFR::shadow AFR::ota AFR::wifi AFR::ble AFR::ble_wifi_provisioning AFR::demo_ble_numeric_comparison AFR::demo_wifi_provisioning)
```
but none of it can compile.

tgsong, 2019-12-20:
@chegewara I can confirm this issue with the newest release. [My example repo](https://github.com/tgsong/amazon-freertos-examples) does not have this problem though. I'll get back to you once I figure what's going on.

chegewara, 2019-12-20:
Thanks @tgsong for help and support.

tgsong, 2019-12-20:
@chegewara I found a bug introduced from a previous PR, I've raised a PR to fix it.

Thanks for trying it out and report the issue. I'll also coordinate with our CI team to add a check in our PR so that we can catch this kind of problem earlier in the future.

chegewara, 2019-12-20:
Thank you so much for taking care of it and fixing it so quick.

chegewara, 2019-12-21:
I can confirm that this fix works, thanks.

chegewara, 2019-12-21:
@tgsong One more thing, could you confirm that extra components in components subfolder works with this setup in CMakeLists.txt, because my app is complaining about missing header files:
```
# Add some extra components. IDF_EXTRA_COMPONENT_DIRS is an variable used by ESP-IDF
# to collect extra components.
get_filename_component(
    EXTRA_COMPONENT_DIRS
    "components/foo" ABSOLUTE
)
list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})
```

tgsong, 2019-12-21:
@chegewara Yes it does work for me. Do you also have the CMakeLists.txt file under `components/foo` folder?

chegewara, 2019-12-21:
Yes, i have all configured, but i made mistake. Ive been trying to use `get_filename_component` to load few components at the same time, but i need to use it to load 1 component at the time.

I think this is last issue i have, and it seems to be AFR issue, or i have to add component required to component CMakeLists.txt:
```
In file included from ../amazon-freertos/vendors/espressif/boards/esp32/aws_demos/config_files/iot_config.h:76:0,
                 from ../amazon-freertos/demos/include/iot_demo_logging.h:35,
                 from ../components/stockd_ble/example/demo_client_callbacks.c:13:
../amazon-freertos/demos/include/iot_config_common.h:35:50: fatal error: platform/iot_platform_types_freertos.h: No such file or directory
```
Here is my current CMakeLists.txt from component:
```
set(COMPONENT_ADD_INCLUDEDIRS 
    include 
    example 
)

# set(COMPONENT_REQUIRES )
# set(COMPONENT_PRIV_REQUIRES )

set(COMPONENT_SRCDIRS source example)

register_component()
```
Anyway, i think i will use ESP_LOGX in my components instead.

chegewara, 2019-12-21:
I have one more question, it is related to my last post and issue with compiling.
Now, when AFR is used as component, and i have my own components in folder `components`, how i can add for example `AFR::wifi` to my component dependency or linking?

Let say i have CMakeLists.txt like this:
```
set(COMPONENT_ADD_INCLUDEDIRS 
    include 
    example 
)

# set(COMPONENT_REQUIRES )
# set(COMPONENT_PRIV_REQUIRES )

set(COMPONENT_SRCDIRS source example)

register_component()
```
I cant use `add_library` and `target_link_libraries` in my component because i get error.

chegewara, 2019-12-21:
I have second project with AFR as component, without my own components this time and this is error i have (which i didnt have with you demo repo):
```
../amazon-freertos/libraries/freertos_plus/aws/ota/src/mqtt/aws_iot_ota_mqtt.c:39:40: fatal error: aws_iot_ota_agent_internal.h: No such file or directory
```

tgsong, 2019-12-23:
@chegewara How are you using `target_link_libraries` function? I did try this myself and it was working, here's my CMakeLists.txt for the component,

```cmake
# include paths of this components.
set(COMPONENT_ADD_INCLUDEDIRS include)

# source files of this components.
set(COMPONENT_SRCDIRS src)
# Alternatively, use COMPONENT_SRCS to specify source files explicitly
# set(COMPONENT_SRCS src/foo.c)

# add this components, this will define a CMake library target.
register_component()

# standard CMake function can be used to specify dependencies. ${COMPONENT_TARGET} is defined
# from esp-idf when you call register_component, by default this should be the folder name.
target_link_libraries(${COMPONENT_TARGET} PRIVATE AFR::mqtt)
```
and within foo.c I'm able to include the header.

chegewara, 2019-12-25:
Looks like i had to copy/paste too many config files from aws to custom configs folder.
At least with project that is not using additional components.

chegewara, 2019-12-25:
With project that is using additional components i have bigger problem, because i have to modify AFR::ble. My modifications are rather big, because i implemented BLE client API, which is missing in AFR. It is including update existing API with ble scan and adding gattc API.

chegewara, 2019-12-25:
I have solution, but it is only partially solution. I could use esp32 BLE library or even esp-idf client API, but i have to use AFR::wifi in my own component and here is a problem. 
Sorry for being so lame, but i didnt have to work with cmake before and using cmake with AFR seems to be for pro.

tgsong, 2019-12-26:
Looks like you mentioned few things, I'm trying to summarize it here,

1. You need copy some configuration files from AFR. Otherwise it won't build.
  A: The solution we have for configurations is just workaround. If you don't override anything then the configuration in our demo code will be used, which in most case is not what you want. Especially there are some configurations related to demos and not libraries so it could cause confusion.
1. You have to modify the BLE library because it's missing client API. 
  A: This indeed sounds rather big, it's probably out of my scope, if you have questions or feature request regarding to the BLE, I suggest you open another issue.
1. You need to link `AFR::wifi` in your own component but ran into build error.
  A: Looks like you're using the `201906.00` version because `aws_wifi.h` only exist in that version, and we renamed it in following release. I did try to link to `AFR::wifi` in the `foo` component in my example repo, it's building fine and I was able to do `#include <iot_wifi.h>`. I couldn't tell exactly what you're trying to do from your error message, would you mind uploading your code to a repo? It would be easier to explain from there.

chegewara, 2019-12-26:
Hi @tgsong 
1. Yes, i know i have to copy some config files and i did it without problems.
2. Yes, its big change and i have it working with make, but i had to modify AFR code, either .c files and headers, but this is not that big issue and we can ignore it now.
3. Yes, i want to link AFR::wifi in my component, which i failed. Logs are from old code (pre release-1.5), but when i change headers to <iot_wifi.h> i am getting the same error. 
Do i have to add AFR::wifi in component CMakeLists.txt as required component or modify it some special way (i will try with clean AFR and foo component)?

This is my current CMakeLists.txt for component that needs to use AFR::wifi:
```
set(COMPONENT_ADD_INCLUDEDIRS 
    include  example // <--- my own component files
)
set(COMPONENT_PRIV_REQUIRES bt ) // <---    use esp-idf ble
set(COMPONENT_SRCDIRS source example) // <--- my own component files
register_component()
```

tgsong, 2019-12-27:
No you don't need to modify it, just do the normal `target_link_libraries(${COMPONENT_TARGET} PRIVATE AFR::wifi)` in your component CMake file and you should be able to use it. One thing I noticed is that the error is from `vendors/espressif/boards/esp32/aws_demos/application_code/components/stockd_m5stack/source/display_system_menu.c`, this is something we don't have in our codebase. Did you add this component yourself? I was referring to the components sitting outside of our codebase, so this might be something different though.

chegewara, 2019-12-27:
Yes, it is my own component with pre release-1.5. I have issue with adding components in cmake, but my code is compiling with GNU make. Now i am trying to move my components from `vendors/espressif/boards/esp32/aws_demos/application_code/components/` to AFR as component and master branch and im having exactly the same issues compiling with cmake.

I can create private repository, can i invite you to check what i am doing wrong?

tgsong, 2019-12-27:
Sure, I can check it for you. Can you share the link?

chegewara, 2019-12-27:
Ok, i am missing this line in component CMakeLists.txt:
https://github.com/tgsong/amazon-freertos-examples/blob/master/components/foo/CMakeLists.txt#L14

It is not shown in this docs:
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project

Thanks for help, again.

chegewara, 2019-12-27:
Now i have another issue, but it will be fixed in AFR. I am using some functions from AFR in my component, which are in this demo folder:
https://github.com/aws/amazon-freertos/tree/master/demos/network_manager

No CMakeLists.txt in it, so i cant add it to my component CMakeLists.txt with `target_link_libraries`.

tgsong, 2019-12-27:
@chegewara Network manger is in `AFR::demo_base`, see [here](https://github.com/aws/amazon-freertos/blob/6bc2f67f9b769f3f90956585778c4d884256d5a2/demos/CMakeLists.txt#L12). However, network manager is not supposed to be here, we will probably move it in the future, but for now you can link to this as a work around.

I'll work with our doc team to add the use of `target_link_libraries` for components.

chegewara, 2019-12-27:
Thanks, i hope this will help others with AFR as component.

With your help I fixed almost all issues i have. What can you tell about that one:
```
[7/107] Building C object amazon-freertos/esp-idf/stockd_m5stack/CMakeFiles/idf_comp..._/__/amazon-freertos/libraries/freertos_plus/aws/ota/src/http/aws_iot_ota_http.c.ob
FAILED: amazon-freertos/esp-idf/stockd_m5stack/CMakeFiles/idf_component_stockd_m5stack.dir/__/__/amazon-freertos/libraries/freertos_plus/aws/ota/src/http/aws_iot_ota_http.c.obj 
/home/chegewara/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DGCC_NOT_5_2_0=0 -DHAVE_CONFIG_H -DIDF_VER=\"v3.3-163-g601a03e\" -DMBEDTLS_CONFIG_FILE=\"/home/chegewara/demos/afr-submodule/hello_world/amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/include/mbedtls/esp_config.h\" -I../aws-configs -I../components/stockd_m5stack/include -I../components/stockd_m5stack/example -Iconfig -I../amazon-freertos/vendors/espressif/esp-idf/components/nvs_flash/include -I../amazon-freertos/vendors/espressif/esp-idf/components/spi_flash/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/mbedtls/port/include -I../amazon-freertos/libraries/3rdparty/mbedtls/include -I../amazon-freertos/libraries/3rdparty/mbedtls/include/mbedtls -I../amazon-freertos/libraries/abstractions/pkcs11/mbedtls -I../amazon-freertos/vendors/espressif/esp-idf/components/fatfs/src -I../amazon-freertos/vendors/espressif/esp-idf/components/wear_levelling/include -I../amazon-freertos/vendors/espressif/esp-idf/components/sdmmc/include -I../amazon-freertos/vendors/espressif/esp-idf/components/driver/include -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_ringbuf/include -I../components/tft -I../components/spidriver -I../amazon-freertos/vendors/espressif/esp-idf/components/bt/include -I../amazon-freertos/vendors/espressif/esp-idf/components/bt/bluedroid/api/include/api -I../components/ws2812 -I../amazon-freertos/vendors/espressif/esp-idf/components/esp32/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos_tcpip/tcpip_adapter/include -I../amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/include -I../amazon-freertos/libraries/freertos_plus/standard/freertos_plus_tcp/source/portable/Compiler/GCC -I../amazon-freertos/vendors/espressif/esp-idf/components/esp_event/include -I../amazon-freertos/vendors/espressif/esp-idf/components/log/include -I../amazon-freertos/vendors/espressif/esp-idf/components/efuse/include -I../amazon-freertos/vendors/espressif/esp-idf/components/efuse/esp32/include -I../amazon-freertos/vendors/espressif/esp-idf/components/newlib/platform_include -I../amazon-freertos/vendors/espressif/esp-idf/components/newlib/include -I../amazon-freertos/vendors/espressif/esp-idf/components/vfs/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos/include -I../amazon-freertos/freertos_kernel/include -I../amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/include -I../amazon-freertos/vendors/espressif/boards/esp32/aws_demos/config_files -I../amazon-freertos/vendors/espressif/esp-idf/components/heap/include -I../amazon-freertos/vendors/espressif/esp-idf/components/soc/esp32/include -I../amazon-freertos/vendors/espressif/esp-idf/components/soc/include -I../amazon-freertos/demos/include -I../amazon-freertos/demos/network_manager -I../amazon-freertos/vendors/espressif/esp-idf/components/app_update/include -I../amazon-freertos/vendors/espressif/esp-idf/components/bootloader_support/include -I../amazon-freertos/vendors/espressif/esp-idf/components/pthread/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos_tcpip/ethernet/include -I../amazon-freertos/vendors/espressif/boards/esp32/components/freertos_tcpip/smartconfig_ack/include -I../amazon-freertos/libraries/c_sdk/standard/common/include -I../amazon-freertos/libraries/abstractions/platform/include -I../amazon-freertos/libraries/abstractions/platform/freertos/include -I../amazon-freertos/libraries/abstractions/secure_sockets/include -I../amazon-freertos/libraries/c_sdk/standard/common/include/private -I../amazon-freertos/demos/dev_mode_key_provisioning/include -I../amazon-freertos/libraries/freertos_plus/standard/pkcs11/include -I../amazon-freertos/libraries/abstractions/pkcs11/include -I../amazon-freertos/libraries/freertos_plus/standard/utils/include -I../amazon-freertos/libraries/freertos_plus/standard/crypto/include -I../amazon-freertos/libraries/c_sdk/standard/ble/include -I../amazon-freertos/libraries/c_sdk/standard/serializer/include -I../amazon-freertos/libraries/3rdparty/tinycbor -I../amazon-freertos/libraries/abstractions/ble_hal/include -I../amazon-freertos/libraries/abstractions/wifi/include -I../amazon-freertos/libraries/c_sdk/standard/mqtt/include -I../amazon-freertos/libraries/c_sdk/standard/https/include -I../amazon-freertos/libraries/freertos_plus/aws/ota/include -I../amazon-freertos/libraries/c_sdk/aws/shadow/include -isystem ../amazon-freertos/libraries/3rdparty/pkcs11 -mlongcalls   -mlongcalls -Og -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -std=gnu99 -Wno-old-style-declaration -Wno-maybe-uninitialized -MD -MT amazon-freertos/esp-idf/stockd_m5stack/CMakeFiles/idf_component_stockd_m5stack.dir [... cut ...] s/libraries/freertos_plus/aws/ota/src/http/aws_iot_ota_http.c
../amazon-freertos/libraries/freertos_plus/aws/ota/src/http/aws_iot_ota_http.c:43:40: fatal error: aws_iot_ota_agent_internal.h: No such file or directory
compilation terminated.
[8/107] Building C object amazon-freertos/esp-idf/stockd_m5stack/CMakeFiles/idf_comp..._/__/amazon-freertos/libraries/freertos_plus/aws/ota/src/mqtt/aws_iot_ota_cbor.c.ob
ninja: build stopped: subcommand failed.```

BTW i sent you invite to private repository

aggarg, 2019-12-30:
@chegewara Does it work for you after applying the fix as you mentioned?

Thanks.

chegewara, 2019-12-30:
Hi,
that fix works with new project and with my project that does not have additional components. 
With project that has additional components i am having such weird errors so im embarrassed to post logs. Im getting this:
```
../main/main.c:177:9: error: implicit declaration of function 'SYSTEM_Init' [-Werror=implicit-function-declaration]
     if( SYSTEM_Init() == pdPASS )
```
and errors about implicit declarations BT functions from esp-idf, which i dont understand.

aggarg, 2019-12-30:
```SYSTEM_Init``` is defined here: https://github.com/aws/amazon-freertos/blob/31c4d9b5175a88c0578337cf63b3ce85a96a2e5f/libraries/freertos_plus/standard/utils/include/iot_system_init.h#L33

Does your ```main.c``` not include ```iot_system_init.h```? If it does, can you get the preprocessed file to confirm that the declaration of this function does exist in the preprocessed file?

Thanks.

chegewara, 2019-12-31:
Thanks, i dont know why i didnt have that header and BT errors also gone when i uncomment out headers (why i did have it commented out?).

Now all is compiling to the end, but cant link my components:
```

/home/chegewara/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lstockd_m5stack
/home/chegewara/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lspidriver
/home/chegewara/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lws2812
/home/chegewara/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -ltftspi
/home/chegewara/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -ltft
/home/chegewara/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lble
```
All components are included in CMakeLists.txt:
```
get_filename_component(
    EXTRA_COMPONENT_DIRS
    "components/spidriver" ABSOLUTE
)
list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})

get_filename_component(
    EXTRA_COMPONENT_DIRS
    "components/stockd_m5stack" ABSOLUTE
)
list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})
get_filename_component(
    EXTRA_COMPONENT_DIRS
    "components/tft" ABSOLUTE
)
list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})
get_filename_component(
    EXTRA_COMPONENT_DIRS
    "components/ws2812" ABSOLUTE
)
list(APPEND IDF_EXTRA_COMPONENT_DIRS ${EXTRA_COMPONENT_DIRS})
```

aggarg, 2019-12-31:
Pick one of the component, say spidriver, and check if it is being built. This will help to isolate whether:

* The component itself is not built.
* Or the linker is not able to find the object/binary to link.

Thanks.

chegewara, 2019-12-31:
It seems that all components are build, at least all libNAME.a are in place in build/subfolders.

tgsong, 2019-12-31:
@chegewara Sorry for getting back to you late, I didn't see your invite to the repo in my email. Can you check again?

If you can find the static libraries containing those functions, then it's somehow not linked to your application code. Theoretically ESP-IDF does this automatically, so this shouldn't happen. Did you manually specify any of those components in `target_link_libraries`?

chegewara, 2019-12-31:
> @chegewara Sorry for getting back to you late, I didn't see your invite to the repo in my email. Can you check again?

No worries about that.

> Did you manually specify any of those components in target_link_libraries?

Indeed, i have all libraries in included and this was the last one problem, now my project is building.

Thanks for all your support and sorry for being so lame.

closed 2019-12-31 by chegewara

chegewara, 2019-12-31:
To make full image of building AFR as component, i am trying to add `sdkconfig.defaults` to my build system, but adding it in project top folder seems to not work. Any idea?

tgsong, 2020-01-03:
@chegewara Have you checked this? https://github.com/aws/amazon-freertos/issues/832#issuecomment-514756608 and related PR https://github.com/aws/amazon-freertos/pull/987. There were some discussions at that time about `sdkconfig`. But looks like I forgot to add it into our document, I'll provide this to our doc team.

chegewara, 2020-01-03:
Thanks, it works perfectly. 
In addition i can say that partition-table.csv can be put in project top folder without changing in menuconfig. https://github.com/aws/amazon-freertos/issues/832#issuecomment-516236030
> Edit: In the custom sdkconfig it's now necessary to specify the partition table, e.g.:

tgsong, 2020-01-03:
Yes, you can make it pointing to any location in your own sdkconfig file.

kaizio, 2020-12-28:
This is still an issue. Not in documentation about making sure to git init and commit [getting_started_espressif_cmake_project ](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html#getting_started_espressif_cmake_project) had me pulling my hair out for a couple hours.