aws/amazon-freertos #1656: Conflicting errno.h in include path when using LwIP

cypress-neil, 2019-12-20:
**Describe the bug**

libraries/3rdparty/lwip/src/include/lwip should not be in the include path because it contains a file named errno.h that conflicts with the standard C errno.h.
Instead, the include path should only contain libraries/3rdparty/lwip/src/include. Source files that include unqualified lwip header files need to be updated. Of the in-tree targets it looks like only cypress and mediatek uses 3rdparty::lwip.

For GCC, the Cypress build scripts specify LWIP_ERRNO_STDINCLUDE so libraries/3rdparty/lwip/src/include/lwip/errno.h attempts to include the compiler-provided errno.h via <errno.h> but when libraries/3rdparty/lwip/src/include/lwip is in the include path it will just re-include libraries/3rdparty/lwip/src/include/lwip/errno.h instead.

This was fixed as part of https://github.com/aws/amazon-freertos/pull/1526 but it was re-added in the 201912.00 version:
https://github.com/aws/amazon-freertos/blame/master/libraries/3rdparty/CMakeLists.txt#L89

**System information**
- Project: CY8CPROTO_062_4343W (out of tree)
- IDE: CMake 3.15/gnu-arm-embedded 7-2018-q2-update
- Operating System: Linux
- Version of Amazon FreeRTOS: 201912.00
- Project: aws_demos

**Screenshots or console output**
```
$ ninja
[0/2] Re-checking globbed directories...
[16/258] Building C object CMakeFiles/aws_demos.dir/vendors/cypress/clib_support/TOOLCHAIN_GCC_ARM/cy_newlib_freertos.c.obj
FAILED: CMakeFiles/aws_demos.dir/vendors/cypress/clib_support/TOOLCHAIN_GCC_ARM/cy_newlib_freertos.c.obj
/home/tester/.cache/cydep/gcc-7.2.1-1.0.0.1-linux/bin/arm-none-eabi-gcc -DCY8C624ABZI_D44 -DCYBSP_WIFI_CAPABLE -DCY_RTOS_AWARE -DCY_USE_LWIP -DCY_USING_HAL -DLWIP_ERRNO_STDINCLUDE -DLWIP_TIMEVAL_PRIVATE=0 -DTARGET_CY8CPROTO_062_4343W -I../libraries/freertos_plus/standard/utils/include -I../libraries/abstractions/secure_sockets/include -I../libraries/c_sdk/standard/common/include/private -I../freertos_kernel/include -I../vendors/cypress/boards/CY8CPROTO_062_4343W/aws_demos -I../vendors/cypress/boards/CY8CPROTO_062_4343W/aws_demos/application_code -I../vendors/cypress/boards/CY8CPROTO_062_4343W/aws_demos/application_code/cy_code -I../vendors/cypress/boards/CY8CPROTO_062_4343W/aws_demos/application_code/cy_code/GeneratedSource -I../vendors/cypress/boards/CY8CPROTO_062_4343W/aws_demos/config_files -I../vendors/cypress/clib_support -I../vendors/cypress/clib_support/TOOLCHAIN_GCC_ARM -I../vendors/cypress/psoc6 -I../vendors/cypress/psoc6/common -I../vendors/cypress/psoc6/mw -I../vendors/cypress/psoc6/mw/capsense -I../vendors/cypress/psoc6/mw/emeeprom -I../vendors/cypress/psoc6/mw/objstore -I../vendors/cypress/psoc6/psoc6csp -I../vendors/cypress/psoc6/psoc6csp/abstraction -I../vendors/cypress/psoc6/psoc6csp/abstraction/resource -I../vendors/cypress/psoc6/psoc6csp/abstraction/resource/include -I../vendors/cypress/psoc6/psoc6csp/abstraction/rtos -I../vendors/cypress/psoc6/psoc6csp/abstraction/rtos/include -I../vendors/cypress/psoc6/psoc6csp/abstraction/rtos/include/COMPONENT_FREERTOS -I../vendors/cypress/psoc6/psoc6csp/core_lib -I../vendors/cypress/psoc6/psoc6csp/core_lib/include -I../vendors/cypress/psoc6/psoc6csp/hal -I../vendors/cypress/psoc6/psoc6csp/hal/include -I../vendors/cypress/psoc6/psoc6csp/hal/include/pin_packages -I../vendors/cypress/psoc6/psoc6pdl -I../vendors/cypress/psoc6/psoc6pdl/cmsis -I../vendors/cypress/psoc6/psoc6pdl/cmsis/include -I../vendors/cypress/psoc6/psoc6pdl/devices -I../vendors/cypress/psoc6/psoc6pdl/devices/include -I../vendors/cypress/psoc6/psoc6pdl/devices/include/ip -I../vendors/cypress/psoc6/psoc6pdl/drivers -I../vendors/cypress/psoc6/psoc6pdl/drivers/include -I../vendors/cypress/whd -I../vendors/cypress/whd/inc -I../vendors/cypress/whd/resources -I../vendors/cypress/whd/resources/firmware -I../vendors/cypress/whd/resources/firmware/COMPONENT_4343W -I../vendors/cypress/whd/resources/nvram -I../vendors/cypress/whd/resources/nvram/COMPONENT_CY8CPROTO_062_4343W -I../vendors/cypress/whd/resources/resource_imp -I../vendors/cypress/whd/src -I../vendors/cypress/whd/src/bus_protocols -I../vendors/cypress/whd/src/include -I../freertos_kernel/portable/GCC/ARM_CM4F -I../demos/include -I../libraries/freertos_plus/standard/crypto/include -I../libraries/3rdparty/mbedtls/include -I../libraries/3rdparty/mbedtls/include/mbedtls -I../libraries/abstractions/pkcs11/mbedtls -I../libraries/abstractions/wifi/include -I../libraries/c_sdk/standard/common/include -I../libraries/3rdparty/lwip/src/include -I../libraries/3rdparty/lwip/src/include/lwip -I../demos/network_manager -I../libraries/abstractions/platform/include -I../libraries/abstractions/platform/freertos/include -I../demos/dev_mode_key_provisioning/include -I../libraries/freertos_plus/standard/pkcs11/include -I../libraries/abstractions/pkcs11/include -I../libraries/c_sdk/aws/defender/include -I../libraries/c_sdk/standard/mqtt/include -I../libraries/c_sdk/standard/serializer/include -I../libraries/3rdparty/tinycbor -I../libraries/freertos_plus/aws/greengrass/include -I../libraries/c_sdk/standard/https/include -I../libraries/c_sdk/aws/shadow/include -isystem ../vendors/cypress/lwip -isystem ../libraries/3rdparty/lwip_osal/include -isystem ../vendors/cypress/boards/CY8CPROTO_062_4343W/ports/wifi -isystem ../libraries/3rdparty/pkcs11 -mcpu=cortex-m4 -Os -mfloat-abi=softfp -mfpu=fpv4-sp-d16 -mthumb -ffunction-sections -fdata-sections -funsigned-char -fno-exceptions -g -Wall -fdiagnostics-color=always -MD -MT CMakeFiles/aws_demos.dir/vendors/cypress/clib_support/TOOLCHAIN_GCC_ARM/cy_newlib_freertos.c.obj -MF CMakeFiles/aws_demos.dir/vendors/cypress/clib_support/TOOLCHAIN_GCC_ARM/cy_newlib_freertos.c.obj.d -o CMakeFiles/aws_demos.dir/vendors/cypress/clib_support/TOOLCHAIN_GCC_ARM/cy_newlib_freertos.c.obj   -c ../vendors/cypress/clib_support/TOOLCHAIN_GCC_ARM/cy_newlib_freertos.c
../vendors/cypress/clib_support/TOOLCHAIN_GCC_ARM/cy_newlib_freertos.c: In function '_sbrk':
../vendors/cypress/clib_support/TOOLCHAIN_GCC_ARM/cy_newlib_freertos.c:51:17: error: 'ENOMEM' undeclared (first use in this function)
         errno = ENOMEM;
                 ^~~~~~
../vendors/cypress/clib_support/TOOLCHAIN_GCC_ARM/cy_newlib_freertos.c:51:17: note: each undeclared identifier is reported only once for each function it appears in
[21/258] Building C object CMakeFiles/aws_demos.dir/vendors/cypress/psoc6/common/cybsp_wifi.c.obj
ninja: build stopped: subcommand failed.
```

**To reproduce**

https://gist.githubusercontent.com/cypress-neil/2f088e680902030ece68c4d8a4fae003 contains some unreleased changes to build with LWIP_ERRNO_STDINCLUDE and fix some 201912.00 compatibility issues.

I used version 0f698ccd96749dac499fd4e41e9cdbaee6668916 of the https://github.com/cypresssemiconductorco/amazon-freertos repository.

```
$ curl -o lwip.patch https://gist.githubusercontent.com/cypress-neil/2f088e680902030ece68c4d8a4fae003/raw/a0bdf651cccecf479764b902ebc25019abd8c5cc/gistfile1.txt
$ git clone https://github.com/cypresssemiconductorco/amazon-freertos.git cypress
$ git clone https://github.com/aws/amazon-freertos.git
$ cd amazon-freertos
$ git checkout 201912.00
$ cp -r ../cypress/vendors/cypress/* vendors/cypress/
$ patch -p0 -i ../lwip.patch
$ mkdir build && cd build
$ cmake -G Ninja -DVENDOR=cypress -DBOARD=CY8CPROTO_062_4343W -DCOMPILER=arm-gcc ..
$ ninja
```

qiutongs, 2019-12-24:
Sorry about the compilation regression. I agree we should remove libraries/3rdparty/lwip/src/include/lwip from include path. 

Some background first. This issue happened in this https://github.com/aws/amazon-freertos/pull/1624 , which was to fix compilation issue for Mediatek board.  Specifically, this line of Mediatek code requires that: https://github.com/aws/amazon-freertos/blob/master/vendors/mediatek/sdk/middleware/MTK/dhcpd/inc/dhcpd.h#L42

Actually, when this https://github.com/aws/amazon-freertos/pull/1526 merged, the Mediateck project should have failed compiling but it didn't because (1) PR 1526 doesn't update project file for Mediateck and (2) we don't have auto CMake build check for Mediateck. That's why the problem to Mediateck was caught so late.

We need to fix dhcpd.h before removing libraries/3rdparty/lwip/src/include/lwip.

aggarg, 2019-12-31:
Fixed with this PR: https://github.com/aws/amazon-freertos/pull/1660

closed 2019-12-31 by aggarg