aws/amazon-freertos #1686: POSIX layer is not well developed

edbek, 2020-01-11:
POSIX layer is not well developed. Look at how TI has it:

http://software-dl.ti.com/simplelink/esd/simplelink_msp432_sdk/2.40.00.10/docs/tiposix/Users_Guide.html#summary-of-functions-supported-in-ti-posix

sarenameas, 2020-01-13:
Thank you for this feedback. We will take into consideration improvements to the POSIX layer and related items and prioritize it.

sarenameas, 2020-01-14:
Hello edbk, 

By the way, are there any specific functions or variables that you looking for?

The https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/v4_beta uses a posix implementation for their Linux port. It may be helpful to check out.

edbek, 2020-01-16:
that's what I would like to see:

variables: 
**pthread_key_t;**

functions: 
**pthread_getspecific();
pthread_setspecific();
pthread_key_create();
pthread_setcancelstate();**

alfred2g, 2020-01-21:
Hi @edbek 

Thanks for your interest in Amazon FreeRTOS!

We will look into adding your feature request of adding thread local storage Api functions to our POSIX API support.
I will be closing this issue, and if you have any question or concern please reopen the issue

Thanks

yuhui-zheng, 2020-01-21:
Closing this issue on behalf of @alfred2g.

closed 2020-01-21 by yuhui-zheng

edbek, 2020-01-28:
Well, it would be nice to implement a set of Linux functions epoll.