aws/amazon-freertos #1687: PC/Win demo shows asserts and crashes

htibosch, 2020-01-13:
**Describe the bug**
PC/Win demo shows asserts and crashes. Ofen a queue function is called while the scheduler is disabled.

**System information**

- Which hardware board or part numbers?
Asus laptop, i7

- IDE used
VS 2017 with SDK version 10.0.17763.0

- Operating System [Windows|Linux|MacOS]
W10

- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
Today's version, cloned from here

- Project
pc - demo
pc - test

**Expected behavior**
No asserts or crashes

**Screenshots or console output**
One example is from `xQueueSemaphoreTake()`:

~~~c
configASSERT( !( ( xTaskGetSchedulerState() == taskSCHEDULER_SUSPENDED ) &&
                 ( xTicksToWait != 0 ) ) );
~~~

What I think is that despite the scheduler was suspended, and this (asserting) task is still running.

**To reproduce**
Steps to reproduce the behavior:

Start the demo one or more times, with or without enabling `ipconfigUSE_DHCP`.

**Additional context**
I saw this kind of behaviour in the PC/Win before: it may happen when a FreeRTOS task calls directly a blocking Windows API, like e.g. Sleep(), or any I/O call ( `printf()` ).

Would one of the tracing functions call API's of the Windows?

Thanks, Hein

sarenameas, 2020-01-14:
Hi Hein,

>Project
pc - demo
pc - test

Which demo are you using? Is it the default MQTT demo? 

>Would one of the tracing functions call API's of the Windows?

Also I do not believe that there are any tracealyzer functions being invoked by default. Are these the functions you are speaking of here?

htibosch, 2020-01-14:
Hi Sarena, I mostly tested with :

`projects/pc/windows/visual_studio/aws_demos/aws_demos.sln`

It rarely finishes the demo, because it hangs somewhere, it asserts, or it crashes.

I worked from scratch:
- git clone
- Change the SDK version in VS
- Adapted `configNETWORK_INTERFACE_TO_USE`
- Changed `aws_clientcredential.h` and `aws_clientcredential_keys.h`

Is anything else needed?

sarenameas, 2020-01-15:
I made a fresh clone of the latest release: https://github.com/aws/amazon-freertos/releases/tag/201912.00 and using VS2019 and SDK v10.0.17763.0, and I could not reproduce the issue.
I tried the latest release with VS2017 and SDK v8.1, and I could not reproduce the issue.
Could you post the logs when it crashes or hangs?

htibosch, 2020-01-15:
Thanks a lot for doing all this work, really appreciated!
I will try out your latest release, install VS2019, and also I will run the demo on my earlier laptop. And after that I will report back here. Hein

htibosch, 2020-01-15:
Good news: the problem does not occur on an older W7 laptop, borrowed from my son:

~~~
Windows 7 Ultimate, Service Pack 1
CPU : i7-2630QM @ 2.00 GHz
RAM 8GB, 64 bits OS.
~~~

I tried it 10 times, and each time the demo finished correctly. Glad to see that.

The problem does occur when I use my own laptop:

~~~
Windows 10 Home
CPU : i7-6700HQ @ 2.60 GHz
RAM 16GB, 64 bits OS.
~~~

My laptop has a different OS version and the same type of CPU, only a bit faster.

When debugging the demo, I observed that a task was running while another task had suspended the FreeRTOS scheduler.

Scheduling in WinSim happens here in `prvProcessSimulatedInterrupts()`:

~~~c
    SuspendThread( previous_task ); // always returns 0, correct
    ResumeThread( new_task ); // always returns 1, also good
~~~

Although both functions return a correct value, I see that 'previous_task' tries to call `vTaskDelay()` while the scheduler is not running. In stead, 'previous_task' should be suspended.

There is some randomness: sometimes it crashes after a second, sometimes it runs for 3 seconds. The cause is probably always the same.

Does this ring a bell to anyone?

gordonwang0, 2020-01-27:
Hi,

We have tried reproducing this on several different Windows machines, and weren't able to reproduce.  As we can't reproduce this, there isn't any action we can take.  I am closing this issue; please let me know if you have any questions.

closed 2020-01-27 by gordonwang0

htibosch, 2020-01-28:
Hello Gordon, it seems to be that the problem only occurs on some hardware, or on some versions of Windows.
@RichardBarry wrote me that more users have reported the same type of problems.
Both Richard and I worked on finding a solution. The result is a new port that will be merged with this Github repository.

For those who are experiencing this instability:

I used [this port](https://sourceforge.net/p/freertos/code/HEAD/tree/trunk/FreeRTOS/Source/portable/MSVC-MingW).

beside that, I made a change to task.c :

Here is my small change, it is adding 4 lines of code:

Add to the port's header file:

~~~c
+extern volatile BaseType_t xInsideInterrupt;
+#define portWAIT_FOR_INTERRUPT() \
+    while( xInsideInterrupt != 0 )
~~~

Add to task.c :

~~~c
 void vTaskSuspendAll( void )
 {
     /* A critical section is not required as the variable is of type
     BaseType_t.  Please read Richard Barry's reply in the following link to a
     post in the FreeRTOS support forum before reporting this as a bug! -
     http://goo.gl/wu4acr */
+    portWAIT_FOR_INTERRUPT();
     ++uxSchedulerSuspended;
     portMEMORY_BARRIER();
 }
~~~

**Without** the above change the application will hang quickly: `uxSchedulerSuspended` becomes non-zero while a different task is running, causing a total dead-lock.

~~~c
>    while( xInsideInterrupt != 0 )
~~~

I hope this helps other users who are not able to run the current PC/Win port.

Discussion: I am having doubts if the `SetThreadAffinityMask()` still behaves as it should.

Hein

htibosch, 2020-02-13:
FreeRTOS V10.3.0 is out, and it contains the reparation for the above mentioned issue.

I tested it with the PC project and it worked well: no more exceptions or asserts.

It gave one compilation error in:

	libraries/3rdparty/tracealyzer_recorder/Include/trcKernelPort.h

because `uxPendedTicks` is unknown to my compiler.

Hein