aws/amazon-freertos #1688: Use ESP32-Wifi-Manager with AWS FreeRTOS

elieDaan, 2020-01-14:
Hello, 

I'm trying to use [esp32-wifi-manager](https://github.com/tonyp7/esp32-wifi-manager) with AWS FreeRTOS. Thanks to the AWS team, LWIP and esp-http-server is now functioning on AFR. 

I'm able to connect with Access point to the ESP32 then it opens a browser window with all wifi network around like this [video](https://www.youtube.com/watch?v=hxlZi15bym4). 

But i's impossible to use this wifi parameters to connect the esp32 to AWS IOT. 

When I've got a wifi connection I'm creating the Iot Task : 
```c
void cb_connection_ok(void *pvParameter){
     /* Initialize mqttDemoContext */
    static demoContext_t mqttDemoContext =
    {
        .networkTypes                  = AWSIOT_NETWORK_TYPE_WIFI,
        .demoFunction                  = RunMqttBob,
        .networkConnectedCallback      = NULL,
        .networkDisconnectedCallback   = NULL
    };
	ESP_LOGI(TAG, "I have a connection!");

    
    Iot_CreateDetachedThread( task_test_mqtt,
                              &mqttDemoContext,
                              democonfigDEMO_PRIORITY,
         
```
But this is the terminal output : 

I (6775) main: I have a connection!

 END AwsIotNetworkManager_Init 
I (6785) BTDM_INIT: BT controller compile version [9891bc6]
1 673 [iot_thread] [INFO ][INIT][6730] SDK successfully initialized.
I (6795) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
GAP procedure initiated: stop advertising.
GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=480 adv_itvl_max=960
E (7095) WIFI: WIFI_On: Failed to init event loop -1
2 704 [iot_thread] [ERROR][DEMO][7040] Failed to intialize all the networks configured for the device.
3 704 [iot_thread] [INFO ][INIT][7040] SDK cleanup done.

Thanks for your help

sarenameas, 2020-01-14:
Hello @elieDaan,

Does *task_test_mqtt()* match this function https://github.com/aws/amazon-freertos/blob/master/demos/demo_runner/iot_demo_freertos.c#L309? From the logs it appears so.

There is no point in using the **AwsIotNetworkManager** at the same time as another network manager (esp32-wifi-manager), because they may try to reconnect at the same time if a network down even it detected.  In the initialization sequence https://github.com/aws/amazon-freertos/blob/master/demos/demo_runner/iot_demo_freertos.c#L200, the AwsIotNetworkManager, registers to receive state changes in wifi, and tries to turn the wifi on (which it already is on in your example, so the failure is here).

The easiest thing to do to with your current code is to delete lines 226 through 281 here https://github.com/aws/amazon-freertos/blob/master/demos/demo_runner/iot_demo_freertos.c#L226. There will be minor syntax fixes you need to make after doing this. This will enable the network manager only (with function AwsIotNetworkManager_Init()), but not let it mess with your existing wifi connection. 

If you still have a strong desire to use esp32-wifi-manager, for a cleaner application you will need to remove AwsIotNetworkManager completely.

elieDaan, 2020-01-16:
Hello @sarenameas ,

task_test_mqtt is based on this function[ ]( https://github.com/aws/amazon-freertos/blob/master/demos/demo_runner/iot_demo_freertos.c#L309).  

I've commented the lines suggested in my initialize function but it's still not working and always rebooting.

I've got this terminal output : 
 
I (6482) main: I have a connection!

 END AwsIotNetworkManager_Init 
/** START RUN MQTT BOB 
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:

I've found that if in my __establishMqttConnection_ like [iot_demo_mqtt](https://github.com/aws/amazon-freertos/blob/master/demos/mqtt/iot_demo_mqtt.c#L393), I comment the IotMqtt_Connect
```c
connectStatus = IotMqtt_Connect( &networkInfo,
                                         &connectInfo,
                                         MQTT_TIMEOUT_MS,
                                         pMqttConnection );
```c

I will get this terminal output : 
I (7375) main: I have a connection!

 END AwsIotNetworkManager_Init 
/** START RUN MQTT BOB 
 **/2 707 [iot_thread] [INFO ][INIT][7070] SDK successfully initialized.
3 707 [iot_thread] [INFO ][DEMO][7070] Successfully initialized the demo. Network type for the demo: 0
4 707 [iot_thread] [INFO ][MQTT][7070] MQTT library successfully initialized.
5 707 [iot_thread] [INFO ][DEMO][7070] MQTT demo client identifier is Blablablaaa(length 11).
6 707 [iot_thread] [ERROR][DEMO][7070] MQTT CONNECT returned error PENDING.

elieDaan, 2020-01-17:
UPDATE : 

I've got a new terminal output after uncomment some pieces of initialize. 

My _initialize: 
```c
static int _initialize( demoContext_t * pContext )
{
    int status = EXIT_SUCCESS;
    bool commonLibrariesInitialized = false;
    bool semaphoreCreated = false;

    /* Initialize common libraries required by network manager and demo. */
    if( IotSdk_Init() == true )
    {
        commonLibrariesInitialized = true;
    }
    else
    {
        IotLogInfo( "Failed to initialize the common library." );
        status = EXIT_FAILURE;
    }
   
    if( status == EXIT_SUCCESS )
    {
        if( AwsIotNetworkManager_Init() != pdTRUE )
        {
            IotLogError( "Failed to initialize network manager library." );
            status = EXIT_FAILURE;
        }
    }

    printf("\r\n END AwsIotNetworkManager_Init \r\n");

    #if 1
    if( status == EXIT_SUCCESS )
    {
        /* Create semaphore to signal that a network is available for the demo. */
        if( IotSemaphore_Create( &demoNetworkSemaphore, 0, 1 ) != true )
        {
            IotLogError( "Failed to create semaphore to wait for a network connection." );
            status = EXIT_FAILURE;
        }
        else
        {
            semaphoreCreated = true;
        }
    }
    #endif
   #if 0
    if( status == EXIT_SUCCESS )
    {
        /* Subscribe for network state change from Network Manager. */
        if( AwsIotNetworkManager_SubscribeForStateChange( pContext->networkTypes,
                                                          _onNetworkStateChangeCallback,
                                                          pContext,
                                                          &subscription ) != pdTRUE )
        {
            IotLogError( "Failed to subscribe network state change callback." );
            status = EXIT_FAILURE;
        }
   

    }
   #endif

      /*** Disenable wifi initialize because it's wifi manager now ***/
    /* Initialize all the  networks configured for the device. */
     #if 0
    if( status == EXIT_SUCCESS )
[... 162 lines cut ...]
 END AwsIotNetworkManager_Init 

 End initialize, status :0
2 620 [iot_thread] [INFO ][INIT][6200] SDK successfully initialized.
3 620 [iot_thread] [INFO ][DEMO][6200] Successfully initialized the demo. Network type for the demo: 1
/** START RUN MQTT BOB 
 **//** INIT MQTT OK 
 **/4 820 [iot_thread] [INFO ][MQTT][8200] MQTT library successfully initialized.
5 820 [iot_thread] [INFO ][DEMO][8200] M../amazon-freertos/freertos_kernel/queue.c:1425 (xQueQTT demo client ueSemaphoreTake)- assert failed!
abort() wasidentifier is Bob_iot_v42 (length 11).
 called at PC 0x4008b82b on core 0

The backtrace decoding is : 
Decoding stack results
0x4008e0f8: invoke_abort at ../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c line 156
0x4008e325: abort at ../amazon-freertos/vendors/espressif/esp-idf/components/esp32/panic.c line 171
0x4008b82b: xQueueSemaphoreTake at ../amazon-freertos/freertos_kernel/queue.c line 1425
0x400ebe05: WIFI_GetHostIP at ../amazon-freertos/vendors/espressif/boards/esp32/ports/wifi/iot_wifi.c line 1111
0x40138851: SOCKETS_GetHostByName at ../amazon-freertos/vendors/espressif/boards/esp32/ports/secure_sockets/lwip/iot_secure_sockets.c line 817
0x400ec648: IotNetworkAfr_Create at ../amazon-freertos/libraries/abstractions/platform/freertos/iot_network_freertos.c line 370
0x400ede17: IotMqtt_Connect at ../amazon-freertos/libraries/c_sdk/standard/mqtt/src/iot_mqtt_api.c line 980
0x400e9949: _establishMqttConnection at ../components/XX.c line 2201
0x400e9be3: RunMqttBob at ../components/XX.c line 2303
0x400ea615: task_test_mqtt at ../components/XX.c line 2464
0x400ec045: _threadRoutineWrapper at ../amazon-freertos/libraries/abstractions/platform/freertos/iot_threads_freertos.c line 85

dachalco, 2020-01-24:
Hi @elieDaan 

Frame 3 **(0x400ebe05 : WIFI_GetHostIP()** raises some flags. The call attempts to take `xWiFiSem` but aborts because it's null -- see Frame 2 **(0x4008b82b: xQueueSemaphoreTake)**. There's a few ways `xWiFiSem` can be created. Here's how a stack trace of its initial creation in the demo looks:

0: xWiFiSem = xSemaphoreCreateMutexStatic
1: WIFI_On 
2: _wifiEnable 
3: AwsIotNetworkManager_EnableNetwork 
4: _initialize 
5: runDemoTask

In your `_initialize` snippet above, this path would not be executed since it's excluded by preprocessor macro:

```c   
#if 0
    if( status == EXIT_SUCCESS )
    {
        if( AwsIotNetworkManager_EnableNetwork( configENABLED_NETWORKS ) != configENABLED_NETWORKS )
        {
            IotLogError( "Failed to intialize all the networks configured for the device." );
            status = EXIT_FAILURE;
        }
          
    }
#endif
```

Given you're trying to replace the Network Manager with the `esp32-wifi-manager`, you should take care to ensure all necessary initialization is still completed in the correct order. It looks like you will have to take some of the code in function `AwsIotNetworkManager_EnableNetwork`, such as initializing this mutex, and move it into the function you are writing to replace this. Best of luck!

qiutongs, 2020-02-10:
If there is still question or problem, feel free to re-open this issue or create a new one.

closed 2020-02-10 by qiutongs

latonita, 2020-02-12:
@elieDaan did you manage to make it up and running?

elieDaan, 2020-02-12:
@latonita No sorry I don't have the time to work on it until March..

hackaprende, 2021-01-24:
Same here, I am getting this and I am not sure what is the cause, I double checked my wifi credentials and ssid
I (649) wifi:wifi firmware version: 1afdae0
I (649) wifi:config NVS flash: enabled
I (649) wifi:config nano formating: disabled
I (649) wifi:Init dynamic tx buffer num: 32
I (649) wifi:Init data frame dynamic rx buffer num: 32
I (659) wifi:Init management frame dynamic rx buffer num: 32
I (659) wifi:Init management short buffer num: 32
I (669) wifi:Init static rx buffer size: 1600
I (669) wifi:Init static rx buffer num: 10
I (669) wifi:Init dynamic rx buffer num: 32
4 3163 [iot_thread] [ERROR][DEMO][31630] Failed to initialize all the networks configured for the device.
5 3163 [iot_thread] [INFO ][INIT][31630] SDK cleanup done.
6 3163 [iot_thread] [ERROR][DEMO][31630] Failed to initialize the demo. exiting...
7 3163 [iot_thread] [INFO ][DEMO][31630] -------DEMO FINISHED-------

elieDaan, 2021-01-27:
It was a struggle to get it working with AFR. And still now it's not reliable, you've got a lot of issues with the esp_http_server. 

You need to kill the wifi manager task before running your AFR wifi initialize and make AFR wifi initialise reading your wifi credentials to init the network.

dachalco, 2021-01-27:
Hi All,

Yes as the `esp_http_server` and our Wifi Manager rely on the same TCPIP stack, caution must be taken to make sure they don't step on eachother.

@hackaprende 
Are you running on of our demos, using the `NetworkManager`, or our WiFI API? Could you give a rough sequence of how you set things up -- better yet a code snippet (if its custom).

monhack, 2021-03-16:
Hi @elieDaan,

have you managed to add esp32-wifi-manager to freertos?