aws/amazon-freertos #1690: Build executable outside of AmazonFreeRTOS tree

HuyDuongA, 2020-01-14:
I'm following Getting Started guide section Using Amazon FreeRTOS in Your Own CMake Project for ESP32. 

Up until I run `cmake --build build`, I have this error:

`[ 90%] Linking CXX executable my_app
CMakeFiles/my_app.dir/AmazonFreeRTOS/vendors/espressif/boards/esp32/ports/ble/bluedroid/iot_ble_hal_common_gap.c.obj: In function 'prvBTManagerInit':
/home/huy/Documents/GitHub/GardenBuddy_FW/AmazonFreeRTOS/vendors/espressif/boards/esp32/ports/ble/bluedroid/iot_ble_hal_common_gap.c:517: multiple definition of 'prvBTManagerInit'
CMakeFiles/my_app.dir/AmazonFreeRTOS/vendors/espressif/boards/esp32/ports/ble/iot_ble_hal_common_gap.c.obj:/home/huy/Documents/GitHub/GardenBuddy_FW/AmazonFreeRTOS/vendors/espressif/boards/esp32/ports/ble/bluedroid/iot_ble_hal_common_gap.c:517: first defined here
CMakeFiles/my_app.dir/AmazonFreeRTOS/vendors/espressif/boards/esp32/ports/ble/bluedroid/iot_ble_hal_common_gap.c.obj: In function 'prvBTGetAllDeviceProperties':
iot_ble_hal_common_gap.c:(.text.prvBTGetAllDeviceProperties+0x0): multiple definition of 'prvBTGetAllDeviceProperties'
CMakeFiles/my_app.dir/AmazonFreeRTOS/vendors/espressif/boards/esp32/ports/ble/iot_ble_hal_common_gap.c.obj:iot_ble_hal_common_gap.c:(.text.prvBTGetAllDeviceProperties+0x0): first defined here
`

sarenameas, 2020-01-14:
Hello HuyDuongA,

>`multiple definition of 'prvBTManagerInit'` 

It appears that two files with the same function definitions are part of the build. Are building your own copy ESP-IDF? The ESP-IDF is already included with Amazon FreeRTOS, so you will want to make sure that you build the files only from there. 

Let me know if this help.

HuyDuongA, 2020-01-16:
I'm not building my own copy of the ESP-IDF. I'm new to AmazonFreeRTOS, so I just follow the getting started guide instructions. Before this problem, I successfully built MQTT application inside the AmazonFreeRTOS folder, but once I try to have AmazonFreeRTOS as external library, errors start to appear. I notice that there are `iot_ble_hal_common_gap.c  iot_ble_hal_gap.c  iot_ble_hal_gatt_server.c` inside `AmazonFreeRTOS/vendors/espressif/boards/esp32/ports/ble/` and also in `AmazonFreeRTOS/vendors/espressif/boards/esp32/ports/ble/bluedroid/`. But they are dependent on each other, so I think they shouldn't be duplicated. That's the impression I get from the error message.

dachalco, 2020-02-03:
Hi @HuyDuongA 

I'm having trouble reproducing your issue or retracing the build steps in general. I was able to build a simple app following the **_Using Amazon FreeRTOS in Your Own CMake Project for ESP32_** guide [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html). Our configurations must be different. Are you able to provide more details?

What, if any, modifications were made to the ```CMakeLists.txt``` suggested in the guide above?
Are you attempting an external app similar to ```demos/mqtt/iot_demo_mqtt.c```?

It's okay to have two C files with definitions for the same function, so long as **only one** of the function definitions is linked.

qiutongs, 2020-02-12:
If there is still a problem, feel free to re-open this issue or create a new one.

closed 2020-02-12 by qiutongs