aws/amazon-freertos #1697: Dynamic memory call mapping in AFR crypto layer

shubhamkulkarni97, 2020-01-16:
Dynamic memory calls in AFR crypto layer override other memory call mappings. `calloc` is always mapped to `prvCalloc` and `free` is always mapped to `vPortFree`.

The mapping can be seen at: https://github.com/aws/amazon-freertos/blob/master/libraries/freertos_plus/standard/crypto/src/iot_crypto.c#L254

Platform should provide an ability to override default crypto malloc/free calls to its implementations. It would be great if an option is added to override the default mappings of the calls.

Thank you!

alfred2g, 2020-01-16:
Hi (@shubhamkulkarni97) Shubham,

I will be looking into the issue,
Thanks for noticing and reporting!

Regards

dcgaws, 2020-01-16:
@shubhamkulkarni97 can you please clarify the requirement? In iot_crypto.c, prvCalloc calls pvPortMalloc. The latter, and vPortFree, are already part of the FreeRTOS kernel portable layer.

shubhamkulkarni97, 2020-01-17:
@dcgaws `mbedtls_platform_set_calloc_free` is used to register calloc and free functions related to mbedTLS port layer. Currently, these functions are always mapped to `prvCalloc` and `vPortFree`, which are FreeRTOS specific calloc/free functions and there is no provision to use mbedTLS calloc/free functions.

A wrapper function can be added which has an option to use mbedTLS specific calloc/free functions.

mahavirj, 2020-01-17:
@dcgaws 

To elaborate on context: `pvPortMalloc`/`vPortFree` though platform specific, still are too generic. We provide an ability to control allocations at much finer granularity (e.g. for mbedTLS, allocation strategy can be per https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/components/mbedtls/Kconfig#L7). For that we would like to use wrappers from https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/components/mbedtls/port/esp_mem.c#L22, for all mbedTLS and crypto related allocations. However, right now this gets overridden due to call `CRYPTO_ConfigureHeap`, which if made platform specific will help to achieve our purpose.

dcgaws, 2020-01-17:
@mahavirj yes we'd have to refactor our various library calls to CRYPTO_ConfigureHeap in order to better facilitate this. 

Alternatively, we could use different default static definitions for MBEDTLS_PLATFORM_STD_CALLOC and MBEDTLS_PLATFORM_STD_FREE in Amazon FreeRTOS, similar to what you have in mbedtls\esp_config.h. Then the latter could override the former for the ESP32 projects at compile time. Do you see any problem with that approach?

shubhamkulkarni97, 2020-01-20:
> @mahavirj yes we'd have to refactor our various library calls to CRYPTO_ConfigureHeap in order to better facilitate this.
>
> Alternatively, we could use different default static definitions for [...]

@dcgaws This approach seems good. Using default static definitions for MBEDTLS_PLATFORM_STD_CALLOC and MBEDTLS_PLATFORM_STD_FREE in Amazon FreeRTOS would fix the issue.

qiutongs, 2020-02-11:
We are still working on the fix. Leave this issue open.

sachin0x18, 2020-03-31:
Is there any active development ongoing for this issue ?

https://github.com/aws/amazon-freertos/pull/1713 relies on this issue being fixed

lundinc2, 2020-03-31:
@sachin0x18, 

This has been blocked by an effort to sub module mbedTLS, once that is checked in to the repository I will reopen a pull request that resolves this issue.

I apologize for the lack of transparency regarding this.

lundinc2, 2020-04-10:
@sachin0x18,

I have implemented this, and will open a PR soon. I am close, but one platform is not working yet.

lundinc2, 2020-04-13:
https://github.com/aws/amazon-freertos/pull/1898 implements this, I will go ahead and close this issue. Please reopen it if something was missed.

closed 2020-04-13 by lundinc2