aws/amazon-freertos #1702: thing disconnected by aws because of duplicate client id

sameerdasarwad, 2020-01-18:
i am working switch board where i refer the conveyor demo for my code and it was working fine but now its getting disconnected because of duplicate client id i am not understanding its because of my code or aws iot core issue when i see thing activity its shows that msg 
[image]

closed 2020-01-19 by sameerdasarwad

alfred2g, 2020-01-21:
Hi @sameerdasarwad 

Thanks for your interest in Amazon FreeRTOS.

I guess you already found the issue, you cant use two client IDs.
As per our documentation at https://docs.aws.amazon.com/iot/latest/developerguide/life-cycle-events.html

Thanks,

sameerdasarwad, 2020-01-23:
Hi alfred2g thanks for replay i found what was the issue