aws/amazon-freertos #1704: Issue while running latest version of IDT(IoT Device Tester)

shubhamkulkarni97, 2020-01-20:
**Describe the bug**
While running latest version of IDT (v1.6.0), test-groups fail as it does not get the path to thingPolicy.json. The file is present at the specified path, but the code fails to access the file.

**System information**
- Operating System - MacOS
- Version of Amazon FreeRTOS - 201912.00-43-g9db74d7f7

**Screenshots or console output**
Test execution logs:
```time="2020-01-20T14:34:48+05:30" level=info msg="AFQP Suite Info" hostArch=amd64 hostName=Anuj.localdomain hostOs=darwin
time="2020-01-20T14:34:48+05:30" level=info msg="Getting version file url ..."
time="2020-01-20T14:34:49+05:30" level=info msg="Validation result for test group"
time="2020-01-20T14:34:49+05:30" level=info msg="{\"userData\":{\"sourcePath\":\"\",\"vendorPath\":\"\",\"buildTool\":{\"Command\":\"\"},\"flashTool\":{\"Command\":\"\"},\"clientWifiConfig\":{\"wifiSSID\":\"\",\"wifiPassword\":\"\",\"wifiSecurityType\":\"\"},\"testWifiConfig\":{\"wifiSSID\":\"\",\"wifiPassword\":\"\",\"wifiSecurityType\":\"\"},\"otaConfiguration\":{\"otaFirmwareFilePath\":\"\",\"deviceFirmwareFileName\":\"\",\"awsSignerPlatform\":\"\",\"awsSignerCertificateArn\":\"\",\"awsUntrustedSignerCertificateArn\":\"\",\"awsSignerCertificateFileName\":\"\",\"compileCodesignerCertificate\":\"\"},\"cmakeConfiguration\":{\"boardName\":\"\",\"vendorName\":\"\",\"afrToolchainPath\":\"\",\"compilerName\":\"\",\"cmakeToolchainPath\":\"\",\"cmakeInstallation\":\"\",\"compilerInfo\":\"\"},\"echoserverConfiguration\":{\"securePortForSecureSocket\":\"\",\"insecurePortForSecureSocket\":\"\",\"insecurePortForWiFi\":\"\"}},\"setup\":{\"awsCredential\":{\"accessKey\":\"\",\"secretKey\":\"\",\"region\":\"\"},\"productVersion\":\"\"},\"resource\":{\"privateKeyPath\":\"\",\"username\":\"\",\"ip\":\"\",\"bleDevice\":\"\"}}"
time="2020-01-20T14:35:22+05:30" level=info msg="AFQP Suite Info" hostArch=amd64 hostName=Anuj.localdomain hostOs=darwin
time="2020-01-20T14:35:22+05:30" level=info msg="Getting version file url ..."
time="2020-01-20T14:35:22+05:30" level=info msg="Test setup start ..."
time="2020-01-20T14:35:22+05:30" level=info msg="Copying source code to /Users/shubhamkulkarni/gitrepo/afr_DQP/devicetester_afreertos_mac/c19-7p03dc..."
time="2020-01-20T14:35:33+05:30" level=info msg="Copying source code finished"
time="2020-01-20T14:35:33+05:30" level=info msg="Setting client wifi config ..."
time="2020-01-20T14:35:33+05:30" level=info msg="Applying policy retrieved from /Users/shubhamkulkarni/gitrepo/afr_DQP/devicetester_afreertos_mac/tests/AFQ_1/policies/thingPolicy.json.\n"
time="2020-01-20T14:35:33+05:30" level=error msg="failed to create thing with error open : no such file or directory"
time="2020-01-20T14:35:33+05:30" level=info msg="Cleaning test resources"
time="2020-01-20T14:35:36+05:30" level=error msg="Execution failed due to error: failed to setup environment with error: open : no such file or directory"
```

While the file can be accessed from the terminal using same path:
```
cat /Users/shubhamkulkarni/gitrepo/afr_DQP/devicetester_afreertos_mac/tests/AFQ_1/policies/thingPolicy.json
{
	"Version": "2012-10-17",
	"Statement": [
		{
			"Effect": "Allow",
			"Action": "iot:*",
			"Resource": "*"
		}
	]
```

Thank you!

gordonwang0, 2020-01-27:
Hello @shubhamkulkarni97,

From your logs, it seems that the policy has already been applied but it failed to create IoT Thing.

Could you please provide your device.json, userdata.json and log file for failed testcase?

shubhamkulkarni97, 2020-01-27:
@gordonwang0 Thanks for taking a look. I'm attaching required files for the failed test case.
[idt.zip](https://github.com/aws/amazon-freertos/files/4115592/idt.zip)

gordonwang0, 2020-01-27:
Hi,

According to [the public documentation](https://docs.aws.amazon.com/freertos/latest/userguide/qual-steps.html), IDT currently supports two key provisioning types: Import or Onboard.

You have selected Onboard, however the "secureElementConfig" section is not configured in device.json.

Suggestions:
- If the board does not have secure element, select "Import" for "KeyProvisioning".
- If the board does have secure element, configure secureElementConfig section.

Thank you for bringing this up! I have passed this to the IDT team and they realized the error message is not clear. The IDT has created a task to improve it.

pvyawaha, 2020-02-03:
Hi Shubham,

Please re-open the issue if you have more questions related to this topic or open a new one for other.

closed 2020-02-03 by pvyawaha