aws/amazon-freertos #1705: MQTT echo connected and subscribed to freertos/demos/echo but could not publish

ferrim, 2020-01-21:
Hi,
i am using Infineon XMC4800 IoT connectivity Kit for connecting AWS cloud but I can't publish:

1 7 [Tmr Svc] Write device private key...
2 291 [Tmr Svc] Wi-Fi module initialized. Connecting to AP...
.3 6047 [Tmr Svc] Wi-Fi Connected to AP. Creating tasks which use network...
4 6059 [Tmr Svc] IP Address acquired 192.168.1.106
5 6059 [Tmr Svc] Creating MQTT Echo Task...
6 6060 [MQTTEcho] MQTT echo attempting to connect to xxxxxxxxxxxxxx.iot.eu-central-1.amazonaws.com.
7 9456 [MQTTEcho] MQTT echo connected.
8 9456 [MQTTEcho] MQTT echo test echoing task created.
9 9620 [MQTTEcho] MQTT Echo demo subscribed to freertos/demos/echo
10 9758 [MQTTEcho] ERROR:  Echo failed to publish 'Hello World 0'
.11 14758 [MQTTEcho] ERROR:  Echo failed to publish 'Hello World 1'
.12 19758 [MQTTEcho] ERROR:  Echo failed to publish 'Hello World 2'
.13 24758 [MQTTEcho] ERROR:  Echo failed to publish 'Hello World 3'
.14 29758 [MQTTEcho] ERROR:  Echo failed to publish 'Hello World 4'
.15 34758 [MQTTEcho] ERROR:  Echo failed to publish 'Hello World 5'
.16 39758 [MQTTEcho] ERROR:  Echo failed to publish 'Hello World 6'
.17 44758 [MQTTEcho] ERROR:  Echo failed to publish 'Hello World 7'
.18 49758 [MQTTEcho] ERROR:  Echo failed to publish 'Hello World 8'
.19 54758 [MQTTEcho] ERROR:  Echo failed to publish 'Hello World 9'
.20 59758 [MQTTEcho] ERROR:  Echo failed to publish 'Hello World 10'
.21 64758 [MQTTEcho] ERROR:  Echo failed to publish 'Hello World 11'
.22 69758 [MQTTEcho] MQTT echo demo finished.
23 69758 [MQTTEcho] ----Demo finished----

This is my policy:

{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": "iot:Connect",
      "Resource": "arn:aws:iot:eu-central-1:xxxxxxxxxxxx:*"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Publish",
      "Resource": "arn:aws:iot:eu-central.1:xxxxxxxxxxxx:*"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Subscribe",
      "Resource": "arn:aws:iot:eu-central-1:xxxxxxxxxxxx:*"
    },
    {
      "Effect": "Allow",
      "Action": "iot:Receive",
      "Resource": "arn:aws:iot:eu-central-1:xxxxxxxxxxxx:*"
    }
  ]
}

have I I forgot something?
could you advise ?
Thank you
BR
Marcello

gordonwang0, 2020-01-22:
Hello,

You policy appears to be incorrect.

To start, I would suggest testing with `"Resource": "*"` for all Actions.

Once that works, you can restrict the resources as below:
- Connect can be restricted to the client identifier you are using with `"Resource": "arn:aws:iot:eu-central-1:xxxxxxxxxxxx:client/<clientName>"` (Replace `<clientName>`)
- Publish and Receive can be restricted to the demo topic name with `"Resource": "arn:aws:iot:eu-central-1:xxxxxxxxxxxx:topic/freertos/demo/echo"`
- Subscribe can be restricted to the demo topic filter with `"Resource": "arn:aws:iot:eu-central-1:xxxxxxxxxxxx:topicfilter/freertos/demo/echo"`

For examples of policies please see https://docs.aws.amazon.com/iot/latest/developerguide/pub-sub-policy.html

ferrim, 2020-01-23:
Hello Gordon,
these are really useful informations and now I can take a step at a time and understand what I am doing ;-)
Thank you very much,
Marcello

closed 2020-01-31 by pvyawaha