aws/amazon-freertos #1706: Reduce the time required for OTA Update when using HTTP protocol for firmware upgrade

shubhamkulkarni97, 2020-01-22:
This is not a bug report actually.

The time required to perform OTA Update using HTTP should be less as compared to MQTT. But exactly opposite case is observed in case of OTA demo in Amazon FreeRTOS.

When using HTTP protocol, http request is performed to download each block of the firmware, which increases the time required to download the image.

As the firmware image is hosted on S3 bucket, can the entire image be downloaded in a single http request?

This would drastically reduce the time required to perform OTA updates.

Thank you!

pvyawaha, 2020-01-23:
Hi Shubham,

Thank you for posting your question here. When we are downloading the file over MQTT we are requesting number of blocks which is configured as `otaconfigMAX_NUM_BLOCKS_REQUEST`  in the OTA agent config. Whereas when we download the file over HTTP requests are made for each block of data which is configured by `otaconfigLOG2_FILE_BLOCK_SIZE` . To reduce the number of HTTP requests with current implementation we can increase the block size depending on the memory available on the platform. 

We will investigate reducing the HTTP requests made and maximum data request we can make on esp32 platform. Can you please share what is the performance difference you are observing in MQTT vs HTTP ?

shubhamkulkarni97, 2020-01-24:
@pvyawaha 
For downloading a firmware image of 880K,
OTA Update takes 60 seconds when MQTT is used, while it takes around 90 seconds when HTTP is used with block size set to 4K.

The time required reduces to 70 seconds when the block size is increased to 8K with HTTP. But, even in this case, time required is more than that of MQTT.

pvyawaha, 2020-01-29:
Hi Shubham,

Thank you for confirming that the download time reduces for increasing the block size which effectively reduces the HTTP requests. We will track further reducing the requests made during download over HTTP internally and share update. I would like to point out that we expect to observe different latency when downloading file over MQTT(using IOT endpoint) vs HTTP (S3 endpoint) . Also in case of MQTT the OTA streaming service caches the file whereas in case of HTTP the requests are directly made using pres-signed S3 URL. 

Please let me know if you have further questions on this topic.

shubhamkulkarni97, 2020-02-04:
Hi Prasad,

Can you please provide more information or pointers to links, on technical details of working of OTA in case of MQTT and HTTP?

pvyawaha, 2020-02-04:
Hi Shubham,

The OTA Agent in current implementation uses MQTT for all the control operations and gives option for downloading file over MQTT or HTTP protocol.

 While downloading the file the OTA Agent controls the requests for file block using same state machine and request timer. The difference here is when requesting file blocks over MQTT a single request is made for number of blocks as configured in `otaconfigMAX_NUM_BLOCKS_REQUEST` whereas in case of HTTP range requests are made for each block of data. This is the primary reason we have more requests being made when using HTTP downloads. We will prioritize making HTTP download more efficient and reducing these requests.

qiutongs, 2020-02-11:
If there is a still problem, feel free to re-open this issue or create a new one.

closed 2020-02-11 by qiutongs

mahavirj, 2020-02-11:
Per earlier comment this was supposed to be `Feature Request` and not a `Question`:

> We will prioritize making HTTP download more efficient and reducing these requests.

I would recommend to keep this open until support is added in mainline.

qiutongs, 2020-02-11:
Hi Mahavir,

Thank you for the feedback. We currently manage feature requests in our internal backlog tool and keep the open issues low. 

Do you want to keep this open to be notified when the change is in master? Please share what you want. We would love to hear that.

Thanks,
Qiutong

mahavirj, 2020-02-14:
Hi Qiutong,

> Thank you for the feedback. We currently manage feature requests in our internal backlog tool and keep the open issues low.

Any specific reason for taking this approach? (Since external users do not have access to your internal tracking system, there is no way to retrieve list of current known issues or feature requests).

> Do you want to keep this open to be notified when the change is in master? Please share what you want. We would love to hear that.

Yes, definitely. I would like to get notification on when issue gets closed. Moreover if there reference to corresponding PR (https://help.github.com/en/github/managing-your-work-on-github/linking-a-pull-request-to-an-issue) then it can help future users as well.

Thanks.