aws/amazon-freertos #1721: Windows Simulator - no MQTT message received

khteh, 2020-02-04:
**Describe the bug**
A clear and concise description of what the bug is.
I follow https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_windows.html
I manage to build and run the demo but there is no MQTT message received at the AWS IoT Console. What do I miss?

**System information**
- Which hardware board or part numbers? Windows Simulator
- IDE used Visual Studio 2019
- Operating System [Windows|Linux|MacOS] Windows
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it) I download from https://ap-southeast-1.console.aws.amazon.com/iot/home?region=ap-southeast-1#/software/freertos
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server] Sub pub Demo
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
A clear and concise description of what you expected to happen.
Hit F5 to run the demo from Visual Studio 2019 and expect some MQTT message (What's the default message?) in the IoT Console but there is none at all.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 
[image]
[image]

**To reproduce**
Steps to reproduce the behavior:

1. ...
2. ...

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
Add any other context about the problem here.

Thank you!

pvyawaha, 2020-02-05:
Hello,

Thank you for reporting this issue.

The screenshot of the console output that you shared shows that you are running greengrass discovery demo but the aws document that you are referring to is for the default MQTT demo. Can you please confirm which demo you are trying to run? 

Please go to `\amazon-freertos\vendors\pc\boards\windows\aws_demos\config_files\aws_demo_config.h` and check which demo is enabled.

For the default MQTT demo following macro must be defined as 
#define CONFIG_MQTT_DEMO_ENABLED
and AWS Documentation for that is [MQTT Demo](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_windows.html)

For Greengrass discovery it should be -
#define CONFIG_GREENGRASS_DISCOVERY_DEMO_ENABLED 
and AWS Documentation is [Greengrass Demo](https://docs.aws.amazon.com/freertos/latest/userguide/gg-demo.html)

khteh, 2020-02-05:
Just tried. It doesn't work. I put a breakpoint at the demo entry point `RunMqttDemo` but it is not reached at all.

khteh, 2020-02-06:
There is no selection process at all. Just scroll to the Windows Simulator, click the URL and a .zip file is downloaded. https://ap-southeast-1.console.aws.amazon.com/iot/home?region=ap-southeast-1#/freertos/configuration/Connect_to_AWS_IoT_-_Windows-Simulator?type=predefined

khteh, 2020-02-06:
@pvyawaha I downloaded and ran the GreenGrass Windows Simulator. I just downloaded the AWS IoT Windows Simulator and `CONFIG_MQTT_DEMO_ENABLED` is configured by default in the `aws_demo_config.h`. However, the entry point `RunMqttDemo` is still not reached at all.

qiutongs, 2020-02-06:
> There is no selection process at all. Just scroll to the Windows Simulator, click the URL and a .zip file is downloaded. https://ap-southeast-1.console.aws.amazon.com/iot/home?region=ap-southeast-1#/freertos/configuration/Connect_to_AWS_IoT_-_Windows-Simulator?type=predefined

I just tried it myself and everything works fine. Here are my steps.

1. downloaded from above link you pasted. 
2. opened the project file with Visual Studio 2017: projects\pc\windows\visual_studio\aws_demos\aws_demos.sln
3. update configNETWORK_INTERFACE_TO_USE to 1 in FreeRTOSConfig.h 
4. Build & Run, I can reach RunMqttDemo.

If you wasn't running MQTT demo, which one were you running? Can you put a break point here? https://github.com/aws/amazon-freertos/blob/master/demos/demo_runner/iot_demo_freertos.c#L352

khteh, 2020-02-07:
There are 5 network interfaces listed in the console output of the demo. However, it doesn't show the 802.11 wifi interface which I am using. That's the root cause of the problem. Ethernet is listed but I don't use it.

closed 2020-02-07 by qiutongs