aws/amazon-freertos #1733: Windows Simulator - 802.11 interface missing from interface list

khteh, 2020-02-07:
**Describe the bug**
A clear and concise description of what the bug is.
I downloaded IoT Windows Simulator from https://ap-southeast-1.console.aws.amazon.com/iot/home?region=ap-southeast-1#/freertos/configuration/Connect_to_AWS_IoT_-_Windows-Simulator?type=predefined and run it from Visual Studio 2019. There is no 802.11 in the list of interfaces when it runs. 
**System information**
- Which hardware board or part numbers? Windows Simulator
- IDE used: Visual Studio 2019
- Operating System [Windows|Linux|MacOS] Windows
- Version of Amazon FreeRTOS (run ``git describe --tags`` to find it)
- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server] MQTT
- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect to see 802.11 in the list of interfaces when the application runs.

**Screenshots or console output**
If appropriate, please cut-and-paste the console output to explain the issue. If possible, include the call stack. 
[image]

**To reproduce**
Steps to reproduce the behavior:

1. ...
2. ...

**Code to reproduce the bug**
The code should be wrapped in the cpp tag in order to be displayed clearly. For example:

```cpp
printf("Hello World")
```

If the code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

**Additional context**
Add any other context about the problem here.

Thank you!

khteh, 2020-02-07:
https://github.com/aws/amazon-freertos/issues/1721

qiutongs, 2020-02-08:
Unfortunately, the windows simulator doesn't support 802.11 interfaces, while only Ethernet interfaces are supported. There is a limitation stated in the WinPcap FAQ: https://www.winpcap.org/misc/faq.htm#Q-16

Please try an Ethernet interface. If there is still some problem. feel free to re-open this ticket.

closed 2020-02-08 by qiutongs