aws/amazon-freertos #1734: Heap available ignores external RAM after BLE init

danahern, 2020-02-07:
On version 201910.00 of Amazon FreeRTOS using an ESP32-wrover we setup the 4MB external RAM and initialize it.  Before we start initializing the Bluetooth we have ~4MB available and after we initialize we only have ~115KB available.  

Log:
```
AWSInitialize 71 Stack watermark: 2004w, heap available: 4283296b
33 39 [main] [DEBUG][AWS][390] Starting BLE init
35 39 [main] [INFO ][BLE][390] Releasing memory for Classic BT
36 39 [main] [INFO ][BLE][390] Initializing BLE Stack
I (677) BTDM_INIT: BT controller compile version [382a548]
I (707) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (767) phy: phy_version: 4009, e7844c0, Jul 15 2019, 13:20:06, 0, 0
AWSInitialize 79 Stack watermark: 2004w, heap available: 114920b
```

Function where the heap available gets ignored
```cpp
esp_err_t espBLEInit(){
    esp_err_t espRetVal;
    IotLogInfo("Releasing memory for Classic BT");
    espRetVal = esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);
    if(ESP_OK != espRetVal)
    {
        IotLogError("esp_bt_controller_mem_release error %d", espRetVal);
        return espRetVal;
    }
    IotLogInfo("Initializing BLE Stack");
    espRetVal = esp_nimble_hci_and_controller_init();
    if (ESP_OK != espRetVal)
    {
        IotLogError("esp_nimble_hci_and_controller_init error %d", espRetVal);
    }
    return espRetVal;
}
```

To reproduce modify sdkconfig
```
CONFIG_SPIRAM_SUPPORT=y
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_TYPE_ESPPSRAM32=y
CONFIG_SPIRAM_SIZE=4194304
CONFIG_SPIRAM_SPEED_40M=y
CONFIG_SPIRAM_MEMTEST=y
```
Then use the above function to release BT Classic and initialize nimble.  Heap allocation should be reset to the number it would be at without the external SPI RAM.

We are continuing to investigate and will update with more information as we discover more.

doothez, 2020-02-07:
More detailed Log:

brgAWSInitialize 71 Stack watermark: 2012w, heap available: 4288120b
33 37 [main] [DEBUG][BRG_AWS][370] Starting BLE init
34 37 [main] [INFO ][BLE][370] starting brgBLEInit
brgBLEInit 65 Stack watermark: 2012w, heap available: 4287924b
35 40 [main] [INFO ][BLE][400] Initializing BLE Stack
esp_nimble_hci_and_controller_init 378 Stack watermark: 2012w, heap available: 4319004b
I (693) BTDM_INIT: BT controller compile version [382a548]

**esp_bt_controller_init 953 Stack watermark: 2012w, heap available: 4318916b
esp_bt_controller_init 958 Stack watermark: 2012w, heap available: 119924b**
esp_nimble_hci_and_controller_init 382 Stack watermark: 2012w, heap available: 119924b
I (773) phy: phy_version: 4009, e7844c0, Jul 15 2019, 13:20:06, 0, 0
brgBLEInit 78 Stack watermark: 2012w, heap available: 119744b
36 73 [main] [INFO ][BLE][730] Initializing BLE Stack SUCCESS 
brgAWSInitialize 79 Stack watermark: 2012w, heap available: 119744b
37 74 [main] [DEBUG][BRG_AWS][740] Starting Enable Network(0x3)
38 79 [main] [ERROR][NET_MGR][790] Failed to initialize BLE.
I (1093) wifi: wifi driver task: 3ffb668c, prio:23, stack:3584, core=0
I (1093) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1103) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (1123) wifi: wifi firmware version: 3c46a62

So, looks like the huge memory consumption is happening at "btdm_controller_init()"

ravibhagavandas, 2020-02-10:
@danahern ,

Thank you for posting this issue.

I tried to reproduce the issue on FreeRTOS  201910.00 version and ESP32 wrover board, but the BLE controller init code consumes only 5k of RAM.

Here is the output log from the test run:

2 8 [main] Releasing memory for Classic BT
3 9 [main] Initializing BLE Stack
I (121) BTDM_INIT: BT controller compile version [382a548]

**esp_bt_controller_init 953 Stack watermark: 1948w, heap available: 4368040b**
I (131) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
**esp_bt_controller_init 958 Stack watermark: 1948w, heap available: 4363312b**
I (211) phy: phy_version: 4009, e7844c0, Jul 15 2019, 13:20:06, 0, 0

sdkconfig changes for enabling SPIRAM:

```
CONFIG_SPIRAM_SUPPORT=y
CONFIG_SPIRAM_BOOT_INIT=y
# CONFIG_SPIRAM_IGNORE_NOTFOUND is not set
# CONFIG_SPIRAM_USE_MEMMAP is not set
# CONFIG_SPIRAM_USE_CAPS_ALLOC is not set
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_TYPE_ESPPSRAM32=y
CONFIG_SPIRAM_SIZE=4194304
CONFIG_SPIRAM_SPEED_40M=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
# CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST is not set
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768
# CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY is not set
```

Could you verify that the sdkconfig changes matches yours ? 

Also what startup steps are performed from `app_main()` before `espBLEInit()` is invoked.

danahern, 2020-02-11:
Modifying the sdkconfig appears to have solved the problem.  I'm no longer seeing the issue where the memory amount changes.  I'll continue testing tomorrow. and make sure.

danahern, 2020-02-11:
Did more testing today.  It looks like the issue is resolved.  It looks like when we switched from the wroom to the wrover kit the pin mapping for one of our peripherals didn't get changed.  I'm not sure why it the memory drop was happening when it was but switching the pin mapping to not use 16 and 17 and setting the suggested values in the sdkconfig has resolved it.

closed 2020-02-11 by danahern