aws/amazon-freertos #1740: How to use older version

heshananupama, 2020-02-08:
Hi,
Could I please know the way to use the older versions? I could download the zip files, but I could not find the way to generate build files and then compiling the demo projects as I did for the latest version as described in the documentation. Could somebody provide me any place to get an idea Thanks

qiutongs, 2020-02-11:
Unfortunately, we don't have user documents for the Amazon FreeRTOS versions you want. You can open projects on the target platform. For example, here is the Windows Simulator project of v1.2.7 https://github.com/aws/amazon-freertos/tree/v1.2.7/demos/pc/windows/visual_studio

And you can still use our latest document for reference(how to use IDE, change configuration, what to expect in demo). https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_windows.html

heshananupama, 2020-02-11:
I figured this out. The main issue was I was using Cmake to do the builds in the newest version. I was thinking and trying to do the same for the older versions. However, later I realized that Cmake was added like more recently and did used the IDE based approach as it was described in the user documentation. It worked. Thanks. I am closing the issue.

closed 2020-02-11 by heshananupama