aws/amazon-freertos #1766: Sample/Demo code for WiFi SoftAP mode

teuteuguy, 2020-02-17:
Looking at the documentation for the WiFi library there is the SoftAP mode.
Could you provide sample/demo code that shows how to setup wifi credentials on a device using this mode?

ie. Have mobile or laptop connect to the device SoftAP, show a page, enter the SSID and Password and reboot with the new credentials.

Thank you!

dnoal, 2020-02-18:
Up to recently, our support for wifi provisioning is based upon BLE.  SoftAP mode is supported in our wifi abstraction, but the complete functionality to support WiFi provisioning using that abstraction has not been supported.  We have started to make progress here.  On ESP32 specifically, Espressif has recently contributed two PRs (1519 and 1571) that are now merged into master.  Using those two PRs (read the details on configuration), it should be possible to use the Espressif documented method of configuring the wifi module to SoftAP mode, serving an IP address (via DHCP) to a mobile client, serving a simple web page to that client, entering the WiFI credentials from that client, and doing the tap dance to restore the wifi module to client/ station mode and using the entered WiFi credentials.  That approach is in the ESP32 documentation.  We are working on documenting that approach and providing a code sample, but you should be enabled now.  We are working to make WiFi based provisioning uniform across our supported platforms but that is not available at this time.  Note:  Some wifi modules (Microchip) act differently than the ESP32 Wifi, and can support wifi provisioning now.  Please advise on your success.

pisukeman, 2020-03-05:
Hi @dnoal ! thanks for your response. We are currently developing a FreeRTOS firmware on a ESP32 device.
Our customer wants a wifi provisioning system using a wifi SoftAP, because the wifi provisioning via BLE is using too much memory on the device because we also need to make some HTTP call (BLE+HTTP is consuming too much memory on the ESP32)

As i can see in your latest message you are working on it. Is this possible to have some early version of your work in order to make the wifi based provisioning? or some workaround that should do the trik?
If this is not possible, how can we manage to do the provisioning using BLE and also make an HTTP call to download certificates as it's explained in this aws doc? (https://aws.amazon.com/es/blogs/iot/ensure-secure-communication-with-aws-iot-core-using-the-certificate-vending-machine-reference-application/). 
In the official documentation of FreeRTOS is explicity writen that: "Due to a limited amount of RAM, you must turn off BLE when you enable HTTP as an OTA data protocol. "

Thanks in advance for your time.

leegeth, 2020-03-11:
@pisukeman ,
Thanks for bringing this question to our attention. As you have correctly identified from Dan's comments we are working on getting the SoftAP support on ESP32. Please find an early version of this feature done by an espressif engineer and the instructions to try out the demo application.

Application to demonstrate SoftAP provisioning on ESP32 in Amazon FreeRTOS as shown in this branch 
https://github.com/shubhamkulkarni97/amazon-freertos/tree/feature/add_softap_provisioning_support.
 
esp_prov_example.c is the application code for SoftAP provisioning.
 
Steps to verify the application:
 
1.	Build and flash the application using LwIP stack.
2.	Download ESP SoftAP Prov app for android smartphone from Play Store (https://play.google.com/store/apps/details?id=com.espressif.provsoftap&hl=en)
3.	Connect to WiFi network created by ESP32 with SSID like PROV_XXXXXX with password abcd1234.
4.	Enter Proof Of Possession, which is set to abcd1234.
5.	Enter Credentials of a local WiFi network.

Some of these changes have already merged and we are currently in process of merging the remaining changes to master branch.

While we are exploring more on this provisioning option, we would like to point you about good amount memory foot print savings that you can achieve by turning off or tearing down BLE when not in use.

BLE can be turned off by 2 different ways depending on if a network manager is used or not. 
1. If network manager is used, invoking **AwsIotNetworkManager_DisableNetwork(AWS_IOT_NETWORK_TYPE_BLE)** will turn off BLE.
2. If network manager is not used, PAL function **IotBle_Off()** can be invoked to turn BLE off.
Support for this is available in latest master.  By doing this step, **~17kB** can be saved.

BLE can be completely teared down using a espressif function.
**esp_bt_controller_mem_release( ESP_BT_MODE_BLE )** can be invoked to completely tear down BLE. This can save **about 22kB** additional memory. If BLE needs to be re-enable a power cycle would be needed. However this would be required only if there is a re provisioning required.

Please let us know if there are any follow up questions.

TheIronNinja, 2020-04-27:
Hello @leegeth.

Has there been any progress towards the Wifi provisioning via SoftAP? Is the branch you linked in your post still the reference we should follow in order to implement this ourselves?

Thanks.

dachalco, 2021-03-01:
@TheIronNinja 
> Has there been any progress towards the Wifi provisioning via SoftAP? 

Yes. Compiling the info. Will follow up shortly.

dachalco, 2021-03-01:
Reference was requested and provided, so I'm closing this issue as the scope is snowballing.

For more general SoftAP questions, or any other issues outside this scope, please open a separate issue with details for you particular use-case. Thanks.

closed 2021-03-01 by dachalco