aws/amazon-freertos #1775: ota demo killing shadow demo mqtt connection

sameerdasarwad, 2020-02-20:
Hi, i am running two demos first shadow demo executing and after  that i am calling OTA demo but when i call OTA demo after some time it kills my shadow connection below are the logs 

32 900 [iot_thread] [INFO ][MQTT][9000] Establishing new MQTT connection.
33 900 [iot_thread] [INFO ][MQTT][9000] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
34 900 [iot_thread] [INFO ][MQTT][9000] (MQTT connection 0x3ffb40ec, CONNECT operation 0x3ffb6090) Waiting for operation completion.
35 930 [iot_thread] [INFO ][MQTT][9300] (MQTT connection 0x3ffb40ec, CONNECT operation 0x3ffb6090) Wait complete with result SUCCESS.
36 930 [iot_thread] [INFO ][MQTT][9300] New MQTT connection 0x3ffb3e08 established.
37 930 [iot_thread] [INFO ][Shadow][9300] (ESP_246F2825D19C) Modifying Shadow DELTA callback.
38 930 [iot_thread] [INFO ][Shadow][9300] (ESP_246F2825D19C) Adding new DELTA callback.
39 930 [iot_thread] [INFO ][MQTT][9300] (MQTT connection 0x3ffb40ec) SUBSCRIBE operation scheduled.
40 930 [iot_thread] [INFO ][MQTT][9300] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffb9dcc) Waiting for operation completion.
41 950 [iot_thread] [INFO ][MQTT][9500] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffb9dcc) Wait complete with result SUCCESS.
42 950 [iot_thread] [INFO ][Shadow][9500] (ESP_246F2825D19C) Shadow DELTA callback operation complete with result SUCCESS.
43 950 [iot_thread] [INFO ][Shadow][9500] (ESP_246F2825D19C) Modifying Shadow UPDATED callback.
44 950 [iot_thread] [INFO ][Shadow][9500] (ESP_246F2825D19C) Adding new UPDATED callback.
45 950 [iot_thread] [INFO ][MQTT][9500] (MQTT connection 0x3ffb40ec) SUBSCRIBE operation scheduled.
46 950 [iot_thread] [INFO ][MQTT][9500] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffb5e6c) Waiting for operation completion.
47 970 [iot_thread] [INFO ][MQTT][9700] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffb5e6c) Wait complete with result SUCCESS.
48 970 [iot_thread] [INFO ][Shadow][9700] (ESP_246F2825D19C) Shadow UPDATED callback operation complete with result SUCCESS.
49 970 [iot_thread] [INFO ][MQTT][9700] MQTT library successfully initialized.
50 970 [iot_thread] [INFO ][Shadow][9700] Shadow library successfully initialized.
51 971 [iot_thread] [INFO ][MQTT][9710] (MQTT connection 0x3ffb40ec) SUBSCRIBE operation scheduled.
52 971 [iot_thread] [INFO ][MQTT][9710] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffb9dcc) Waiting for operation completion.
53 990 [iot_thread] [INFO ][MQTT][9900] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffb9dcc) Wait complete with result SUCCESS.
54 990 [iot_thread] [INFO ][MQTT][9900] (MQTT connection 0x3ffb40ec) SUBSCRIBE operation scheduled.
55 990 [iot_thread] [INFO ][MQTT][9900] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffb6090) Waiting for operation completion.
56 1011 [iot_thread] [INFO ][MQTT][10110] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffb6090) Wait complete with result SUCCESS.
57 1011 [iot_thread] [INFO ][MQTT][10110] (MQTT connection 0x3ffb40ec) MQTT PUBLISH operation queued.
58 1043 [iot_thread] [INFO ][DEMO][10430] Shadow was updated!
Previous: {"state":{"desired":{"switch_1":1,"switch_2":1,"switch_3":0,"switch_4":0},"reported":{"switch_1":0,"switch_2":1,"switc59 1044 [iot_thread] [INFO ][DEMO][10440] ESP_246F2825D19C changing state from 0 to 0.
60 1044 [iot_thread] [INFO ][MQTT][10440] (MQTT connection 0x3ffb40ec) MQTT PUBLISH operation queued.
61 1045 [iot_thread] [INFO ][DEMO][10450] ESP_246F2825D19C sent new state report.
62 1063 [iot_thread] [INFO ][Shadow][10630] Shadow UPDATE of ESP_246F2825D19C was ACCEPTED.
63 1104 [iot_thread] [INFO ][DEMO][11040] Shadow was updated!
Previous: {"state":{"desired":{"switch_1":0,"switch_2":0,"switch_3":0,"switch_4":0},"reported":{"switch_1":0,"switch_2":1,"s65 1423 [iot_thread] [INFO ][MQTT][14230] (MQTT connection 0x3ffb40ec) MQTT PUBLISH operation queued.
66 1443 [IP-task] Network buffers: 57 lowest 57
67 1444 [iot_thread] [INFO ][DEMO][14440] Shadow was updated!
Previous: {"state":{"desired":{"switch_1":0,"switch_2":0,"switch_3":0,"switch_4":0},"reported":{"switch_1":0,"switch_2":0,"switc68 1445 [iot_thread] [INFO ][DEMO][14450] ESP_246F2825D19C changing state from 0 to 1.
69 1445 [iot_thread] [INFO ][MQTT][14450] (MQTT connection 0x3ffb40ec) MQTT PUBLISH operation queued.
70 1445 [iot_thread] [INFO ][DEMO][14450] ESP_246F2825D19C sent new state report.
71 1462 [iot_thread] [INFO ][Shadow][14620] Shadow UPDATE of ESP_246F2825D19C was ACCEPTED.
72 1462 [iot_thread] [INFO ][MQTT][14620] MQTT library successfully initialized.
73 1462 [iot_thread] OTA demo version 0.9.2
74 1462 [iot_thread] Creating MQTT Client...
75 1498 [IP-task] Heap: current 37340 lowest 37200
76 1499 [IP-task] Heap: current 36616 lowest 36480
77 1499 [iot_thread] [INFO ][DEMO][14990] Shadow was updated!
Previous: {"state":{"desired":{"switch_1":0,"switch_2":1,"switch_3":0,"switch_4":0},"reported":{"switch_1":0,"switch_2":0,"switc78 1500 [IP-task] Heap: current 37340 lowest 36260
79 1500 [iot_thread] [INFO ][Shadow][15000] Shadow UPDATE of ESP_246F2825D19C was ACCEPTED.
80 1522 [wifi] Heap: current 36072 lowest 35796
81 1523 [wifi] Heap: current 36004 lowest 35600
82 1523 [wifi] Heap: current 35956 lowest 34848
83 1523 [wifi] Heap: current 35908 lowest 34800
84 1523 [IP-task] Network buffers: 57 lowest 56
85 1523 [IP-task] Heap: current 32712 lowest 32248
86 1523 [IP-task] Heap: current 33100 lowest 32240
87 1594 [wifi] Heap: current 21912 lowest 21100
88 1599 [IP-task] Heap: current 21340 lowest 20428
89 1625 [wifi] Heap: current 20172 lowest 20068
90 1625 [wifi] Heap: current 21576 lowest 19636
91 1666 [wifi] Heap: current 17708 lowest 17604
92 1696 [wifi] Heap: current 16248 lowest 10972
93 1778 [wifi] Heap: current 9760 lowest 9672
94 1789 [wifi] Heap: current 12268 lowest 9388
95 1901 [wifi] Heap: current 5816 lowest 5720
96 1922 [wifi] Heap: current 5816 lowest 5436
97 2003 [wifi] Heap: current 9104 lowest 5288
246F2825D19C
thing_name:ESP_246F2825D19C
246F2825D19C
[... 77 lines cut ...]
168 4962 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
169 5062 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
170 5162 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
171 5262 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
172 5362 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
173 5462 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
174 5562 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
175 5662 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
176 5762 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
177 5862 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
178 5962 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
179 6062 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
180 6162 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
181 6262 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
182 6362 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
183 6462 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
184 6562 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
185 6662 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
186 6762 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
187 6862 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
188 6930 [iot_thread] [ERROR][NET][69300] Error -27648 while sending data.
189 6930 [iot_thread] [ERROR][MQTT][69300] (MQTT connection 0x3ffb40ec) Failed to send PINGREQ.
190 6931 [iot_thread] [INFO ][MQTT][69310] (MQTT connection 0x3ffb40ec) Network connection closed.
191 6962 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
192 7062 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

sameerdasarwad, 2020-02-20:
this is the another logs when i call two demos at a time 
7 630 [iot_thread] [INFO ][MQTT][6300] MQTT library successfully initialized.
28 630 [iot_thread] [INFO ][Shadow][6300] Shadow library successfully initialized.
29 630 [iot_thread] [INFO ][DEMO][6300] Shadow Thing Name is ESP_246F2825D19C (length 16).
30 636 [iot_thread] DNS[0xCDF5]: The answer to 'a1l1zbts62o6u8-ats.iot.eu-west-1.amazonaws.com' (22fb8fc7ip) will be stored
31 637 [IP-task] Heap: current 110224 lowest 97668
32 638 [main] DNS[0x8F46]: The answer to 'a1l1zbts62o6u8-ats.iot.eu-west-1.amazonaws.com' (22fb8fc7ip) will be stored
I (6900) PKCS11: Initializing NVS partition: "storage"
33 670 [wifi] Heap: current 96036 lowest 95428
34 676 [IP-task] Heap: current 76504 lowest 76304
35 693 [IP-task] Heap: current 45552 lowest 45404
36 693 [wifi] Heap: current 44296 lowest 44148
37 695 [wifi] Heap: current 44296 lowest 43864
38 695 [IP-task] Heap: current 41804 lowest 41656
39 696 [wifi] Heap: current 41072 lowest 40924
40 696 [IP-task] Heap: current 41616 lowest 39964
41 728 [IP-task] Heap: current 39404 lowest 39256
42 729 [wifi] Heap: current 38672 lowest 38524
43 729 [IP-task] Heap: current 39208 lowest 37556
44 764 [wifi] Heap: current 37868 lowest 36828
45 768 [wifi] Heap: current 28172 lowest 26644
46 772 [wifi] Heap: current 25172 lowest 25116
47 773 [wifi] Heap: current 26580 lowest 24832
48 781 [wifi] Heap: current 24860 lowest 24820
49 781 [IP-task] Heap: current 24620 lowest 23852
50 865 [wifi] Heap: current 20908 lowest 15984
51 1049 [wifi] Heap: current 15628 lowest 15352
52 1069 [wifi] Heap: current 10752 lowest 10748
53 1073 [IP-task] Heap: current 11780 lowest 10464
54 1121 [wifi] Heap: current 10336 lowest 10332
55 1172 [wifi] Heap: current 13008 lowest 10048
56 1182 [wifi] Heap: current 4080 lowest 4080
57 1195 [IP-task] Heap: current 5116 lowest 3796
58 1196 [iot_thread] [INFO ][MQTT][11960] Establishing new MQTT connection.
59 1196 [iot_thread] [ERROR][NET][11960] Failed to create network receive task.
60 1196 [iot_thread] [ERROR][MQTT][11960] Failed to set MQTT network receive callback.
61 1196 [iot_thread] [ERROR][MQTT][11960] Failed to establish new MQTT connection, error NETWORK ERROR.
62 1196 [iot_thread] [INFO ][MQTT][11960] Network connection closed on error.
63 1198 [iot_thread] [INFO ][MQTT][11980] (MQTT connection 0x3ffd83b4) Network connection destroyed.
64 1198 [iot_thread] [ERROR][DEMO][11980] MQTT CONNECT returned error NETWORK ERROR.
65 1198 [iot_thread] [INFO ][DEMO][11980] Shadow Thing Name is ESP_246F2825D19C (length 16).
66 1402 [iot_thread] ERROR: Handshake failed with error code -16 
67 1402 [iot_thread] [ERROR][NET][14020] Failed to establish new connection. Socket status: -1002.
68 1403 [iot_thread] [ERROR][MQTT][14030] Failed to establish new MQTT connection, error NETWORK ERROR.
69 1403 [iot_thread] [ERROR][DEMO][14030] MQTT CONNECT returned error NETWORK ERROR.
70 1404 [iot_thread] [INFO ][DEMO][14040] Shadow Thing Name is ESP_246F2825D19C (length 16).
71 1405 [IP-task] Heap: current 55788 lowest 1228
72 1615 [iot_thread] ERROR: Handshake failed with error code -16 
73 1615 [iot_thread] [ERROR][NET][16150] Failed to establish new connection. Socket status: -1002.
74 1616 [iot_thread] [ERROR][MQTT][16160] Failed to establish new MQTT connection, error NETWORK ERROR.
75 1616 [iot_thread] [ERROR][DEMO][16160] MQTT CONNECT returned error NETWORK ERROR.
76 1617 [iot_thread] [INFO ][DEMO][16170] Shadow Thing Name is ESP_246F2825D19C (length 16).
77 1618 [IP-task] Heap: current 52564 lowest 1072
78 1748 [iot_thread] ERROR: Handshake failed with error code -9984 
79 1748 [iot_thread] [ERROR][NET][17480] Failed to establish new connection. Socket status: -1002.
80 1748 [iot_thread] [ERROR][MQTT][17480] Failed to establish new MQTT connection, error NETWORK ERROR.
81 1749 [iot_thread] [ERROR][DEMO][17490] MQTT CONNECT returned error NETWORK ERROR.
82 1750 [iot_thread] [INFO ][DEMO][17500] Shadow Thing Name is ESP_246F2825D19C (length 16).
83 1956 [iot_thread] ERROR: Handshake failed with error code -16 
84 1956 [iot_thread] [ERROR][NET][19560] Failed to establish new connection. Socket status: -1002.
85 1956 [iot_thread] [ERROR][MQTT][19560] Failed to establish new MQTT connection, error NETWORK ERROR.
86 1956 [iot_thread] [ERROR][DEMO][19560] MQTT CONNECT returned error NETWORK ERROR.
87 1957 [iot_thread] [INFO ][DEMO][19570] Shadow Thing Name is ESP_246F2825D19C (length 16).
88 1959 [main] Connecting to broker...
89 1959 [main] printing mac address:246F2825D19C
thing_name:ESP_246F2825D19C
90 1960 [main] printing mac address:246F2825D19C
thing_name:ESP_246F2825D19C
91 1960 [main] [INFO ][MQTT][19600] Establishing new MQTT connection.
92 1961 [main] [INFO ][MQTT][19610] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
[... 136 lines cut ...]
221 4866 [main] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
222 4977 [iot_thread] ERROR: Handshake failed with error code -16 
223 4977 [iot_thread] [ERROR][NET][49770] Failed to establish new connection. Socket status: -1002.
224 4977 [iot_thread] [ERROR][MQTT][49770] Failed to establish new MQTT connection, error NETWORK ERROR.
225 4977 [iot_thread] [ERROR][DEMO][49770] MQTT CONNECT returned error NETWORK ERROR.
226 4978 [iot_thread] [INFO ][DEMO][49780] Shadow Thing Name is ESP_246F2825D19C (length 16).
[... cut ...]
239 5408 [iot_thread] [ERROR][DEMO][54070] MQTT CONNECT returned error NETWORK ERROR.
240 5409 [iot_thread] [INFO ][DEMO][54090] Shadow Thing Name is ESP_246F2825D19C (length 16).
241 5410 [main] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
242 5511 [main] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
243 5622 [iot_thread] ERROR: Handshake failed with error code -16 
244 5622 [iot_thread] [ERROR][NET][56220] Failed to establish new connection. Socket status: -1002.
245 5622 [iot_t

yanjos-dev, 2020-02-21:
Hi sameerdasarwad ,

Are you using the same client ID for the Shadow and OTA connections? If this is the case, then you will need update the demo client ids so that they are unique relative to each other. 

Two clients with the same client ID are not allowed to be connected concurrently to the message broker. When a client connects to the message broker using a client ID that another client is using, the new client connection is accepted and the previously connected client is disconnected. 

See the following link for details:
https://docs.aws.amazon.com/iot/latest/developerguide/mqtt.html

sameerdasarwad, 2020-02-21:
both demo using separate network , but not sure about client id or client token your saying i mean client id is client token

sameerdasarwad, 2020-02-21:
@param[in] pIdentifier NULL-terminated MQTT client identifier. The Shadow
 * demo will use the Thing Name as the client identifier.
above line for shadow demo but i didn't find about ota demo

yanjos-dev, 2020-02-22:
> both demo using separate network , but not sure about client id or client token your saying i mean client id is client token

Just to clarify, MQTT client IDs aren't associated with a [Thing in AWS IoT](https://docs.aws.amazon.com/iot/latest/developerguide/iot-thing-management.html). They are only used to uniquely identify the MQTT connection. Also, the client ID does not need to be set to the Thing name, that is just what was used in the demos. 
 
The [client token](https://docs.aws.amazon.com/iot/latest/developerguide/device-shadow-document.html#client-token) is unrelated to the MQTT client ID. It's a property of a [Device Shadow Service Document](https://docs.aws.amazon.com/iot/latest/developerguide/device-shadow-document.html). Client tokens are used by the Shadow Library to associate responses with requests in an MQTT environment. The OTA Library does not have the concept of client tokens.

> @param[in] pIdentifier NULL-terminated MQTT client identifier. The Shadow
> 
> * demo will use the Thing Name as the client identifier.
>   above line for shadow demo but i didn't find about ota demo
>   if  i want to send ota update for same thing at that time if ota demo uses same client id as per shadow demo comment, it means  that it will kill shadow demo connection and shadow will not update by default until ota demo finish or terminate

Your understanding is correct, the Shadow Library demo is using the thing name for the client identifier. The OTA demo is also using the Thing name as the client identifier. The define that the OTA and Shadow demos are using can be found in the "aws_clientcredential.h" file. https://github.com/aws/amazon-freertos/blob/94797e54a8ad1f61bb08a63857e7b3d1541d1d64/demos/include/aws_clientcredential.h#L41

As you pointed out, this causes the MQTT connection created by the Shadow Library demo to disconnect when the OTA demo tries to connect. 
 
In the OTA demo code, you can see where the demo:
1.	Creates a "xConnectInfo" variable of type "IotMqttConnectInfo_t" for setting the MQTT connection information: https://github.com/aws/amazon-freertos/blob/94797e54a8ad1f61bb08a63857e7b3d1541d1d64/demos/ota/aws_iot_ota_update_demo.c#L209
2.	Sets the "pClientIdentifier" field of "xConnectInfo" to be the Thing name: https://github.com/aws/amazon-freertos/blob/94797e54a8ad1f61bb08a63857e7b3d1541d1d64/demos/ota/aws_iot_ota_update_demo.c#L242-L243
3.	Creates the MQTT connection with the previously defined connection information https://github.com/aws/amazon-freertos/blob/94797e54a8ad1f61bb08a63857e7b3d1541d1d64/demos/ota/aws_iot_ota_update_demo.c#L246-L248
 
You will need to update the code in step 2 to use a client identifier that is different from the shadow library. In this case, you will need to set it to something other than the Thing name.

sameerdasarwad, 2020-02-22:
@yanjos-dev  thanks for valueable information, but when i change the OTA client Identifier same thing happening again below are the logs
  

20 640 [iot_thread] [INFO ][MQTT][6400] MQTT library successfully initialized.
21 640 [iot_thread] [INFO ][Shadow][6400] Shadow library successfully initialized.
22 641 [iot_thread] [INFO ][DEMO][6410] Shadow Thing Name is ESP_246F2825D19C (length 16).
23 650 [iot_thread] DNS[0x40E4]: The answer to 'xxxxxxxxxxxx-ats.iot.eu-west-1.amazonaws.com' (34d4a7ebip) will be stored
24 650 [IP-task] Heap: current 115396 lowest 102124
I (7045) PKCS11: Initializing NVS partition: "storage"
25 680 [wifi] Heap: current 101104 lowest 100428
26 687 [IP-task] Heap: current 81872 lowest 81800
27 711 [wifi] Heap: current 81052 lowest 80104
28 711 [IP-task] Heap: current 78760 lowest 78584
29 711 [wifi] Heap: current 78024 lowest 77076
30 712 [IP-task] Heap: current 78564 lowest 76304
31 782 [wifi] Heap: current 66300 lowest 65580
32 788 [IP-task] Heap: current 65720 lowest 64908
33 803 [wifi] Heap: current 64896 lowest 64896
34 813 [wifi] Heap: current 64680 lowest 64452
35 814 [IP-task] Heap: current 64900 lowest 64396
36 885 [wifi] Heap: current 59936 lowest 55468
37 967 [wifi] Heap: current 54148 lowest 54148
38 977 [wifi] Heap: current 57152 lowest 53864
39 1089 [wifi] Heap: current 50200 lowest 50200
40 1100 [wifi] Heap: current 50200 lowest 49916
41 1171 [wifi] Heap: current 49784 lowest 49784
42 1182 [wifi] Heap: current 49784 lowest 49340
43 1224 [iot_thread] [INFO ][MQTT][12240] Establishing new MQTT connection.
44 1224 [iot_thread] [INFO ][MQTT][12240] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
45 1224 [iot_thread] [INFO ][MQTT][12240] (MQTT connection 0x3ffb9b9c, CONNECT operation 0x3ffb9cb8) Waiting for operation completion.
46 1255 [iot_thread] [INFO ][MQTT][12550] (MQTT connection 0x3ffb9b9c, CONNECT operation 0x3ffb9cb8) Wait complete with result SUCCESS.
47 1255 [iot_thread] [INFO ][MQTT][12550] New MQTT connection 0x3ffb3e08 established.
48 1255 [iot_thread] [INFO ][Shadow][12550] (ESP_246F2825D19C) Modifying Shadow DELTA callback.
49 1255 [iot_thread] [INFO ][Shadow][12550] (ESP_246F2825D19C) Adding new DELTA callback.
50 1255 [iot_thread] [INFO ][MQTT][12550] (MQTT connection 0x3ffb9b9c) SUBSCRIBE operation scheduled.
51 1255 [iot_thread] [INFO ][MQTT][12550] (MQTT connection 0x3ffb9b9c, SUBSCRIBE operation 0x3ffd7ba0) Waiting for operation completion.
52 1275 [iot_thread] [INFO ][MQTT][12750] (MQTT connection 0x3ffb9b9c, SUBSCRIBE operation 0x3ffd7ba0) Wait complete with result SUCCESS.
53 1275 [iot_thread] [INFO ][Shadow][12750] (ESP_246F2825D19C) Shadow DELTA callback operation complete with result SUCCESS.
54 1275 [iot_thread] [INFO ][Shadow][12750] (ESP_246F2825D19C) Modifying Shadow UPDATED callback.
55 1275 [iot_thread] [INFO ][Shadow][12750] (ESP_246F2825D19C) Adding new UPDATED callback.
56 1275 [iot_thread] [INFO ][MQTT][12750] (MQTT connection 0x3ffb9b9c) SUBSCRIBE operation scheduled.
57 1275 [iot_thread] [INFO ][MQTT][12750] (MQTT connection 0x3ffb9b9c, SUBSCRIBE operation 0x3ffb9cb8) Waiting for operation completion.
58 1295 [iot_thread] [INFO ][MQTT][12950] (MQTT connection 0x3ffb9b9c, SUBSCRIBE operation 0x3ffb9cb8) Wait complete with result SUCCESS.
59 1295 [iot_thread] [INFO ][Shadow][12950] (ESP_246F2825D19C) Shadow UPDATED callback operation complete with result SUCCESS.
60 1295 [iot_thread] [INFO ][MQTT][12950] MQTT library successfully initialized.
61 1295 [iot_thread] [INFO ][Shadow][12950] Shadow library successfully initialized.
62 1296 [iot_thread] [INFO ][MQTT][12960] (MQTT connection 0x3ffb9b9c) SUBSCRIBE operation scheduled.
63 1296 [iot_thread] [INFO ][MQTT][12960] (MQTT connection 0x3ffb9b9c, SUBSCRIBE operation 0x3ffba020) Waiting for operation completion.
64 1326 [iot_thread] [INFO ][MQTT][13260] (MQTT connection 0x3ffb9b9c, SUBSCRIBE operation 0x3ffba020) Wait complete with result SUCCESS.
65 1326 [iot_thread] [INFO ][MQTT][13260] (MQTT connection 0x3ffb9b9c) SUBSCRIBE operation scheduled.
66 1326 [iot_thread] [INFO ][MQTT][13260] (MQTT connection 0x3ffb9b9c, SUBSCRIBE operation 0x3ffb9cb8) Waiting for operation completion.
67 1357 [iot_thread] [INFO ][MQTT][13560] (MQTT connection 0x3ffb9b9c, SUBSCRIBE operation 0x3ffb9cb8) Wait complete with result SUCCESS.
68 1357 [iot_thread] [INFO ][MQTT][13570] (MQTT connection 0x3ffb9b9c) MQTT PUBLISH operation queued.
69 1396 [iot_thread] [INFO ][DEMO][13960] ESP_246F2825D19C changing state from 0 to 0.
70 1396 [iot_thread] [INFO ][MQTT][13960] (MQTT connection 0x3ffb9b9c) MQTT PUBLISH operation queued.
71 1396 [iot_thread] [INFO ][DEMO][13960] ESP_246F2825D19C sent new state report.
72 1429 [iot_thread] [INFO ][DEMO][14290] Shadow was updated!
Previous: {"state":{"desired":{"switch_1":0,"switch_2":0,"switch_3":0,"switch_4":0,"otajobflag":1},"reported":{"switch_1":0,"swi73 1430 [iot_thread] [INFO ][Shadow][14300] Shadow UPDATE of ESP_246F2825D19C was ACCEPTED.
74 1451 [iot_thread] [INFO ][DEMO][14510] Shadow was updated!
Previous: {"state":{"desired":{"switch_1":0,"switch_2":0,"switch_3":0,"switch_4":0,"otajobflag":0},"reported":{"switch_1":0,"swi75 1469 [iot_thread] [INFO ][Shadow][14690] Shadow UPDATE of ESP_246F2825D19C was ACCEPTED.
76 2566 [iot_thread] [INFO ][DEMO][25660] Shadow was updated!
[... cut ...]
130 3917 [iot_thread] [INFO ][MQTT][39170] (MQTT connection 0x3ffd8fd0) Disconnecting connection.
131 3917 [iot_thread] [INFO ][MQTT][39170] (MQTT connection 0x3ffd8fd0, DISCONNECT operation 0x3ffd8d18) Waiting for operation completion.
132 3920 [iot_thread] [INFO ][MQTT][39200] (MQTT connection 0x3ffd8fd0, DISCONNECT operation 0x3ffd8d18) Wait complete with result SUCCESS.
133 3920 [iot_thread] [INFO ][MQTT][39200] (MQTT connection 0x3ffd8fd0) Connection disconnected.
134 3921 [iot_thread] [INFO ][MQTT][39210] (MQTT connection 0x3ffd8fd0) Network connection closed.

sameerdasarwad, 2020-02-22:
look like  below line is doing something
114 3192 [iot_thread] [OTA_AgentInit_internal] Failed to start the OTA Task, Error Code :ffffffff Queue:00000000

sameerdasarwad, 2020-02-22:
@yanjos-dev  thanks it was connection problem with mqtt not i am able to run both mqtt and shadow together with different client ID  thank you

sameerdasarwad, 2020-02-22:
it was worked only once  

I (6895) PKCS11: Initializing NVS partition: "storage"
28 670 [IP-task] Heap: current 76536 lowest 76292
29 687 [IP-task] Heap: current 45460 lowest 45324
30 688 [wifi] Heap: current 44640 lowest 44504
31 688 [wifi] Heap: current 44356 lowest 44220
32 688 [IP-task] Heap: current 41948 lowest 41812
33 688 [wifi] Heap: current 41408 lowest 41272
34 688 [IP-task] Heap: current 41948 lowest 40304
35 708 [wifi] Heap: current 37624 lowest 37488
36 708 [IP-task] Heap: current 38532 lowest 36520
37 757 [wifi] Heap: current 25160 lowest 25092
38 757 [wifi] Heap: current 26568 lowest 24808
39 808 [wifi] Heap: current 24608 lowest 24580
40 809 [wifi] Heap: current 24324 lowest 24296
41 809 [wifi] Heap: current 25632 lowest 24012
42 829 [wifi] Heap: current 14800 lowest 14772
43 839 [wifi] Heap: current 20972 lowest 14488
44 1044 [wifi] Heap: current 10672 lowest 10652
45 1054 [wifi] Heap: current 10672 lowest 10368
46 1095 [wifi] Heap: current 10260 lowest 10208
47 1115 [wifi] Heap: current 10260 lowest 9796
48 1169 [IP-task] Heap: current 4748 lowest 4232
49 1170 [wifi] Heap: current 5100 lowest 3104
50 1170 [iot_thread] [INFO ][MQTT][11700] Establishing new MQTT connection.
51 1171 [iot_thread] [ERROR][NET][11710] Failed to create network receive task.
52 1171 [iot_thread] [ERROR][MQTT][11710] Failed to set MQTT network receive callback.
53 1171 [iot_thread] [ERROR][MQTT][11710] Failed to establish new MQTT connection, error NETWORK ERROR.
54 1171 [iot_thread] [INFO ][MQTT][11710] Network connection closed on error.
55 1172 [iot_thread] [INFO ][MQTT][11720] (MQTT connection 0x3ffd84b4) Network connection destroyed.
56 1172 [iot_thread] [ERROR][DEMO][11720] MQTT CONNECT returned error NETWORK ERROR.
57 1173 [iot_thread] [INFO ][DEMO][11730] Shadow Thing Name is ESP_246F2825D19C (length 16).
58 1376 [iot_thread] ERROR: �%shake failed with error code -16 
59 1376 [iot_thread] [ERROR][NET][13760] Failed to establish new connection. Socket status: -1002.
60 1377 [iot_thread] [ERROR][MQTT][13770] Failed to establish new MQTT connection, error NETWORK ERROR.
61 1377 [iot_thread] [ERROR][DEMO][13770] MQTT CONNECT returned error NETWORK ERROR.
62 1378 [iot_thread] [INFO ][DEMO][13780] Shadow Thing Name is ESP_246F2825D19C (length 16).
63 1379 [IP-task] Heap: current 55216 lowest 1160
64 1591 [iot_thread] ERROR: Handshake failed with error code -16 
65 1591 [iot_thread] [ERROR][NET][15910] Failed to establish new connection. Socket status: -1002.
66 1591 [iot_thread] [ERROR][MQTT][15910] Failed to establish new MQTT connection, error NETWORK ERROR.
67 1591 [iot_thread] [ERROR][DEMO][15910] MQTT CONNECT returned error NETWORK ERROR.
68 1592 [iot_thread] [INFO ][DEMO][15920] Shadow Thing Name is ESP_246F2825D19C (length 16).
69 1722 [iot_thread] ERROR: Handshake failed with error code -9984 
70 1722 [iot_thread] [ERROR][NET][17220] Failed to establish new connection. Socket status: -1002.
71 1723 [iot_thread] [ERROR][MQTT][17230] Failed to establish new MQTT connection, error NETWORK ERROR.
72 1723 [iot_thread] [ERROR][DEMO][17230] MQTT CONNECT returned error NETWORK ERROR.
73 1724 [iot_thread] [INFO ][DEMO][17240] Shadow Thing Name is ESP_246F2825D19C (length 16).
74 1930 [iot_thread] ERROR: Handshake failed with error code -16 
75 1930 [iot_thread] [ERROR][NET][19300] Failed to establish new connection. Socket status: -1002.
76 1930 [iot_thread] [ERROR][MQTT][19300] Failed to establish new MQTT connection, error NETWORK ERROR.
77 1930 [iot_thread] [ERROR][DEMO][19300] MQTT CONNECT returned error NETWORK ERROR.
78 1931 [iot_thread] [INFO ][DEMO][19310] Shadow Thing Name is ESP_246F2825D19C (length 16).
79 1932 [IP-task] Heap: current 52144 lowest 1092
80 1934 [main] Connecting to broker...
81 1934 [main] printing mac address:246F2825D19C
thing_name:ESP_246F2825D19C
82 1934 [main] [INFO ][MQTT][19340] Establishing new MQTT connection.
83 1935 [main] [INFO ][MQTT][19350] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
84 1937 [main] [INFO ][MQTT][19370] (MQTT connection 0x3ffd8648, CONNECT operation 0x3ffd9fec) Waiting for operation completion.
85 1975 [main] [INFO ][MQTT][19740] (MQTT connection 0x3ffd8648, CONNECT operation 0x3ffd9fec) Wait complete with result SUCCESS.
86 1975 [main] [INFO ][MQTT][19750] New MQTT connection 0x3ffbf65c established.
87 1975 [main] Connected to broker.
88 1976 [main] printing mac address:246F2825D19C
thing_name:ESP_246F2825D19C
89 1976 [main] [OTA_AgentInit_internal] OTA Task is Ready.
90 1977 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [Ready] Event [Start] New state [RequestingJob] 
91 1977 [OTA Agent Task] [INFO ][MQTT][19770] (MQTT connection 0x3ffd8648) SUBSCRIBE operation scheduled.
92 1977 [OTA Agent Task] [INFO ][MQTT][19770] (MQTT connection 0x3ffd8648, SUBSCRIBE operation 0x3ffd9fec) Waiting for operation completion.
[... 25 lines cut ...]
111 2040 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: files
112 2040 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: filepath
113 2040 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: filesize
114 2040 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: fileid
115 2040 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: certfile
116 2040 [OTA Agent Task] [prvParseJSONbyModel] parameter not present: sig-sha256-ecdsa
[... cut ...]
130 2362 [iot_thread] [ERROR][MQTT][23620] Failed to establish new MQTT connection, error NETWORK ERROR.
131 2362 [iot_thread] [ERROR][DEMO][23620] MQTT CONNECT returned error NETWORK ERROR.
132 2363 [iot_thread] [INFO ][DEMO][23630] Shadow Thing Name is ESP_246F2825D19C (length 16).
133 2364 [IP-task] Heap: current 57744 lowest 948
134 2365 [main] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
135 2466 [main] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0

htibosch, 2020-02-22:
You are not telling what platform you are using? But as I see "ESP_246F2825D19C", I assume it is an ESP32?

~~~
> 49 1170 [wifi] Heap: current 5100 lowest 3104
> 51 1171 [iot_thread] [ERROR][NET][11710] Failed to create network receive task.
~~~

It looks like it is running out of heap space.
I recently investigated the usage of RAM in the ESP demo project, and I made recommendations to lower the RAM foot-print.
Please have a look [here](https://github.com/aws/amazon-freertos/issues/1250)

sameerdasarwad, 2020-02-22:
@htibosch i am using esp32 for two demos one is shadow and another one is OTA i just want to run both demo,
yes esp32 is running out of heap space that way i just made separate calls for both demo first i initialize shadow demo ones its initialization complete i am initiating OTA demo but  i am getting error like below

2695 [OTA_task] mbedTLS sign failed with error -17168 
86 2696 [OTA_task] ERROR: Failure in signing callback: 6 
87 2696 [OTA_task] ERROR: Handshake failed with error code -2003 
88 2696 [OTA_task] [ERROR][NET][26960] Failed to establish new connection. Socket status: -1002.
89 2696 [OTA_task] Network Connection failed, retry delay 5000 ms, retries left 100
90 3638 [wifi] Heap: current 1980 lowest 1628
91 3645 [OTA_task] mbedTLS sign failed with error -17168 
92 3645 [OTA_task] ERROR: Failure in signing callback: 6 
93 3645 [OTA_task] ERROR: Handshake failed with error code -2003 
94 3645 [OTA_task] [ERROR][NET][36450] Failed to establish new connection. Socket status: -1002.
95 3645 [OTA_task] Network Connection failed, retry delay 5000 ms, retries left 99
96 3659 [wifi] Heap: current 59668 lowest 1600

sameerdasarwad, 2020-02-22:
@htibosch  i was successful with ota and shadow demo but when i am using IRAM_ATTR handler and  interrupts and my shadow size also increases so this all above logs are happens ,
when i calls both demo same time in app_main stack overflow happens for preventing this i have created separate task in shadow demo which call ota demo once it running ok but tls handshake fails

sameerdasarwad, 2020-02-22:
@htibosch  as you mention i saw again yes i am running out of heap so is there any chance to resolve this issue

htibosch, 2020-02-23:
Have you already changed you copy of `FreeRTOSIPConfig.h` ?
After changing, rebuild the whole project and check if the heap logging changes, there should be more heap available.

~~~c
+#define ipconfigIP_TASK_STACK_SIZE_WORDS   ( configMINIMAL_STACK_SIZE * 3 )
~~~

For me this worked fine, but make sure you don't get a stack overflow.

sameerdasarwad, 2020-02-23:
now reducing size help me little bit but still running out of heap and also even i mange to run both demo are they perform their operations  below are logs
```
24 1146 [iot_thread] [INFO ][MQTT][11460] MQTT library successfully initialized.
25 1146 [iot_thread] [INFO ][Shadow][11460] Shadow library successfully initialized.
26 1147 [iot_thread] [INFO ][DEMO][11470] Shadow Thing Name is ESP_246F2825D19C (length 16).
27 1148 [IP-task] Heap: current 103996 lowest 103920
28 1173 [wifi] Heap: current 123332 lowest 103724
29 1173 [iot_thread] DNS[0xD69B]: The answer to 'xxxxxxxxxxxxxx-ats.iot.eu-west-1.amazonaws.com' (22f8a986ip) will be stored
30 1174 [IP-task] Heap: current 109572 lowest 92688
I (12285) PKCS11: Initializing NVS partition: "storage"
31 1210 [IP-task] Heap: current 76048 lowest 75464
32 1213 [wifi] Heap: current 75228 lowest 73768
33 1234 [IP-task] Heap: current 72936 lowest 72436
34 1235 [wifi] Heap: current 72012 lowest 70740
35 1255 [wifi] Heap: current 72200 lowest 70552
36 1326 [wifi] Heap: current 68740 lowest 67348
37 1377 [wifi] Heap: current 67044 lowest 65652
38 1423 [IP-task] Heap: current 59820 lowest 58600
39 1459 [wifi] Heap: current 58704 lowest 58348
40 1459 [IP-task] Heap: current 59420 lowest 57824
41 1490 [wifi] Heap: current 53960 lowest 52352
42 1500 [wifi] Heap: current 54100 lowest 49472
43 1674 [wifi] Heap: current 48904 lowest 48828
44 1695 [wifi] Heap: current 50380 lowest 48400
45 1766 [wifi] Heap: current 43868 lowest 43808
46 1797 [wifi] Heap: current 43868 lowest 43524
47 1849 [iot_thread] [INFO ][MQTT][18490] Establishing new MQTT connection.
48 1849 [iot_thread] [INFO ][MQTT][18490] Anonymous metrics (SDK language, SDK version) will be provided to AWS IoT. Recompile with AWS_IOT_MQTT_ENABLE_METRICS set to 0 to disable.
49 1849 [iot_thread] [INFO ][MQTT][18490] (MQTT connection 0x3ffb40ec, CONNECT operation 0x3ffb6090) Waiting for operation completion.
50 1873 [iot_thread] [INFO ][MQTT][18730] (MQTT connection 0x3ffb40ec, CONNECT operation 0x3ffb6090) Wait complete with result SUCCESS.
51 1873 [iot_thread] [INFO ][MQTT][18730] New MQTT connection 0x3ffb3e10 established.
52 1873 [iot_thread] [INFO ][Shadow][18730] (ESP_246F2825D19C) Modifying Shadow DELTA callback.
53 1873 [iot_thread] [INFO ][Shadow][18730] (ESP_246F2825D19C) Adding new DELTA callback.
54 1873 [iot_thread] [INFO ][MQTT][18730] (MQTT connection 0x3ffb40ec) SUBSCRIBE operation scheduled.
55 1874 [iot_thread] [INFO ][MQTT][18730] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffd8c30) Waiting for operation completion.
56 1901 [iot_thread] [INFO ][MQTT][19010] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffd8c30) Wait complete with result SUCCESS.
57 1901 [iot_thread] [INFO ][Shadow][19010] (ESP_246F2825D19C) Shadow DELTA callback operation complete with result SUCCESS.
58 1901 [iot_thread] [INFO ][Shadow][19010] (ESP_246F2825D19C) Modifying Shadow UPDATED callback.
59 1901 [iot_thread] [INFO ][Shadow][19010] (ESP_246F2825D19C) Adding new UPDATED callback.
60 1902 [iot_thread] [INFO ][MQTT][19020] (MQTT connection 0x3ffb40ec) SUBSCRIBE operation scheduled.
61 1902 [iot_thread] [INFO ][MQTT][19020] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffb5e6c) Waiting for operation completion.
62 1931 [iot_thread] [INFO ][MQTT][19310] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffb5e6c) Wait complete with result SUCCESS.
63 1931 [iot_thread] [INFO ][Shadow][19310] (ESP_246F2825D19C) Shadow UPDATED callback operation complete with result SUCCESS.
64 1931 [iot_thread] [INFO ][MQTT][19310] MQTT library successfully initialized.
65 1932 [iot_thread] [INFO ][Shadow][19320] Shadow library successfully initialized.
66 1932 [iot_thread] [INFO ][MQTT][19320] (MQTT connection 0x3ffb40ec) SUBSCRIBE operation scheduled.
67 1932 [iot_thread] [INFO ][MQTT][19320] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffd8c30) Waiting for operation completion.
68 1962 [iot_thread] [INFO ][MQTT][19620] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffd8c30) Wait complete with result SUCCESS.
69 1962 [iot_thread] [INFO ][MQTT][19620] (MQTT connection 0x3ffb40ec) SUBSCRIBE operation scheduled.
70 1962 [iot_thread] [INFO ][MQTT][19620] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffb6108) Waiting for operation completion.
71 1992 [iot_thread] [INFO ][MQTT][19920] (MQTT connection 0x3ffb40ec, SUBSCRIBE operation 0x3ffb6108) Wait complete with result SUCCESS.
72 1992 [iot_thread] [INFO ][MQTT][19920] (MQTT connection 0x3ffb40ec) MQTT PUBLISH operation queued.
73 2023 [iot_thread] [INFO ][Shadow][20230] Shadow DELETE of ESP_246F2825D19C was ACCEPTED.
74 2023 [iot_thread] [INFO ][MQTT][20230] (MQTT connection 0x3ffb40ec) UNSUBSCRIBE operation scheduled.
75 2023 [iot_thread] [INFO ][MQTT][20230] (MQTT connection 0x3ffb40ec, UNSUBSCRIBE operation 0x3ffb9dcc) Waiting for operation completion.
76 2074 [iot_thread] [INFO ][MQTT][20740] (MQTT connection 0x3ffb40ec, UNSUBSCRIBE operation 0x3ffb9dcc) Wait complete with result SUCCESS.
77 2074 [iot_thread] [INFO ][MQTT][20740] (MQTT connection 0x3ffb40ec) UNSUBSCRIBE operation scheduled.
78 2074 [iot_thread] [INFO ][MQTT][20740] (MQTT connection 0x3ffb40ec, UNSUBSCRIBE operation 0x3ffb9b9c) Waiting for operation completion.
79 2105 [iot_thread] [INFO ][MQTT][21050] (MQTT connection 0x3ffb40ec, UNSUBSCRIBE operation 0x3ffb9b9c) Wait complete with result SUCCESS.
[... cut ...]
123 4665 [OTA_task] mbedTLS sign failed with error -17168 
124 4665 [OTA_task] ERROR: Failure in signing callback: 6 
125 4665 [OTA_task] ERROR: Handshake failed with error code -2003 
126 4666 [OTA_task] [ERROR][NET][46660] Failed to establish new connection. Socket status: -1002.
127 4667 [OTA_task] Network Connection failed, retry delay 5000 ms, retries left 100
128 4726 [wifi] Heap: current 62396 lowest 1804
129 5408 [iot_thread] [INFO ][MQTT][54080] (MQTT connection 0x3ffb40ec) MQTT PUBLISH operation queued.
```

sameerdasarwad, 2020-02-23:
@htibosch  @yanjos-dev  after some time later it got connected but when i try to perform shadow operation it show that connection closed mean that ota demo still killing shadow demo

sameerdasarwad, 2020-02-23:
@htibosch  i follow given link and i am successfully connected to both demo but when i am performing shadow action it  failed to send below are the logs and thanks for the above help ;)
```
202 6791 [OTA_task] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0
203 6891 [OTA_task] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0
204 6991 [OTA_task] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0
205 7087 [iot_thread] [INFO ][MQTT][70870] (MQTT connection 0x3ffb5fa8) MQTT PUBLISH operation queued.
206 7087 [iot_thread] [ERROR][NET][70870] Error -27648 while sending data.
207 7091 [OTA_task] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0
208 7191 [OTA_task] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0
209 7291 [OTA_task] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0
210 7391 [OTA_task] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0
211 7398 [iot_thread] [ERROR][NET][73980] Error -27648 while sending data.
212 7398 [iot_thread] [ERROR][MQTT][73980] (MQTT connection 0x3ffb5fa8) Failed to send PINGREQ.
213 7398 [iot_thread] [WARN ][NET][73980] Failed to close connection.
214 7398 [iot_thread] [INFO ][MQTT][73980] (MQTT connection 0x3ffb5fa8) Network connection closed.
215 7491 [OTA_task] State: WaitingForJob  Received: 2   Queued: 0   Processed: 0   Dropped: 0

```

htibosch, 2020-02-24:
I have pointed out how you can reduce the consumption of RAM in the FreeRTOS+TCP library by changing the configuration in `FreeRTOSIPConfig.h`.

It is also useful to define `configUSE_MALLOC_FAILED_HOOK` in your `FreeRTOSConfig.h`, and also define this call-back, usually in main.c:

~~~c
#if( configUSE_MALLOC_FAILED_HOOK != 0 )
    void vApplicationMallocFailedHook( void )
    {
        taskDISABLE_INTERRUPTS();
        {
            for( ;; )
            {
                /* Put a break on the next instruction. */
                portNOP();
            }
        }
        /* Will never reach here. */
        taskENABLE_INTERRUPTS();
    }
#endif /* ( configUSE_MALLOC_FAILED_HOOK != 0 ) */
~~~

I don't know what to think of the above logging. `Error -27648` doesn't sound like a valid errno.

sameerdasarwad, 2020-02-24:
@htibosch  yes consumption of RAM is reduced with your changes  thanks for the valuable information really appreciated, 
but above logs are just showing that their is still not enough space to send shadow update

sameerdasarwad, 2020-02-24:
and after connection established to ota, shadow pingrep fail

sameerdasarwad, 2020-02-24:
@ravibhagavandas is there any solution for this issues

abhidixi11, 2020-02-25:
Hello @sameerdasarwad, 

You should able to make use of shared MQTT connection to reduce heap usage. OTA and Shadow libraries both take reference to an MQTT connection. In your current implementation, you are creating two separate MQTT connections. Using single MQTT connection will reduce your heap usage. 
Instead of calling two demo function, you would create an MQTT connection and then pass it to the OTA and Shadow libraries. 
You can use code from _establishMqttConnection() in shadow demo code to create the shared connection. Once the connection is successfully established, pass the created connection to the shadow task and to the OTA task.
You will need to change demo code to make use of the shared connection. For example, in RunShadowDemo() you would remove code that calls _establishMqttConnection() instead would use shared MQTT connection.  Similarly, in vRunOTAUpdateDemo() you would remove MQTT connection logic and assign xConnection.xMqttConnection to the shared MQTT connection.

Please note, at this time we do not have a working example to demonstrate use of shared MQTT connection. We shall try to create a proof of concept showing use of shared MQTT connection, but meanwhile you can give it a try and let us know if that works or you have any question. 

Thanks.

sameerdasarwad, 2020-02-25:
@abhidixi11  thanks for valuable information i will give try on sharing connection might be it will works

shubhamkulkarni97, 2020-02-25:
Hi @sameerdasarwad As you are running out of heap memory, can you try building the example using LwIP networking stack?

Memory consumption of LwIP stack is significantly low as compared to FreeRTOS TCP stack.

To build using LwIP stack, run command `cmake -DVENDOR=espressif -DBOARD=esp32_wrover_kit -DCOMPILER=xtensa-esp32 -S . -B build -DAFR_ESP_LWIP=1`

sameerdasarwad, 2020-02-25:
@shubhamkulkarni97  i am using make so is there any other command

sameerdasarwad, 2020-02-25:
@abhidixi11  i have given try to use shared mqttconnection but still facing same issues of low heap while sending shadow payload and downloading ota image

shubhamkulkarni97, 2020-02-25:
> @shubhamkulkarni97 i am using make so is there any other command

set the environment variable AFR_ESP_LWIP using `export AFR_ESP_LWIP=1` and the build using make

abhidixi11, 2020-02-25:
Thanks @sameerdasarwad  Can you please send logs after you have changed code to use shared MQTT connection? I'm assuming that you did make some progress but got stuck in middle of doing OTA.  Thanks.

sameerdasarwad, 2020-02-26:
@abhidixi11  thank you so much for good idea and encouragement i successfull run both demo but after download ota job failed below are the logs 
`1099 17897 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock] 
1100 17958 [OTA Agent Task] [prvIngestDataBlock] Received file block 239, size 4096
1101 17958 [OTA Agent Task] [prvIngestDataBlock] Remaining: 3
1102 17958 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock] 
1103 17960 [OTA Agent Task] [INFO ][MQTT][179580] (MQTT connection 0x3ffb5f30) MQTT PUBLISH operation queued.
1104 17960 [OTA Agent Task] [prvRequestFileBlock_Mqtt] OK: $aws/things/ESP_246F2825D19C/streams/AFR_OTA-0e68aa19-5e9b-4d72-b15d-4a3857fd615b/get/cbor
1105 17960 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [RequestFileBlock] New state [WaitingForFileBlock] 
1106 17966 [iot_thread] State: WaitingForFileBlock  Received: 241   Queued: 0   Processed: 0   Dropped: 0
1107 18050 [OTA Agent Task] [prvIngestDataBlock] Received file block 240, size 4096
1108 18050 [OTA Agent Task] [prvIngestDataBlock] Remaining: 2
1109 18050 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock] 
1110 18076 [iot_thread] State: WaitingForFileBlock  Received: 242   Queued: 0   Processed: 0   Dropped: 0
1111 18111 [OTA Agent Task] [prvIngestDataBlock] Received file block 241, size 4096
1112 18112 [OTA Agent Task] [prvIngestDataBlock] Remaining: 1
1113 18112 [OTA Agent Task] [prvOTAAgentTask] Called handler. Current State [WaitingForFileBlock] Event [ReceivedFileBlock] New state [WaitingForFileBlock] 
I (181856) ota_pal: No such certificate file: arn:aws:acm:eu-west-1:956711000280:certificate/aff86477-6fbc-4ffd-9cc9-766c3b75c5a2. Using aws_ota_codesigner_certificate.h.

1114 18152 [OTA Agent Task] [prvIngestDataBlock] Received file block 242, size 3120
1115 18152 [OTA Agent Task] [prvIngestDataBlock] Received final expected block of file.
1116 18152 [OTA Agent Task] [prvStopRequestTimer] Stopping request timer.
1117 18186 [iot_thread] State: WaitingForFileBlock  Received: 244   Queued: 0   Processed: 0   Dropped: 0
1118 18297 [iot_thread] State: WaitingForFileBlock  Received: 244   Queued: 0   Processed: 0   Dropped: 0
1119 18321 [OTA Agent Task] [prvIngestDataBlock] File receive complete and signature is valid.
1120 18321 [OTA Agent Task] [prvStopRequestTimer] Stopping request timer.
1121 18321 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] Msg: {"status":"IN_PROGRESS","statusDetails":{"self_test":"ready","updatedBy":"0x1000000"}}
1122 18323 [OTA Agent Task] [INFO ][MQTT][183210] (MQTT connection 0x3ffb5f30) MQTT PUBLISH operation queued.
1123 18323 [OTA Agent Task] [INFO ][MQTT][183230] (MQTT connection 0x3ffb5f30, PUBLISH operation 0x3ffd8ab4) Waiting for operation completion.
I (183796) boot_comm: chip revision: 1, min. application chip revision: 0
I (183796) esp_image: segment 0: paddr=0x001a0020 vaddr=0x3f400020 size=0x2835c (164700) map
1124 18346 [OTA Agent Task] [INFO ][MQTT][183460] (MQTT connection 0x3ffb5f30, PUBLISH operation 0x3ffd8ab4) Wait complete with result SUCCESS.
1125 18346 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] 'IN_PROGRESS' to $aws/things/ESP_246F2825D19C/jobs/AFR_OTA-final_test/update
1126 18346 [OTA Agent Task] Received eOTA_JobEvent_Activate callback from OTA Agent.
I (183946) esp_image: segment 1: paddr=0x001c8384 vaddr=0x3ffbdb60 size=0x034e4 ( 13540) 
I (183966) esp_image: segment 2: paddr=0x001cb870 vaddr=0x40080000 size=0x00400 (  1024) 
0x40080000: _WindowOverflow4 at /home/firewires/Desktop/esp/starboard/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

I (183966) esp_image: segment 3: paddr=0x001cbc78 vaddr=0x40080400 size=0x04398 ( 17304) 
I (183986) esp_image: segment 4: paddr=0x001d0018 vaddr=0x400d0018 size=0xb0488 (722056) map
0x400d0018: _flash_cache_start at ??:?

1127 18408 [iot_thread] State: WaitingForFileBlock  Received: 244   Queued: 0   Processed: 0   Dropped: 0
I (184546) esp_image: segment 5: paddr=0x002804a8 vaddr=0x40084798 size=0x12758 ( 75608) 
0x40084798: set_chan_dig_gain at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:2462 (discriminator 1)

I (184616) boot_comm: chip revision: 1, min. application chip revision: 0
I (184616) esp_image: segment 0: paddr=0x001a0020 vaddr=0x3f400020 size=0x2835c (164700) map
I (184746) esp_image: segment 1: paddr=0x001c8384 vaddr=0x3ffbdb60 size=0x034e4 ( 13540) 
I (184756) esp_image: segment 2: paddr=0x001cb870 vaddr=0x40080000 size=0x00400 (  1024) 
0x40080000: _WindowOverflow4 at /home/firewires/Desktop/esp/starboard/amazon-freertos/freertos_kernel/portable/ThirdParty/GCC/Xtensa_ESP32/xtensa_vectors.S:1685

I (184756) esp_image: segment 3: paddr=0x001cbc78 vaddr=0x40080400 size=0x04398 ( 17304) 
I (184776) esp_image: segment 4: paddr=0x001d0018 vaddr=0x400d0018 size=0xb0488 (722056) map
0x400d0018: _flash_cache_start at ??:?

I (185346) esp_image: segment 5: paddr=0x002804a8 vaddr=0x40084798 size=0x12758 ( 75608) 
0x40084798: set_chan_dig_gain at /home/aiqin/git_tree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:2462 (discriminator 1)

[... cut ...]
94 1156 [iot_thread] [INFO ][MQTT][11560] (MQTT connection 0x3ffb5f30) MQTT PUBLISH operation queued.
95 1186 [IP-task] Network buffers: 14 lowest 13
96 1208 [iot_thread] [INFO ][DEMO][12080] Shadow was updated!
Previous: {"state":{"desired":{"switch_1":0,"switch_2":0,"switch_3":0,"switch_4":0,"otajobflag":0},"reported":{"appVersion":"1.097 1209 [iot_thread] [INFO ][Shadow][12090] Shadow UPDATE of ESP_246F2825D19C was ACCEPTED.
98 1320 [iot_thread] State: WaitingForJob  Received: 1   Queued: 0   Processed: 0   Dropped: 0
`

sameerdasarwad, 2020-02-26:
i think below logs are saying something.. :/

I (11000) ota_pal: prvPAL_SetPlatformImageState, 3
W (11000) ota_pal: Set image as invalid!
I (11010) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
I (11010) esp_ota_ops: [1] aflags/seq:0x1/0x2, pflags/seq:0x2/0x1
I (11020) esp_ota_ops: aws_esp_ota_set_boot_flags: 3 1
I (11030) esp_ota_ops: [1] aflags/seq:0x1/0x2, pflags/seq:0x2/0x1
68 1067 [OTA Agent Task] [prvParseJobDoc] Size of OTA_FileContext_t [64]
69 1067 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ clientToken: 0:ESP_246F2825D19C ]
70 1067 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ jobId: AFR_OTA-final_test ]
71 1067 [OTA Agent Task] [prvParseJSONbyModel] Identified parameter [ self_test ]
72 1067 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ updatedBy: 16777216 ]
73 1067 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ protocols: ["MQTT"] ]
74 1067 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ streamname: AFR_OTA-0e68aa19-5e9b-4d72-b15d-4a3857fd615b ]
75 1067 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ filepath: /var ]
76 1067 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ filesize: 994352 ]
77 1067 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ fileid: 0 ]
78 1067 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ certfile: arn:aws:acm:eu-west-1:956711000280:certificate/aff86477-6fbc-4ffd-9cc9-766c3b75c5a2 ]
79 1067 [OTA Agent Task] [prvParseJSONbyModel] Extracted parameter [ sig-sha256-ecdsa: MEUCIQCgWfBXPXL0tGPP2lh2+aB6Cwzo... ]
80 1067 [OTA Agent Task] [prvParseJobDoc] In self test mode.
81 1067 [OTA Agent Task] [prvParseJobDoc] Failing job. We rebooted and the version is still the same.
82 1077 [iot_thread] State: Ready  Received: 1   Queued: 0   Processed: 0   Dropped: 0
83 1077 [iot_thread] [INFO ][MQTT][10770] (MQTT connection 0x3ffb5f30) SUBSCRIBE operation scheduled.
84 1077 [iot_thread] [INFO ][MQTT][10770] (MQTT connection 0x3ffb5f30, SUBSCRIBE operation 

> 0x3ffb9d3c) Waiting for operation completion.
> I (11090) ota_pal: Disabling RTC hardware watchdog timer
> I (11230) esp_ota_ops: aws_esp_ota_get_boot_flags: 1
> I (11230) esp_ota_ops: [0] [...]

87 1091 [OTA Agent Task] [INFO ][MQTT][10910] (MQTT connection 0x3ffb5f30, PUBLISH operation 0x3ffb9b9c) Waiting for operation completion.
88 1098 [iot_thread] [INFO ][MQTT][10970] (MQTT connection 0x3ffb5f30, SUBSCRIBE operation 0x3ffb9d3c) Wait complete with result SUCCESS.
89 1098 [iot_thread] [INFO ][MQTT][10980] (MQTT connection 0x3ffb5f30) SUBSCRIBE operation scheduled.
90 1098 [iot_thread] [INFO ][MQTT][10980] (MQTT connection 0x3ffb5f30, SUBSCRIBE operation 0x3ffb5e74) Waiting for operation completion.
91 1116 [OTA Agent Task] [INFO ][MQTT][11160] (MQTT connection 0x3ffb5f30, PUBLISH operation 0x3ffb9b9c) Wait complete with result SUCCESS.
92 1116 [OTA Agent Task] [prvUpdateJobStatus_Mqtt] 'FAILED' to $aws/things/ESP_246F2825D19C/jobs/AFR_OTA-final_test/update
93 1116 [OTA Agent Task] [prvResetDevice] Attempting forced reset of device...
94 1119 [iot_thread] [INFO ][MQTT][11190] (MQTT connection 0x3ffb5f30, SUBSCRIBE operation 0x3ffb5e74) Wait complete with result SUCCESS.
95 1119 [iot_thread] [INFO ][MQTT][11190] (MQTT connection 0x3ffb5f30) MQTT PUBLISH operation queued.
96 1147 [iot_thread] [INFO ][Shadow][11470] Shadow UPDATE of ESP_246F2825D19C was ACCEPTED.
I (11990) wifi: state: run -> init (0)
I (11990) wifi: pm stop, total sleep time: 6359910 us / 9385944 us

``

sameerdasarwad, 2020-02-26:
@abhidixi11 thanks it works fine it was little mistake that ota job failed 
` 1067 [OTA Agent Task] [prvParseJobDoc] Failing job. We rebooted and the version is still the same.`
above line was hint and now ota and shadow demo both works fine with extra remaining :))
really appreciated all help
@yanjos-dev @htibosch @shubhamkulkarni97 @abhidixi11  thank you all of you

closed 2020-02-26 by sameerdasarwad