aws/amazon-freertos #1788: CMake support for Xilinx MicroZed

cdelledonne, 2020-02-25:
I was wondering when the Xilinx MicroZed board will support CMake, is that planned for the near future? What would be needed to support CMake?

ravibhagavandas, 2020-02-27:
@cdelledonne 

Currently, we don't have a plan to support CMake for Xilinx MicroZed  in near future. We faced some challenges in the past, trying to make  XSDK( Xilinx Software Development Kit) IDE steps work with CMake for building the project. One of the issue was to program the FPGA binary as mentioned [here](https://www.xilinx.com/html_docs/xilinx2018_1/SDK_Doc/SDK_tasks/sdk_t_configure_fpga.html) via CMake commandline. Also we found other issues having a command line tool to flash the binary onto device.

cdelledonne, 2020-02-28:
@ravibhagavandas ACK!

ravibhagavandas, 2020-02-28:
I am closing this issue. Feel free to re-open if you have additional questions.

closed 2020-02-28 by ravibhagavandas

cdelledonne, 2020-05-11:
@ravibhagavandas I've started to look into this again today, here are my findings.

It seems that Xilinx has moved from XSDK to [Vitis Unified Software Platform](https://www.xilinx.com/products/design-tools/vitis/vitis-platform.html). To begin with, I would think that the [MicroZed Getting Started Guide](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_xilinx.html) should be updated.

The installation of Vitis includes the [Xilinx Software Command-Line Tool (XSCT)](https://www.xilinx.com/html_docs/xilinx2018_1/SDK_Doc/xsct/intro/xsct_introduction.html), through which I've managed to program the target FPGA and upload the binary onto the device. The `xsct` utility is in `<path_to_vitis>/<version>/bin/xsct`, so I guess to make it work with your CMake framework one should either have that in their `PATH`, or pass some command-line options to specify the path as it's done with the toolchain (`-DAFR_TOOLCHAIN_PATH`).

`xsct` can be run as an interactive tool, or passed a Tcl script as input. What has worked for me is running `xsct myscript.tcl`, with `myscript.tcl` being the following Tcl script, which contains commands to connect to the target, program the FPGA, initialize the PS and upload the binary:
```tcl
# Connect to local hw_server
connect
# Select target based on name
targets -set -filter {name =~ "ARM Cortex-A9 MPCore #0"}
# Reset the system before initializing the PS and configuring the FPGA
rst
# Configure the FPGA with the bitstream (not provided in amazon-freertos?)
fpga /path/to/project/hw/somename.bit
# Source the ps7_init.tcl script and run ps7_init and ps7_post_config commands
source /path/to/project/hw/ps7_init.tcl
ps7_init
ps7_post_config
# Download the binary onto the device
dow /path/to/project/build/example.elf
# Resume the processor
con
```

I noticed a `reset.tcl` script in `vendors/xilinx/boards/microzed/aws_demos`, which is intended to be run with `xsct`, so it seems that there were some attempts at using `xsct`. Not sure if that was explored further though.

---

Some additional feedback regarding the compilation of `amazon-freertos` for the MicroZed board follows.

I've downloaded the source code from the AWS IoT console, selecting the predefined configuration for the "Microzed Industrial IoT Bundle". All good.

I've tried to run
```
cmake -DBOARD=microzed -DCOMPILER=arm-gcc -S . -B build
```
but that resulted in an error, apparently because the following include path
https://github.com/aws/amazon-freertos/blob/76720527c1d0289d6c396f486e258ece4ec0c7cb/vendors/xilinx/boards/microzed/CMakeLists.txt#L115
cannot be found. If I comment this line out, CMake seems to successfully generate the build system.

Another apparent problem is that the pre-compiled libraries in `vendors/xilinx/aws_bsp/ps7_cortexa9_0/lib/` seem to be built with some compiler that is not the `arm-gcc` I want to use. Running `make` from `vendors/xilinx/aws_bsp` seems to fix this problem.

Finally, there's one more linker error which I haven't tracked down yet:
```
.../vendors/xilinx/aws_bsp/ps7_cortexa9_0/lib/libxil.a(inbyte.o) uses VFP register arguments, aws_demos.elf does not
```
which might be due to both `-mfloat-abi=hard` and `-mfloat-abi=soft` linker flags being passed somewhere across the project.

---

Sorry for the long comment, I hope it can help give some directions to support the CMake framework for the MicroZed. Clearly, I'd be willing to contribute actively if needed.