aws/amazon-freertos #1795: vLoggingPrintf assert failed on IotMqtt_Init();

aviler, 2020-02-27:
**Describe the bug**
Just trying to "recreate" mqtt_demo the code only calls IotMqtt_Init(); but when run keeps rebooting then I got logging errors like in the gist below

**System information**
- Which hardware board or part numbers?
ESP32 DEVKITV1

- IDE used
VSCode

- Operating System [Windows|Linux|MacOS]
Linux

- Version of FreeRTOS (run ``git describe --tags`` to find it)
201912.00-96-g306c3bca9

- Project [Custom Application|MQTT|Shadow|Greengrass|OTA|Device Defender|TCP Echo|Sub pub Demo|Simple TCP Server]
Just MQTT

- If your project is a Custom Application, please add the relevant code snippet in the section `Code to reproduce the bug`. 

**Expected behavior**
Well I'm just calling the init() so I would expect IOT_MQTT_SUCCESS or even FAILURE. But definetelly not the ESP32 rebooting with panic

**GIST with Code and Error logs**
https://gist.github.com/aviler/70594006336b47bde64050d5bad769c9

Thank you!

htibosch, 2020-02-27:
It sounds like `xQueue` or the logging task could not be created.
Could you put some logging into `xLoggingTaskInitialize()`?

aviler, 2020-03-01:
Hi @htibosch ! Thanks for the reply.

I added a esp-idf log in the beginning of the `xLoggingTaskInitialize()` and you're right is not been called. Thanks again!

closed 2020-03-01 by aviler