aws/amazon-freertos #1799: Wi-Fi provisioning without executing the demo code

TheIronNinja, 2020-02-28:
**Is your feature request related to a problem? Please describe.**  
Hello, I am working on using FreeRTOS for a custom application. I've managed to build it properly following the guide at https://docs.aws.amazon.com/freertos/latest/qualificationguide/building-cmake.html and the comments made at #832, but once I pair the ESP with the phone the device seems to not know what to do with that. When executing the demo I could test the app on the phone and it worked great; it showed a little display that I had to press to confirm the pairing once the devices connected. But if I try to execute the same process without calling the DEMO_RUNNER but the functions it calls instead (so I can expand the code from there) it stops working. Specifically, it stops after the numeric comparison.

**Describe the solution you would like.**  
As people said in #832, some more documentation to do this process would be nice. For now, I'd like to know exactly what should I do for this part of the program to work, as I could figure out the rest with some work.

**Additional context**  
The code I want to substitute basically is this:
`void DEMO_RUNNER_RunDemos( void )
{
    /* These demos are shared with the C SDK and perform their own initialization and cleanup. */

    static demoContext_t mqttDemoContext =
    {
        .networkTypes                = democonfigNETWORK_TYPES,
        .demoFunction                = DEMO_entryFUNCTION,
        .networkConnectedCallback    = DEMO_networkConnectedCallback,
        .networkDisconnectedCallback = DEMO_networkDisconnectedCallback
    };

    Iot_CreateDetachedThread( runDemoTask,
                              &mqttDemoContext,
                              democonfigDEMO_PRIORITY,
                              democonfigDEMO_STACKSIZE );
}`

With the DEMO_entryFUNCTION in this case being defined as vGattDemoSvcInit.

I'm trying to call the vGattDemoSvcInit, copying the runDemoTask code and executing it modifying some texts (so the final code doesn't say DEMO STARTING and so on), but I am still missing something to make it work.

Thank you!

ravibhagavandas, 2020-02-28:
The demo runner initializes the required libraries, and enables networks (for WiFi provisioning, its BLE and WiFi) [here](https://github.com/aws/amazon-freertos/blob/master/demos/demo_runner/iot_demo_freertos.c#L200). Using network manager it also waits for a network to be connected, using which you can run the demo.

 If you wan't one of our demo for example, MQTT demo to run after a WiFi network is connected, you can set the respective demo in board specific config [here](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_demo_config.h#L46) and its network type to only WiFi [here](https://github.com/aws/amazon-freertos/blob/master/vendors/espressif/boards/esp32/aws_demos/config_files/aws_demo_config.h#L55).

If you are not using the demo runner and have your own version of demo, then you may still need to call these initialization functions before calling modified version of the demo.

TheIronNinja, 2020-03-02:
So I didn't manage to get it working with the numeric comparison enabled. However, I changed to "Just Works" pairing and this one works perfectly. Because this is the version I'll be using from now on I won't be following the original issue anymore (at least for now), so feel free to close the thread.

In case this was really a bug, and just for documentation purposes in case anyone stumbles upon this thread, here's what I saw running only the Wifi-prov demo (Numeric comparison):
ESP waiting for connection -> Phone detects ESP -> Tap to enable the pairing in the phone -> ESP asks for Y/N while showing a number. Phone does the same with the same number. -> If both agreed, the pairing is completed and I can give the credentials.

And this is what I saw with my custom code (Numeric comparison):
ESP waiting for connection -> Phone detects ESP -> Tap to enable the pairing in the phone -> ESP asks for Y/N while showing a number. Phone does nothing. -> Nothing happens until a timeout occurs and they disconnect from eachother.

"Just works" pairing works properly with the same code, just changing the config and CONFIG_NIMBLE_SM_LEGACY=y in the sdkconfig.

closed 2020-03-02 by ravibhagavandas