aws/amazon-freertos #1813: esp32 partitions table

chegewara, 2020-03-05:
Its not a bug, more like a question.
Is there reason that partition table is not using whole available flash:
```
nvs,      data, nvs,     0x10000,  0x6000
otadata,  data, ota,     0x16000,  0x2000
phy_init, data, phy,     0x18000,  0x1000
ota_0,    0,    ota_0,   0x20000,  1500K
ota_1,    0,    ota_1,   ,         1500K
storage,  data, nvs,     ,         0x10000
```
Its a waste of 800kB of flash.

mahavirj, 2020-03-11:
@chegewara Default partition table is for reference purpose only, hence tuned for maximum firmware size that can potentially happen with various Amazon-FreeRTOS features. Keeping `ota_0`/`ota_1` to optimal sizes also allows for quick development cycle due to flash erase/write cycles. For production, there could be some additional (product specific) partitions (e.g. extra-storage, NVS encryption keys, custom-partitions etc.), hence sufficient headroom is kept available. 

If you have any other suggestion or opinion, we can consider that and make relevant changes as well.

chegewara, 2020-03-11:
> If you have any other suggestion or opinion, we can consider that and make relevant changes as well.

Its not big deal, i can adjust partitions table in my projects, just it pop up when ive been moving to AFR as component option. I had very old partition table in which storage partition start address was 0x320000 and all scripts to build and flash certificates are prepared to work with that address. 
Now starting address is 0x317000 and i missed it at some point. 
Also with partition size 1500K esp-idf is doing memory aligning and real partition size is slightly bigger than 1500K.

Since i have all answers i will close this issue, thanks.

closed 2020-03-11 by chegewara