aws/amazon-freertos #1818: GattServerDemo does not work with Nordic

jnz86, 2020-03-07:
I can't tell if this applies to other systems. Using nordic and 202002.00, changing the demo type in aws_demo_config.h to

`#define CONFIG_BLE_GATT_SERVER_DEMO_ENABLED`

Does not infer the correct change in iot_ble_config_defaults.h to
`IOT_BLE_ADD_CUSTOM_SERVICES 1`

Result is that the custom services are never added to the BLE init.

Also, with out adjusting sdk_config.h to
`#ifndef NRF_SDH_BLE_GATTS_ATTR_TAB_SIZE
#define NRF_SDH_BLE_GATTS_ATTR_TAB_SIZE (1200*6)
#endif

// <o> NRF_SDH_BLE_VS_UUID_COUNT - The number of vendor-specific UUIDs.
#ifndef NRF_SDH_BLE_VS_UUID_COUNT
#define NRF_SDH_BLE_VS_UUID_COUNT 16
#endif
`
The stack either fails to initialize, or the "custom UUIDs" won't show.

The ATTB TABLE SIZE is a magic number. Some explanation of how to set this (in this case lowering it) would be handy in the readme or documentation.

This may not appear to be an issue if using iphone with a previously working connection. Any time the gatts are changed, requies to "forget this device" in bluetooth settings, and cycle bluetooth control off and on again.

**System information**
- nrf52840
- Segger
- Win10
- GattServerDemo 

**Additional context**
I can not confirm the counter code actually works. Have two characteristics load. The first with READ and NOTIFY, works fine although always reads zero. The other is a READ and WRITE, this one seems to not be handled correctly as READ returns no data, and write seems to have no effect I can determine. After using the second, for me, BLE becomes unresponsive entirely and will eventually disconnect.

Thank you!

ravibhagavandas, 2020-03-10:
@jnz86 

Regarding the GATT demo configuration question:

> Does not infer the correct change in iot_ble_config_defaults.h to
IOT_BLE_ADD_CUSTOM_SERVICES 1

It doesn't infer this flag automatically when `CONFIG_BLE_GATT_SERVER_DEMO_ENABLED` flag is enabled. So as mentioned in the document [here](https://docs.aws.amazon.com/freertos/latest/userguide/ble-demo.html#ble-demo-server) you need to set both flags to enable GATT server demo.

Regarding the issue, GATT demo not working, this is because as you mentioned, with all GATT services enabled, Nordic port is reaching the maximum attribute table size limit.

As a temporary fix, if you are not transferring MQTT packets over BLE, you can disable  MQTT and data transfer GATT services, by setting following configs to 0 in `vendors/<vendor>/boards/<board>/aws_demos/config_files/iot_ble_config.h`

```
#define IOT_BLE_ENABLE_MQTT    ( 0 )
#define IOT_BLE_ENABLE_DATA_TRANSFER_SERVICE ( 0 )
```

With this I was able to see the demo working ( you need to forget bond, turn off - turn on BLE on mobile  if needed ). Please let us know if this works.

We will work with vendor on a long term fix for this.

jnz86, 2020-03-10:
R,

 

Ok. To be fair… I did not consult the documentation! Good point. 

 

1.	IMO, if there is a config_files directory, the config for custom_services should be in there somewhere. It seems odd to move that outside the demo/user area. I’d like to see this in a place I could modify without altering code “inside” the AWS FR library.
2.	“Nordic port is reaching the maximum attribute table size limit." There is more to it than that. Using the AWS defaults of TABLE = (1408*6) and VendorUUIDs 15, it doesn’t work. But increasing the vendor uuids to 19 and DECREASING the table size down to 1300*6 does – for me – show all the UUIDs I expect to see including READ/WRITE and READ/NOTIFY on UUID 0xC6F2…00. It would be great to get an indication of why this is a magic number and how to better guess at what the values should be.

 

Thanks!

ravibhagavandas, 2020-03-12:
Regarding
1. The config for custom services is already under config_files directory in file `iot_ble_config.h`. User can set the config here without changing Amazon FreeRTOS code, for example `#define IOT_BLE_ADD_CUSTOM_SERVICES  ( 1 ) ` . If not set,  the default config defined in `iot_config_defaults.h` will take into effect.

2. Glad that the configuration works for you. Nordic documentation specifies to try adjusting these parameters until the API does not return a no memory error. There is no clear guidelines as to by what percentage should it be increased when adding a new service.  `NRF_SDH_BLE_VS_UUID_COUNT` is for number of different 128bit base UUIDs, so it should be increased by 1 for each of 128 bit service UUID added. `NRF_SDH_BLE_VS_UUID_COUNT` and `NRF_SDH_BLE_GATTS_ATTR_TAB_SIZE ` together contributes to overall size, so `NRF_SDH_BLE_GATTS_ATTR_TAB_SIZE ` needs to be adjusted as well to fit the whole attribute table in whole RAM. 
For more details and a better explanation, I would also suggest going through some of Nordic forum posts regarding this and posting the question in [Nordic forum](https://devzone.nordicsemi.com/f/nordic-q-a).

jnz86, 2020-03-12:
Ok.

You're entirely right, the way everything is ifndef, you can just add it to the iot_ble_config.h. That's a good idea. I suppose this is the reason all files have a comment to load that first. This is obvious now that you say it. I wonder was this intention documented somewhere and I missed it?

Well, for the next person that discovers that the ATTR_TAB_SIZE is a nearly irrational number and that Nordic's own recommendation is ¯\_(ツ)_/¯ "try different numbers", some starter documentation would be helpful. I was very surprised that lowering the table size and keeping the same UUID count fixed a build issue I was having. It makes no sense unless inside the soft device they are allocating memory for something that would be used by table.

Alright. Everything makes sense. I'll close this out.

closed 2020-03-12 by jnz86