aws/amazon-freertos #1819: BLE callbacks are being trigged multiple times per event

jnz86, 2020-03-07:
**Describe the bug**
I can't confirm this is a bug, but registering a callback with IotBle_CreateService() functions as expected but the callback (for example characterisitc READ) is invoked twice per actual BLE command.

**System information**
- nrd52840-DK
- Segger
- Win10
- 202002.00
- GattServerDemo

**Expected behavior**
That an event callback is called exactly once per event

**Screenshots or console output**
```
8 193 [BLE] [ERROR][BLE_HAL][1930] UNKNOWN BLE EVENT 34.
9 197 [BLE] [ERROR][BLE_HAL][1970] UNKNOWN BLE EVENT 35.
10 772 [BLE] [INFO ][DEMO][7720] X
11 772 [BLE] [INFO ][DEMO][7720] X
```
**To reproduce**
I put a LOG("X") inside the callback.

**Code to reproduce the bug**
`aws_ble_gatt_server_demo.c  line 329`
```
void vReadCounter( IotBleAttributeEvent_t * pEventParam )
{
    IotBleReadEventParams_t * pxReadParam;
    IotBleAttributeData_t xAttrData = { 0 };
    IotBleEventResponse_t xResp;

    xResp.pAttrData = &xAttrData;
    xResp.rspErrorStatus = eBTRspErrorNone;
    xResp.eventStatus = eBTStatusFail;
    xResp.attrDataOffset = 0;

    if( pEventParam->xEventType == eBLERead )
    {
        pxReadParam = pEventParam->pParamRead;
        xResp.pAttrData->handle = pxReadParam->attrHandle;
        xResp.pAttrData->pData = ( uint8_t * ) &ulCounter;
        xResp.pAttrData->size = sizeof( ulCounter );
        xResp.attrDataOffset = 0;
        xResp.eventStatus = eBTStatusSuccess;
        IotBle_SendResponse( &xResp, pxReadParam->connId, pxReadParam->transId );
    }
    IotLogInfo( "X\n" );
                
}
```

**Additional context**
This could be entirely expected behavior.

Thank you!

ravibhagavandas, 2020-03-11:
Hi @jnz86 ,

The demo has a `vCounterUpdateTaskFunction` running in a task which updates the counter value every 1 second. It also sends a notification of the update on the characteristic to the BLE peer. BLE peer issues a read request for every notification (as per GATT protocol ), for which the read callback should be invoked. So in normal behavior you should read callback invoked every 1 second to read the value.

Could you elaborate more on the behavior you are seeing ? Are you seeing callbacks invoked twice within 1 second interval ?

jnz86, 2020-03-11:
Inside and outside of this demo. Every BLE event on every characteristic, the callbacks are invoked twice on read/write, and sometimes three times on enabling notify. They have the same time stamp, so within a millisecond.

See my example code. I put a print(”X”) in the callback, then see the log I posted where two X’s show up every atomic read action the phone invokes. It should be extremely easy to verify by adding the same print() I posted. In any BLE callback, but using the custom services demo as an example.

This happens in my actual application, on the demos, anywhere the Nordic BLE is invoked with the Amazon middleware so far as I have tested.

What I can’t tell is if this is correct behavior, that if the Nordic Softdevice is triggering the AFR middleware to call the callback twice, or if there is a bug where the Nordic side would only call once but somehow each BLE event gets two callbacks. 

So maybe nothing is wrong at all, I don’t know. If there are supposed to be two invocations per event, that should probably be documented somewhere.

ravibhagavandas, 2020-03-11:
There should not be two callback invocations per event generated from Nordic port layer ([here](https://github.com/aws/amazon-freertos/blob/master/vendors/nordic/boards/nrf52840-dk/ports/ble/iot_ble_hal_common_gap.c#L379)) or BLE Middleware([here](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/standard/ble/src/iot_ble_gatt.c#L482)). So I suspect in your case two events are generated from underlying stack.

I  included the log statement as you described above in `aws_ble_gatt_server_demo.c vReadCounter()`,  but couldn't reproduce the behavior at my end:
```
 618 [iot_thread] [INFO ][DEMO][6180] Successfully initialized the demo. Network type for the demo: 2
7 692 [BLE] [ERROR][BLE_HAL][6920] UNKNOWN BLE EVENT 34.
8 696 [BLE] [ERROR][BLE_HAL][6960] UNKNOWN BLE EVENT 35.
9 902 [BLE] [INFO ][DEMO][9020] Enabled Notification for Read Characteristic

10 1094 [BLE] [INFO ][DEMO][10940] X

11 1193 [BLE] [INFO ][DEMO][11930] X

12 1292 [BLE] [INFO ][DEMO][12920] X

13 1391 [BLE] [INFO ][DEMO][13910] X
```

As  you see, all logs are generated at different ticks.

I used the IOS demo app from [amazon-freertos-ble-ios-sdk](https://github.com/aws/amazon-freertos-ble-ios-sdk) as the BLE central and Nordic NR52840 board running as peripheral with FreeRTOS version 202002.00 and GATT server demo.

To get more hints about the stack behavior in your case, could you also log events from nordic stack [here](https://github.com/aws/amazon-freertos/blob/master/vendors/nordic/boards/nrf52840-dk/ports/ble/iot_ble_hal_common_gap.c#L300)

jnz86, 2020-03-11:
Hmm. I don't have a mac so I can't build the IOS SDK... I'll try another generic BLE app, and I'll go into common_gap and see if I can debug at least which event is being called. The log for the generic BLE tools I'm using (LightBlue, nrfConnect, and BLE Hero) do not show multiple reads or events on their ends.

I was just assuming this was some obscure BLE stack thing. Now I'm interested and will post back here.

I'll also confirm on an iphone 7, using an 11 pro for primary use.

jnz86, 2020-03-12:
R,

Tested again. iPhone11 pro, iPhone 7. Definitely confirmed this is happening. It's a responseConfirmation event. What I don't remember is if READ commands get responses in BLE.. and on top of that only sometimes like Write() / WriteWithoutResponse(). Again, this could be expected, but it doesn't seem right for a Read().

`iot_ble_gatt.c`
```
static void _responseConfirmationCb( BTStatus_t status,
                                     uint16_t handle )
{
    IotBleRespConfirmEventParams_t respConfirmParam;
    IotBleAttributeEvent_t eventParam;
    IotBleAttributeEventCallback_t eventsCallbacks;

    if( _getCallbackFromHandle( handle, &eventsCallbacks ) == true )
    {
        respConfirmParam.handle = handle;
        respConfirmParam.status = status;

        eventParam.xEventType = eBLEResponseConfirmation;
        eventParam.pParamRespConfirm = &respConfirmParam;

        eventsCallbacks( &eventParam );    //Right here, calls back to registered CB even on read
    }
}
```

`aws_ble_gatt_server_demo.c`
```
void vReadCounter( IotBleAttributeEvent_t * pEventParam )
{
IotLogInfo("ENTER");                                        //Twice
    IotBleReadEventParams_t * pxReadParam;
    IotBleAttributeData_t xAttrData = { 0 };
    IotBleEventResponse_t xResp;

    xResp.pAttrData = &xAttrData;
    xResp.rspErrorStatus = eBTRspErrorNone;
    xResp.eventStatus = eBTStatusFail;
    xResp.attrDataOffset = 0;

    if( pEventParam->xEventType == eBLERead )
    {

  IotLogInfo("BLE READ");                                                   //Once
        pxReadParam = pEventParam->pParamRead;
        xResp.pAttrData->handle = pxReadParam->attrHandle;
        xResp.pAttrData->pData = ( uint8_t * ) &ulCounter;
        xResp.pAttrData->size = sizeof( ulCounter );
        xResp.attrDataOffset = 0;
        xResp.eventStatus = eBTStatusSuccess;
        IotBle_SendResponse( &xResp, pxReadParam->connId, pxReadParam->transId );
    }
IotLogInfo("LEAVE");                                                          //Twice, look at the order though
}
```

Console
```
12 612 [BLE] [INFO ][DEMO][6120] ENTER
13 612 [BLE] [INFO ][DEMO][6120] BLE READ
14 612 [BLE] [INFO ][DEMO][6120] ENTER
15 612 [BLE] [INFO ][DEMO][6120] LEAVE
16 612 [BLE] [INFO ][DEMO][6120] LEAVE
```

The callback never leaves the first time before being called again. I'm open to this being spec I forgot or never knew, and this does seem correct for writes that need responses.

What I can't figure out is why the IOS SDK wouldn't show this behavior. Every generic BLE client I used brought it up. The event is value is eBLEResponseConfirmation (0x05).

If it's still not showing with the IOS app, please try a generic BLE discovery tool like LightBlue or BLE Hero and confirm.

ravibhagavandas, 2020-03-12:
Thank you for more detailed description.

The event type `eBLEResponseConfirmation` or `eBLEIndicationConfirmReceived` are valid event types for the characteristics, so this is an expected behavior. Those are invoked on the respective callbacks when the stack has successfully send back response to a read event or it sends a confirmation that indications/notifications are sent.

So tools like `LightBlue` could be issuing a  GATT read request in response to a GATT notification of counter update on that characteristic. This is valid BLE behavior. Apps like `FreeRTOS BLE app` or `NRFConnect` is not issuing a GATT read request, as data is already in the GATT notification. This is also a valid behavior and could be treated as an optimization as data is already in notification.

ravibhagavandas, 2020-03-12:
To get more insights, you could also log the event types as well by:
`IotLogInfo("Event:%d", pEventParam->xEventType);`
These are the event types:
```
typedef enum
{
    eBLERead,                      /**< Read event. */
    eBLEWrite,                     /**< Write event. */
    eBLEWriteNoResponse,           /**< Write event, no response required. */
    eBLEWritePrepare,              /**< Prepare Write event. */
    eBLEExecWrite,                 /**< Execute Write event. */
    eBLEResponseConfirmation,      /**< Confirmation from remote device. */
    eBLEIndicationConfirmReceived, /**< Received confirm to indication from remote device. */
} IotBleAttributeEventType_t;
```
For me, the event type maps to `eBLEIndicationConfirmReceived` which is for the notifications sent out through the characteristic.

jnz86, 2020-03-12:
Ok.

So... If I understand... the App's command is Read(), this is received by the device, the event callback is issued, the data is manually selected and sent back, it's at that point that if the data was loaded in to the softdevice correctly that a responseConfirmation event will trigger the same callback again.

I suppose I could understand a possible scenario where you load data into the void IotBle_SendResponse() and want to know it will eventualy (but hasn't yet!) go out over the radio, but that's what a function return should do, not calling the same callback you're already in before you've left it the first time. 

*This "confirmation" is that you loaded data into the softdevice. It can't possibly have gone out over the radio yet. We know that because it's the same timestamp and the first callback hasn't had time to leave the eventCallback before a second invocation happens.*

For what it's worth nrfConnect app does the same thing. I really can't understand how the FR BLE App would not be doing this. We're not talking about notifications being enabled, this is just atomic read, it makes no sense.

I'm fine with letting this go and closing it out. Other than putting a little more on the stack for a "recursive" callback there isn't a terrible inefficiency to clear up. Maybe a couple hundred instructions. Still, I'm just not seeing the use-case here compared to a function that just returned a positive.

Closing it out, thanks for taking a look.

closed 2020-03-12 by jnz86